miao1007/Openwrt-NetKeeper #2: 用中文可以吗？重庆地区的netkeeper用户，问题是不能拨号。非常感谢！

Jerry6666, 2015-04-09:
[image]
这是日志截图。我不知道username那儿出现了换行有没有影响。万分感谢！

miao1007, 2015-04-10:
已经安装成功了，可能是你的内外网的静态路由表配置错误了

Jerry6666, 2015-04-10:
好的，我下午再研究一下。真是太感谢了。对了，还有一个问题，就是我这样配置好，reboot之后，用pc浏览器打开直接跳转到学校内网的登陆界面了；而且，进192.168.1.1都进不去了。这是什么问题呢？

miao1007, 2015-04-10:
是你桥接了或者 LAN-WAN插反了吧....

Jerry6666, 2015-04-11:
酱紫啊。我看看。我不是特别懂这些。。。谢谢啦

Jerry6666, 2015-04-13:
亲，我又来了( ╯□╰ )都不好意思继续问你了。但是捣鼓了好久，还是一样的问题。连日志都没变( ╯□╰ )你能告诉我这是什么问题吗？拜谢

Jerry6666, 2015-04-14:
Tue Apr 14 19:53:14 2015 daemon.warn odhcpd[730]: DHCPV6 SOLICIT IA_NA from 0001000118ca2cab208984472ec5 on br-lan: ok fd6a:44bd:6e9e::5ac/128 
Tue Apr 14 19:53:14 2015 daemon.warn odhcpd[730]: DHCPV6 SOLICIT IA_NA from 0001000117ac2db1b8763fec7e9b on br-lan: ok fd6a:44bd:6e9e::c3c/128 
Tue Apr 14 19:53:17 2015 daemon.warn odhcpd[730]: DHCPV6 SOLICIT IA_NA from 000100011c6a024070188bdbb123 on br-lan: ok fd6a:44bd:6e9e::bcf/128 
Tue Apr 14 19:53:17 2015 daemon.info dnsmasq[995]: read /etc/hosts - 1 addresses
Tue Apr 14 19:53:17 2015 daemon.info dnsmasq[995]: read /tmp/hosts/odhcpd - 8 addresses
Tue Apr 14 19:53:17 2015 daemon.info dnsmasq[995]: read /tmp/hosts/dhcp - 1 addresses
Tue Apr 14 19:53:18 2015 daemon.warn odhcpd[730]: DHCPV6 REQUEST IA_NA from 000100011c6a024070188bdbb123 on br-lan: ok fd6a:44bd:6e9e::bcf/128 
Tue Apr 14 19:53:18 2015 daemon.info dnsmasq[995]: read /etc/hosts - 1 addresses
Tue Apr 14 19:53:18 2015 daemon.info dnsmasq[995]: read /tmp/hosts/odhcpd - 9 addresses
Tue Apr 14 19:53:18 2015 daemon.info dnsmasq[995]: read /tmp/hosts/dhcp - 1 addresses
Tue Apr 14 19:53:18 2015 daemon.warn odhcpd[730]: DHCPV6 SOLICIT IA_NA from 0001000118ca2cab208984472ec5 on br-lan: ok fd6a:44bd:6e9e::5ac/128 
Tue Apr 14 19:53:18 2015 daemon.warn odhcpd[730]: DHCPV6 SOLICIT IA_NA from 0001000117ac2db1b8763fec7e9b on br-lan: ok fd6a:44bd:6e9e::c3c/128 
Tue Apr 14 19:53:26 2015 daemon.warn odhcpd[730]: DHCPV6 SOLICIT IA_NA from 0001000118ca2cab208984472ec5 on br-lan: ok fd6a:44bd:6e9e::5ac/128 
Tue Apr 14 19:53:26 2015 daemon.warn odhcpd[730]: DHCPV6 SOLICIT IA_NA from 0001000117ac2db1b8763fec7e9b on br-lan: ok fd6a:44bd:6e9e::c3c/128 
Tue Apr 14 19:53:28 2015 daemon.warn pppd[2609]: Timeout waiting for PADO packets
Tue Apr 14 19:53:28 2015 daemon.err pppd[2609]: Unable to complete PPPoE Discovery
Tue Apr 14 19:53:28 2015 daemon.info pppd[2609]: Exit.
Tue Apr 14 19:53:28 2015 daemon.notice netifd: Interface 'NetKeeper' is now down
Tue Apr 14 19:53:28 2015 daemon.notice netifd: Interface 'NetKeeper' is disabled
Tue Apr 14 19:53:28 2015 daemon.notice netifd: Interface 'NetKeeper' is enabled
Tue Apr 14 19:53:28 2015 daemon.notice netifd: Interface 'NetKeeper' is setting up now
Tue Apr 14 19:53:28 2015 daemon.notice netifd: VLAN 'eth0.2' link is down
Tue Apr 14 19:53:28 2015 daemon.notice netifd: Interface 'NetKeeper' has link connectivity loss
Tue Apr 14 19:53:28 2015 daemon.notice netifd: VLAN 'eth0.2' link is up
Tue Apr 14 19:53:28 2015 daemon.notice netifd: Interface 'NetKeeper' has link connectivity 
Tue Apr 14 19:53:28 2015 daemon.info pppd[2657]: Plugin rp-pppoe.so loaded.
Tue Apr 14 19:53:28 2015 daemon.info pppd[2657]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Tue Apr 14 19:53:28 2015 daemon.info pppd[2657]: Plugin sxplugin.so loaded.
Tue Apr 14 19:53:28 2015 daemon.info pppd[2657]: sxplugin : init
Tue Apr 14 19:53:28 2015 daemon.info pppd[2657]: sxplugin : using cqxinliradius002
Tue Apr 14 19:53:28 2015 daemon.info pppd[2657]: -------------------------------------
Tue Apr 14 19:53:28 2015 daemon.info pppd[2657]: timenow(Hex)=552cffb8
Tue Apr 14 19:53:28 2015 daemon.info pppd[2657]: Begin : beforeMD5
Tue Apr 14 19:53:28 2015 daemon.info pppd[2657]: 1.<??>
Tue Apr 14 19:53:28 2015 daemon.info pppd[2657]: 2.<??1336******6>
Tue Apr 14 19:53:28 2015 daemon.info pppd[2657]: 3.<??1336******6cqxinliradius002>
Tue Apr 14 19:53:28 2015 daemon.info pppd[2657]: 4.length=<31>
Tue Apr 14 19:53:28 2015 daemon.info pppd[2657]: End : beforeMD5
Tue Apr 14 19:53:28 2015 daemon.info pppd[2657]: Begin : afterMD5
Tue Apr 14 19:53:28 2015 daemon.info pppd[2657]: 1.MD5use_1=< 8>
Tue Apr 14 19:53:28 2015 daemon.info pppd[2657]: 2.MD5use_2=< 4>
Tue Apr 14 19:53:28 2015 daemon.info pppd[2657]: End : afterMD5
Tue Apr 14 19:53:28 2015 daemon.info pppd[2657]: -------------------------------------
Tue Apr 14 19:53:28 2015 daemon.info pppd[2657]: sxplugin : user  is <
]X!Q> 841336******6> 
Tue Apr 14 19:53:28 2015 daemon.info pppd[2657]: sxplugin : passwd loaded
Tue Apr 14 19:53:28 2015 daemon.notice pppd[2657]: pppd 2.4.7 started by root, uid 0
这是我换了一个路由器之后最新的日志，一直都是这一段的循环。“*****”是我的手机号，我给隐去了。完全找不到原因。。。真是要哭了( ╯□╰ )

Jerry6666, 2015-04-14:
我还是附一下截图，看着直观一些。
[image]
[image]

Jerry6666, 2015-04-14:
漏了一张，zhe'zhang这张应该是在前面的
[image]

miao1007, 2015-04-14:
Timeout waiting for PADO packets

你的pppoe没有拨出去，是内网没有配好

Jerry6666, 2015-04-14:
内网没配好吗？我应该怎么做,,ԾㅂԾ,,

Jerry6666, 2015-04-14:
我自己再试试吧。真是太麻烦你了。我都不好意思来问你了。。。

closed 2015-04-20 by miao1007