miao1007/Openwrt-NetKeeper #4: radius值要在什么地方才能查看到？

wangyangdahai, 2015-04-20:
我们这用的是湖北版的netkeeper，求解答

miao1007, 2015-04-20:
把win,mac,android版的发我，我反编译试试

wangyangdahai, 2015-04-20:
这个是我们学校用的pan.baidu.com/s/1gd4EAAZ

syseric, 2015-04-20:
同湖北，一样求下。另外求下wndr4300 gcc地址，实在是找不到下哪一个。Openwrt官网里面的分类是在ar71xx/nand下 能不能用那个GCC

wangyangdahai, 2015-04-20:
@syseric  openwrt的SDK里应该有吧？话说你是在电脑上用，还是路由器里用。

miao1007, 2015-04-20:
win,mac版用UPX加壳了，看来要慢慢研究了，Android版是用http-post登录的，理论上你用路由器curl登录也可以成功的。

syseric, 2015-04-20:
@wangyangdahai  路由器啊！

syseric, 2015-04-20:
@miao1007  路由器curl登录如何实现啊？

miao1007, 2015-04-20:
反编译Android版....自己搞吧，也要好久的

syseric, 2015-04-20:
@miao1007 估计没那能力，大神 我这有老版的E信for win，现在还能使用的，你看需不需要！

miao1007, 2015-04-20:
加壳的无力也无时间研究了....

TeraDew, 2015-04-24:
湖北e信安卓无线版：http://219.140.78.12/wuhanyixin1.0.5.apk，希望能有帮助

miao1007, 2015-04-24:
我下过了，它是HTTP登录的，没有找到radius

Truelovefortony, 2015-04-27:
武汉地区的 旧的2011版E信屏蔽升级后也可以使用
和早期的闪讯是vmp的壳 文件结构和文件夹和Netkeeper几乎一致。
这是2011版的文件夹打包http://pan.baidu.com/s/1jGvfwUi 苦于找不到radius
望帮忙寻找或提供一下思路（我用的是OD）
好编译到目前主流的MT7620中

miao1007, 2015-04-27:
使用od附加到netkeeper上，设置断点rasdialA，运行到返回处，断点，再次拨号，就能看到radius值了。

syseric, 2015-04-27:
@Truelovefortony 我是没能力搞了，兄弟你搞出来了 希望能分享下

miao1007, 2015-04-28:
搞了好久终于搞出来了（不愧是外包产品），Radius是"hubtxinli01"，各位可以测试了，把我的c文件中重庆的radius换成武汉的即可，反馈问题请把Log贴出。

syseric, 2015-04-28:
@miao1007 感谢分享，马上去试试！

Truelovefortony, 2015-04-28:
@miao1007 感谢分享

syseric, 2015-04-29:
@miao1007  大神 为什么我make的时候就出错啊 make: *** [sxplugin.so] Error 126

miao1007, 2015-04-29:
贴出全部啊...

syseric, 2015-04-29:
@miao1007  电脑太卡 虚拟机运行不起来 就用了个云主机，在putty 里面进行的
/root/OpenWrt-Toolchain-ar71xx-nand_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/bin/mips-openwrt-linux-gcc -Os -Wall sxplugin.c -fPIC -I/root/OpenWrt-Toolchain-ar71xx-nand_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include -shared -o sxplugin.so
/root/OpenWrt-Toolchain-ar71xx-nand_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/bin/mips-openwrt-linux-gcc: line 82: /root/OpenWrt-Toolchain-ar71xx-nand_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/bin/mips-openwrt-linux-uclibc-gcc.bin: cannot execute binary file
/root/OpenWrt-Toolchain-ar71xx-nand_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/bin/mips-openwrt-linux-gcc: line 82: /root/OpenWrt-Toolchain-ar71xx-nand_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/bin/mips-openwrt-linux-uclibc-gcc.bin: Success
make: *** [sxplugin.so] Error 126

syseric, 2015-04-29:
@miao1007 换虚拟机后报错
my@my-virtual-machine:~/src$ make
/home/my/tool/bin/mips-openwrt-linux-gcc -Os -Wall sxplugin.c -fPIC -I/home/my/tool/include -shared -o sxplugin.so
mips-openwrt-linux-uclibc-gcc.bin: warning: environment variable 'STAGING_DIR' not defined
In file included from sxplugin.c:22:0:
pppd.h:58:24: fatal error: patchlevel.h: No such file or directory
 #include "patchlevel.h"
 compilation terminated.
make: *** [sxplugin.so] 错误 1
虚拟机实在卡得不行 ，大神能不能帮忙编译下wndr4300的 湖北用的so库

Truelovefortony, 2015-04-29:
@syseric 第一个问题可能是因为编译器是64位运行在了32位系统上 所以报错cannot execute binary file

第二个问题是GCC环境没有搭建好，环境变量未定义environment variable 'STAGING_DIR' not defined

我现在也卡在这里
pppd.h:58:24: fatal error: patchlevel.h: No such file or directory
#include "patchlevel.h"

syseric, 2015-04-30:
@Truelovefortony 上传到路由器中是否成功？能否留个联系方式就E信交流下？

syseric, 2015-04-30:
@miao1007 Thu Apr 30 18:38:48 2015 daemon.info pppd[1126]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Thu Apr 30 18:38:48 2015 daemon.info pppd[1126]: Plugin sxplugin.so loaded.
Thu Apr 30 18:38:48 2015 daemon.info pppd[1126]: sxplugin : init
Thu Apr 30 18:38:48 2015 daemon.info pppd[1126]: sxplugin : using hubtxinli01
Thu Apr 30 18:38:48 2015 daemon.info pppd[1126]: -------------------------------------
Thu Apr 30 18:38:48 2015 daemon.info pppd[1126]: timenow(Hex)=55420638
Thu Apr 30 18:38:48 2015 daemon.info pppd[1126]: Begin : beforeMD5
Thu Apr 30 18:38:48 2015 daemon.info pppd[1126]: 1.<
4q>
Thu Apr 30 18:38:48 2015 daemon.info pppd[1126]: 2.<
4q18062054769>
Thu Apr 30 18:38:48 2015 daemon.info pppd[1126]: 3.<
4q18062054769hubtxinli01>
Thu Apr 30 18:38:48 2015 daemon.info pppd[1126]: 4.length=<26>
Thu Apr 30 18:38:48 2015 daemon.info pppd[1126]: End : beforeMD5
Thu Apr 30 18:38:48 2015 daemon.info pppd[1126]: Begin : afterMD5
Thu Apr 30 18:38:48 2015 daemon.info pppd[1126]: 1.MD5use_1=< 3>
Thu Apr 30 18:38:48 2015 daemon.info pppd[1126]: 2.MD5use_2=< 3>
Thu Apr 30 18:38:48 2015 daemon.info pppd[1126]: End : afterMD5
Thu Apr 30 18:38:48 2015 daemon.info pppd[1126]: -------------------------------------
Thu Apr 30 18:38:48 2015 daemon.info pppd[1126]: sxplugin : user  is <
WV!I" 3318062054769> 
Thu Apr 30 18:38:48 2015 daemon.info pppd[1126]: sxplugin : passwd loaded
Thu Apr 30 18:38:48 2015 daemon.notice pppd[1126]: pppd 2.4.7 started by root, uid 0
Thu Apr 30 18:38:49 2015 kern.info kernel: [   21.370000] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
Thu Apr 30 18:38:49 2015 kern.info kernel: [   21.380000] device wlan1 entered promiscuous mode
Thu Apr 30 18:38:49 2015 kern.info kernel: [   21.380000] br-lan: port 3(wlan1) entered forwarding state
Thu Apr 30 18:38:49 2015 kern.info kernel: [   21.390000] br-lan: port 3(wlan1) entered forwarding state
Thu Apr 30 18:38:49 2015 kern.info kernel: [   21.830000] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
Thu Apr 30 18:38:49 2015 daemon.notice netifd: Network device 'wlan1' link is up
Thu Apr 30 18:38:49 2015 kern.info kernel: [   22.110000] br-lan: port 1(eth0.1) entered forwarding state
Thu Apr 30 18:38:51 2015 kern.info kernel: [   23.390000] br-lan: port 3(wlan1) entered forwarding state
Thu Apr 30 18:38:52 2015 daemon.info dnsmasq[1186]: started, version 2.71 cachesize 150
Thu Apr 30 18:38:52 2015 daemon.info dnsmasq[1186]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-auth no-DNSSEC
Thu Apr 30 18:38:52 2015 daemon.info dnsmasq-dhcp[1186]: DHCP, IP range 192.168.1.100 -- 192.168.1.249, lease time 12h
Thu Apr 30 18:38:52 2015 daemon.info dnsmasq[1186]: using local addresses only for domain lan
Thu Apr 30 18:38:52 2015 daemon.warn dnsmasq[1186]: no servers found in /tmp/resolv.conf.auto, will retry
Thu Apr 30 18:38:52 2015 daemon.info dnsmasq[1186]: read /etc/hosts - 1 addresses
Thu Apr 30 18:38:52 2015 daemon.info dnsmasq[1186]: read /tmp/hosts/dhcp - 1 addresses
Thu Apr 30 18:38:52 2015 daemon.info dnsmasq-dhcp[1186]: read /etc/ethers - 0 addresses
Thu Apr 30 18:38:55 2015 kern.info kernel: [   27.800000] br-lan: port 2(wlan0) entered forwarding state
Thu Apr 30 18:38:55 2015 kern.info kernel: [   27.800000] br-lan: port 2(wlan0) entered forwarding state
Thu Apr 30 18:38:55 2015 kern.info kernel: [   27.810000] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Thu Apr 30 18:38:55 2015 daemon.notice netifd: Network device 'wlan0' link is up
Thu Apr 30 18:38:57 2015 kern.info kernel: [   29.800000] br-lan: port 2(wlan0) entered forwarding state
Thu Apr 30 18:39:04 2015 daemon.warn pppd[1126]: Timeout waiting for PADO packets
Thu Apr 30 18:39:04 2015 daemon.err pppd[1126]: Unable to complete PPPoE Discovery
Thu Apr 30 18:39:04 2015 daemon.info pppd[1126]: Exit.
Thu Apr 30 18:39:04 2015 daemon.notice netifd: Interface 'NetKeeper' is now down
Thu Apr 30 18:39:04 2015 daemon.notice netifd: Interface 'NetKeeper' is disabled
Thu Apr 30 18:39:04 2015 daemon.notice netifd: Interface 'NetKeeper' is enabled
Thu Apr 30 18:39:04 2015 daemon.notice netifd: Interface 'NetKeeper' is setting up now
Thu Apr 30 18:39:04 2015 daemon.notice netifd: VLAN 'eth0.2' link is down
Thu Apr 30 18:39:04 2015 daemon.notice netifd: Interface 'NetKeeper' has link connectivity loss
Thu Apr 30 18:39:04 2015 daemon.notice netifd: VLAN 'eth0.2' link is up
Thu Apr 30 18:39:04 2015 daemon.notice netifd: Interface 'NetKeeper' has link connectivity 
Thu Apr 30 18:39:04 2015 daemon.info pppd[1289]: Plugin rp-pppoe.so loaded.
Thu Apr 30 18:39:04 2015 daemon.info pppd[1289]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Thu Apr 30 18:39:04 2015 daemon.info pppd[1289]: Plugin sxplugin.so loaded.
Thu Apr 30 18:39:04 2015 daemon.info pppd[1289]: sxplugin : init
Thu Apr 30 18:39:04 2015 daemon.info pppd[1289]: sxplugin : using hubtxinli01
Thu Apr 30 18:39:04 2015 daemon.info pppd[1289]: -------------------------------------
Thu Apr 30 18:39:04 2015 daemon.info pppd[1289]: timenow(Hex)=55420648
Thu Apr 30 18:39:04 2015 daemon.info pppd[1289]: Begin : beforeMD5
Thu Apr 30 18:39:04 2015 daemon.info pppd[1289]: 1.<
4t>
Thu Apr 30 18:39:04 2015 daemon.info pppd[1289]: 2.<
4t18062054769>
Thu Apr 30 18:39:04 2015 daemon.info pppd[1289]: 3.<
4t18062054769hubtxinli01>
Thu Apr 30 18:39:04 2015 daemon.info pppd[1289]: 4.length=<26>
[... cut ...]
Thu Apr 30 18:39:19 2015 daemon.notice netifd: Interface 'NetKeeper' is setting up now
Thu Apr 30 18:39:19 2015 daemon.notice netifd: VLAN 'eth0.2' link is down
Thu Apr 30 18:39:19 2015 daemon.notice netifd: Interface 'NetKeeper' has link connectivity loss
Thu Apr 30 18:39:19 2015 daemon.notice netifd: VLAN 'eth0.2' link is up
Thu Apr 30 18:39:19 2015 daemon.notice netifd: Interface 'NetKeeper' has link connectivity 
连接成功后无法上网，我通过电信接口接到电脑IP为100.xx.xx.xx，初步怀疑是内网问题，请问内网如何配置，另外我通过电脑连接成功IP为59.xx.xx.xx，但通过路由拨号netkeeper接口显示IP为27.18.49.109，两者都为武汉电信。

Truelovefortony, 2015-04-30:
@syseric 外网是DHCP+PPPOE混合组网的，直接接外网DHCP获取的100.xx.xx.xx内网地址是用来给Android、IOS移动客户端HTTP POST认证之后给这个IP开通外网权限，电脑获取这个IP没什么用，差不多就是给寝室设备互联（如文件夹共享、局域网对战等） 实际上上网还是经过PPPOE拨号获取到公网IP的。路由拨号netkeeper接口显示IP为27.18.X.X，恭喜，插件已经工作并且认证成功了，路由器已经具备访问互联网权限的，你检查电脑LAN口和网关是不是路由的IP，建议在路由认证后测试一下网络情况。

syseric, 2015-04-30:
@Truelovefortony 你通过电脑拨号成功后IP是27.18.xx.xx吗？我用电脑拨号成功IP是58.49.xx.xx 这个才是应该获取的IP吧  你现在已经成功了是吧？

miao1007, 2015-05-01:
恭喜，插件已经工作并且认证成功！剩下的一般就是配置路由表，traceroute测试了

syseric, 2015-05-01:
@miao1007 确实 昨天测试过路由器能上网，但LAN口接到电脑不能上网，并且pppoe拨号那里账号密码经常会变成root账号密码。

Truelovefortony, 2015-05-01:
路由器能上网后，电脑接LAN口，tracert一下看看经过了路由器转发，如果有误，看看网关IP，配置一下路由表就可以。“pppoe拨号那里账号密码经常会变成root账号密码”是因为你启用了浏览器的自动填表功能，浏览器把那里当成登录页面所以就给你自动填了路由器的root账号密码

syseric, 2015-05-01:
@miao1007 @Truelovefortony  感谢两位，改掉防火墙配置就可以上网了！！

lh123, 2015-05-02:
求把这个加密方法改写为java版的，
这样我好把加密方法写成安卓程序

lh123, 2015-05-02:
就是修改https://github.com/sunflyer/Simple-Netkeeper-Android
他中的用户名加密算法

closed 2015-05-20 by miao1007