miao1007/Openwrt-NetKeeper #6: 求救啊，ubuntu里编译错误

wsy741963, 2015-04-23:
下载的是OpenWrt-Toolchain-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64.tar
ar71xx的wr841n v3
~/桌面/gcc4.8/bin/mips-openwrt-linux-gcc -Os -Wall sxplugin.c -fPIC -I~/桌面/gcc4.8/include -shared -o sxplugin.so
/home/miracle/桌面/gcc4.8/bin/mips-openwrt-linux-gcc: 行 82: /home/miracle/桌面/gcc4.8/bin/mips-openwrt-linux-uclibc-gcc.bin: cannot execute binary file: 可执行文件格式错误
/home/miracle/桌面/gcc4.8/bin/mips-openwrt-linux-gcc: 行 82: /home/miracle/桌面/gcc4.8/bin/mips-openwrt-linux-uclibc-gcc.bin: 成功
make: *** [sxplugin.so] 错误 126
我们学校是@cqmu，代码不用改么

miao1007, 2015-04-23:
```
/home/miracle/桌面/gcc4.8/bin/mips-openwrt-linux-uclibc-gcc.bin
```

改为

```
/home/miracle/桌面/gcc4.8/bin/mips-openwrt-linux-uclibc-gcc
```

wsy741963, 2015-04-23:
还是不对，这是我改的make文件，用gcc应该不用配置环境吧
#Download Lastest GCC in http://downloads.openwrt.org/snapshots/trunk/
#This is a demo for Netgear WNDR3800(AR71XX)

#TODO : Change the location for your GCC’s location
CC=~/桌面/gcc4.8/mips-openwrt-linux-uclibc/bin/mips-openwrt-linux-gcc
CFLAGS=-Os -Wall

all:sxplugin.so

sxplugin.so:
	$(CC) $(CFLAGS) sxplugin.c -fPIC -I~/桌面/gcc4.8/include -shared -o sxplugin.so

miao1007, 2015-04-23:
#直接网上搜索gcc的名字吧，应该是一样的
#Download Lastest GCC in http://downloads.openwrt.org/snapshots/trunk/
#This is a demo for Netgear WNDR3800(AR71XX)

#TODO : Change the location for your GCC’s location
CC=/home/leon/netkeeper/OpenWrt-Toolchain-ar71xx-for-mips_34kc-gcc-4.8-linaro_uClibc-0.9.33.2/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/bin/mips-openwrt-linux-gcc
CFLAGS=-Os -Wall

all:sxplugin.so

sxplugin.so:
	$(CC) $(CFLAGS) sxplugin.c -fPIC -I/home/leon/netkeeper/OpenWrt-Toolchain-ar71xx-for-mips_34kc-gcc-4.8-linaro_uClibc-0.9.33.2/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include -shared -o sxplugin.so

wsy741963, 2015-04-26:
解决了很多问题，试了好几个toolchain后终于把动态库编出来了，为此我还卸了虚拟机装了双系统，最后用动态库拨号拨上了，但是显示拨上了，其实上不了网。。。。郁闷了。。。这是怎么回事

NETKEEPER
 (eth0.2)
pppoe-NetKeeper
运行时间: 0h 12m 9s
接收: 36.26 KB (266 数据包)
发送: 24.01 KB (294 数据包)
IPv4: 125.82.58.85/32

日志：
Sun Apr 26 17:08:01 2015 daemon.notice netifd: Interface 'NetKeeper' is now down
Sun Apr 26 17:08:01 2015 daemon.info avahi-daemon[2269]: Withdrawing address record for fe80::6609:80ff:fe58:9ee4 on eth0.2.
Sun Apr 26 17:08:01 2015 daemon.notice netifd: Interface 'NetKeeper' is disabled
Sun Apr 26 17:08:01 2015 daemon.notice netifd: Interface 'NetKeeper' is enabled
Sun Apr 26 17:08:01 2015 daemon.notice netifd: Interface 'NetKeeper' is setting up now
Sun Apr 26 17:08:01 2015 user.notice root: stopping ntpclient
Sun Apr 26 17:08:01 2015 daemon.info pppd[4674]: Plugin rp-pppoe.so loaded.
Sun Apr 26 17:08:01 2015 daemon.info pppd[4674]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Sun Apr 26 17:08:01 2015 daemon.info pppd[4674]: Plugin sxplugin.so loaded.
Sun Apr 26 17:08:01 2015 daemon.info pppd[4674]: sxplugin : init
Sun Apr 26 17:08:01 2015 daemon.info pppd[4674]: sxplugin : using cqxinliradius002
Sun Apr 26 17:08:01 2015 daemon.info pppd[4674]: -------------------------------------
Sun Apr 26 17:08:01 2015 daemon.info pppd[4674]: timenow(Hex)=553caaf1
Sun Apr 26 17:08:01 2015 daemon.info pppd[4674]: Begin : beforeMD5
Sun Apr 26 17:08:01 2015 daemon.info pppd[4674]: 1.<"0>
Sun Apr 26 17:08:01 2015 daemon.info pppd[4674]: 2.<"018983420228>
Sun Apr 26 17:08:01 2015 daemon.info pppd[4674]: 3.<"018983420228cqxinliradius002>
Sun Apr 26 17:08:01 2015 daemon.info pppd[4674]: 4.length=<31>
Sun Apr 26 17:08:01 2015 daemon.info pppd[4674]: End : beforeMD5
Sun Apr 26 17:08:01 2015 daemon.info pppd[4674]: Begin : afterMD5
Sun Apr 26 17:08:01 2015 daemon.info pppd[4674]: 1.MD5use_1=< 4>
Sun Apr 26 17:08:01 2015 daemon.info pppd[4674]: 2.MD5use_2=< b>
Sun Apr 26 17:08:01 2015 daemon.info pppd[4674]: End : afterMD5
Sun Apr 26 17:08:01 2015 daemon.info pppd[4674]: -------------------------------------
Sun Apr 26 17:08:01 2015 daemon.info pppd[4674]: sxplugin : user  is <
0X (" 4b18983420228> 
Sun Apr 26 17:08:01 2015 daemon.info pppd[4674]: sxplugin : passwd loaded
Sun Apr 26 17:08:01 2015 daemon.notice pppd[4674]: pppd 2.4.7 started by root, uid 0
Sun Apr 26 17:08:01 2015 daemon.info pppd[4674]: PPP session is 20281
Sun Apr 26 17:08:01 2015 daemon.warn pppd[4674]: Connected to 00:18:82:cf:f8:fa via interface eth0.2
Sun Apr 26 17:08:01 2015 daemon.info pppd[4674]: Using interface pppoe-NetKeeper
Sun Apr 26 17:08:01 2015 daemon.notice pppd[4674]: Connect: pppoe-NetKeeper <--> eth0.2
Sun Apr 26 17:08:01 2015 daemon.info pppd[4674]: syncppp not active
Sun Apr 26 17:08:02 2015 daemon.info pppd[4674]: Remote message: 2001
Sun Apr 26 17:08:02 2015 daemon.notice pppd[4674]: PAP authentication succeeded
Sun Apr 26 17:08:02 2015 daemon.notice pppd[4674]: peer from calling number 00:18:82:CF:F8:FA authorized
Sun Apr 26 17:08:02 2015 daemon.notice pppd[4674]: local  IP address 125.82.58.85
Sun Apr 26 17:08:02 2015 daemon.notice pppd[4674]: remote IP address 125.82.56.1
Sun Apr 26 17:08:02 2015 daemon.notice pppd[4674]: primary   DNS address 61.128.128.68
Sun Apr 26 17:08:02 2015 daemon.notice pppd[4674]: secondary DNS address 61.128.192.68
Sun Apr 26 17:08:02 2015 daemon.warn dnsmasq[3734]: no servers found in /tmp/resolv.conf.auto, will retry
Sun Apr 26 17:08:02 2015 daemon.notice netifd: Network device 'pppoe-NetKeeper' link is up
Sun Apr 26 17:08:02 2015 daemon.notice netifd: Interface 'NetKeeper' is now up

wsy741963, 2015-04-26:
还有你教程里那个ssh上传地址写错了，应该是  root@192.168.1.1:/usr/lib/pppd/2.4.7
编出来居然还是不行，是不是学校有限制啊，我反复检查了代码，觉得没有问题啊，为了编这个我还搭载了gcc环境。。。

wsy741963, 2015-04-26:
这个是安卓版的，测试可用，是一个重邮的大神编的
http://pan.baidu.com/s/1i35qLUl

miao1007, 2015-04-26:
就差一步了，这个最后一步就非常折腾了，一般是学校内网配置什么的，注意尽量用官方包，不要用pandoraBox

wsy741963, 2015-04-26:
我去。。。大神你看出来我用的潘多拉？确实是这样，我改了很多设置都不行，但是他就是可以拨上号，用户名的加密也完全没有问题，时间函数计算正确，我很郁闷

TeraDew, 2015-04-28:
#我的也是编译问题，配置如下：
#Download Lastest GCC in http://downloads.openwrt.org/snapshots/trunk/
#This is a demo for Netgear WNDR3800(AR71XX)

#TODO : Change the location for your GCC鈥檚 location
CC=/home/yujay/Desktop/OpenWrt-Toolchain-ramips-rt305x_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_uClibc-0.9.33.2/bin/mipsel-openwrt-linux-uclibc-gcc
CFLAGS=-Os -Wall

all:sxplugin.so

sxplugin.so:	
$(CC) $(CFLAGS) sxplugin.c -fPIC -I/home/yujay/Desktop/OpenWrt-Toolchain-ramips-rt305x_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_uClibc-0.9.33.2/include -shared -o sxplugin.so
#
#报错如下：
yujay@ubuntu:~/Desktop/Openwrt-NetKeeper-master/src$ make
makefile:11: *** missing separator (did you mean TAB instead of 8 spaces?).  Stop.
#求帮助

TeraDew, 2015-04-28:
我在CSDN查的 MakeFile 每个动作都要加一个<TAB>,http://blog.csdn.net/zhuichao001/article/details/5872604 但是加了tab，include 文件夹就灰了

miao1007, 2015-04-28:
换编辑器吧，我用的是Ubuntu默认的记事本

TeraDew, 2015-04-28:
我是用 ubuntu GUI 的 gedit 编辑的，不可以吗

wsy741963, 2015-05-01:
第二天已经解决，已经用了好几天了，达到满速8m，不掉线。不过代码里有点小问题，pppd.h中有一段依赖#include "patchlevel.h"，这段没用是版本信息，有这个会出错中断。。。唉，弄了好久才把路由调上，多谢你的源码了，帮了大忙

wsy741963, 2015-05-02:
今天一看，居然心跳包都反编译出来了，太nb了，为你点个赞

TeraDew, 2015-05-02:
我和 @Truelovefortony 这位朋友开始遇到的问题一样，都是卡在这里："pppd.h:58:24: fatal error: patchlevel.h: No such file or directory" 不过我从SDK目录里 find 到了这个头文件，然后直接丢到 /src 目录里，make，一路warning，还是编译出来了。惭愧这么简单的问题当时竟没有看明白。

closed 2015-05-20 by miao1007