miao1007/Openwrt-NetKeeper #7: 求助啊！拨号不成功怎么帮

wx1183618058, 2015-05-03:
Mon May  4 00:56:52 2015 daemon.warn pppd[3628]: Timeout waiting for PADS packets
Mon May  4 00:56:52 2015 daemon.err pppd[3628]: Unable to complete PPPoE Discovery
Mon May  4 00:56:52 2015 daemon.info pppd[3628]: Exit.
Mon May  4 00:56:52 2015 daemon.notice netifd: Interface 'wan' is now down
Mon May  4 00:56:52 2015 daemon.info avahi-daemon[2197]: Withdrawing address record for fe80::2276:93ff:fe25:d894 on eth0.2.
Mon May  4 00:56:52 2015 daemon.notice netifd: Interface 'wan' is disabled
Mon May  4 00:56:52 2015 daemon.notice netifd: Interface 'wan' is enabled
Mon May  4 00:56:52 2015 daemon.notice netifd: Interface 'wan' is setting up now
Mon May  4 00:56:52 2015 daemon.notice netifd: wan (8591): uci: Entry not found
Mon May  4 00:56:52 2015 daemon.notice netifd: wan (8591): uci: Entry not found
Mon May  4 00:56:53 2015 daemon.info pppd[8611]: Plugin rp-pppoe.so loaded.
Mon May  4 00:56:53 2015 daemon.info pppd[8611]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Mon May  4 00:56:53 2015 daemon.info pppd[8611]: Plugin sxplugin.so loaded.
Mon May  4 00:56:53 2015 daemon.info pppd[8611]: sxplugin : init
Mon May  4 00:56:53 2015 daemon.info pppd[8611]: -------------------------------------
Mon May  4 00:56:53 2015 daemon.info pppd[8611]: timenow(Hex)=55465355
Mon May  4 00:56:53 2015 daemon.info pppd[8611]: Begin : beforeMD5
Mon May  4 00:56:53 2015 daemon.info pppd[8611]: 1.<?>
Mon May  4 00:56:53 2015 daemon.info pppd[8611]: 2.<?18084092094>
Mon May  4 00:56:53 2015 daemon.info pppd[8611]: 3.<?18084092094cqxinliradius002?PwH??T?w>
Mon May  4 00:56:53 2015 daemon.info pppd[8611]: 4.length=<47>
Mon May  4 00:56:53 2015 daemon.info pppd[8611]: End : beforeMD5
Mon May  4 00:56:53 2015 daemon.info pppd[8611]: Begin : afterMD5
Mon May  4 00:56:53 2015 daemon.info pppd[8611]: 1.MD5use_1=< b>
Mon May  4 00:56:53 2015 daemon.info pppd[8611]: 2.MD5use_2=< 0>
Mon May  4 00:56:53 2015 daemon.info pppd[8611]: End : afterMD5
Mon May  4 00:56:53 2015 daemon.info pppd[8611]: -------------------------------------
Mon May  4 00:56:53 2015 daemon.info pppd[8611]: sxplugin : user  is <
$]A(S b018084092094@bsjdxy> 
Mon May  4 00:56:53 2015 daemon.info pppd[8611]: sxplugin : passwd loaded
Mon May  4 00:56:53 2015 daemon.notice pppd[8611]: pppd 2.4.7 started by root, uid 0
Mon May  4 00:56:54 2015 daemon.info avahi-daemon[2197]: Registering new address record for fe80::2276:93ff:fe25:d894 on eth0.2.*.

ousun, 2015-11-20:
你给的信息好像不完整，你试着手动拨号一次，发出完整一点的消息。

miao1007, 2015-11-20:
应该是静态路由没有配置好