miao1007/Openwrt-NetKeeper #8: 重庆Netkeeper 0094版本

sunflyer, 2015-05-19:
Radius似乎更新了，xianxinli1radius，但是我没有看算法是否做出了修改。心跳服务器貌似转移到了 gxyw.online.cq.cn，报文内容结构没太多变化，加密的KEY没变。现在看样子应该是两个都在使用还是怎么。而且不知道重庆电信计费系统是怎么设置的，我们学校新校区的号在老校区不断网，但是新校区会准时断。

sunflyer, 2015-05-19:
现在OD加上StrongOD也会被Nk查到。。。。本人逆向小白，所以找不到更多的了。也许让我老老实实写写代码或许好得多：）

miao1007, 2015-05-19:
算法应该不会修改的，毕竟是外包产品，服务器与客户端同时升级的成本，比少数人使用路由器的损失多得多了。听说电信买这个软件花了百万的价钱，O(∩_∩)O~ 。我逆向主要是针对Mac与Android版，Windows我也不太擅长。

sunflyer, 2015-05-19:
按我个人推测来说的话服务器端应该是有一个Radius列表来控制账号计算的PIN码校对。我以前还觉得如果电信同时计算两个和以上radius的话对服务器负载是不是有点大，现在看来我被打脸了。按照过去的推测PIN码有效期前后10M左右应该都是OK的。如果电信只是改RADIUS的话那么问题就比较简单了，但是这么一来的话因为RADIUS改变所以对应PIN码列表也会变更，可能电信可以借此查出版本然后强制心跳也不好说。心跳服务器只是加一台的问题，毕竟这东西需要的资源也就摆在那里。
另外我逆向了下安卓的APP，心跳部分可能还是有问题，电脑客户端应该是AES之类的，但是很奇怪不知道是我抓的包有问题还是代码有问题，AES和3DES始终是没法解密心跳包的数据。KEY应该是没问题的。比较奇怪的是上次模拟心跳发送电信也没有返回UDP包，看了94版本找到的东西估计是只对强制新版本的学校开心跳验证了吧。
不过电信这个时候换radius也真是吓了我一跳。。。。

sunflyer, 2015-05-19:
现在的情况是能找到心跳的关键call，但是看不懂汇编，也不知道是怎么走的，94版查逆向似乎有手段了一样。。。也许我得多试试：）

sunflyer, 2015-05-20:
已经确定只是修改了radius。

miao1007, 2015-05-20:
我的学校貌似还能用呢，那么这个新的radisu，我这个op项目，还有你那个Android版个APP也需要更新了吗？

sunflyer, 2015-05-20:
现在是两个radius都能用。不管是之前的cqxinliradius002还是现在的xianxinli1radius，现在是并存的，等哪天强制更新94版本的时候估计就是必须更新的时候了

miao1007, 2015-05-20:
好吧，那我也把它加上去了

sunflyer, 2015-05-20:
：） 心跳包依然没进展，有一个字段我始终找不到怎么算的。醉了醉了。。。。。

closed 2015-05-20 by sunflyer