miao1007/Openwrt-NetKeeper #11: 大婶我研究了很久还是编译不出来

doubidan666, 2015-06-10:
太小白了 没办法 你能不能把你的newifi的so传给我？

Truelovefortony, 2015-06-10:
留个邮箱发你

doubidan666, 2015-06-10:
2434712862@qq.com 谢谢

Truelovefortony, 2015-06-10:
已发

doubidan666, 2015-06-10:
请问路由器刷的什么固件？

doubidan666, 2015-06-10:
Wed Jun 10 18:38:13 2015 daemon.notice netifd: Interface 'wan' is setting up now
Wed Jun 10 18:38:13 2015 daemon.info pppd[7906]: Plugin rp-pppoe.so loaded.
Wed Jun 10 18:38:13 2015 daemon.info pppd[7906]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Wed Jun 10 18:38:13 2015 daemon.notice netifd: wan (7906): /usr/sbin/pppd: File not found
Wed Jun 10 18:38:13 2015 daemon.err pppd[7906]: File not found
Wed Jun 10 18:38:13 2015 daemon.notice netifd: wan (7906): /usr/sbin/pppd: Couldn't load plugin RasDialUp.so
Wed Jun 10 18:38:13 2015 daemon.err pppd[7906]: Couldn't load plugin RasDialUp.so
Wed Jun 10 18:38:14 2015 daemon.notice netifd: Interface 'wan' is now down
无法加载so。。。。

doubidan666, 2015-06-10:
貌似你给我的是e信的，我现在在系统日志里看到能算出账号但是拨不上，应该是闪讯和e信算法不一样？

doubidan666, 2015-06-11:
pppd.h:58:24: fatal error: patchlevel.h: No such file or directory
 #include "patchlevel.h"
求解

doubidan666, 2015-06-14:
编译出现pppd.h:58:24: fatal error: patchlevel.h: No such file or directory
#include "patchlevel.h"
解决方法，去https://github.com/squadette/pppd/tree/for-upstream/pppd把patchlevel.h下载到src目录

closed 2015-06-14 by doubidan666