miao1007/Openwrt-NetKeeper #12: 请问小米路由器能用这个么？

lycycycycyc, 2015-06-22:
我最近买了个小米路由器mini，请问能用么

lycycycycyc, 2015-06-22:
http://downloads.openwrt.org/snapshots/trunk/ 里面没有MT7620A怎么办

miao1007, 2015-06-27:
http://downloads.openwrt.org/snapshots/trunk/ramips/mt7620/

kirito269, 2015-06-27:
好像都不行吧……

lycycycycyc, 2015-06-27:
是需要刷成openwrt的系统么

miao1007, 2015-06-27:
不推荐原厂的系统，建议刷成官方系统

lycycycycyc, 2015-06-27:
http://downloads.openwrt.org.cn/PandoraBox/Xiaomi-Mini-R1CM/testing/
这里面的版本可以么

miao1007, 2015-06-27:
可以

lycycycycyc, 2015-06-27:
Sat Jun 27 15:19:00 2015 daemon.notice netifd: Interface 'NetKeeper' is setting up now
Sat Jun 27 15:19:01 2015 daemon.info pppd[20938]: Plugin rp-pppoe.so loaded.
Sat Jun 27 15:19:01 2015 daemon.info pppd[20938]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Sat Jun 27 15:19:01 2015 daemon.notice netifd: NetKeeper (20938): /usr/sbin/pppd: File not found
Sat Jun 27 15:19:01 2015 daemon.err pppd[20938]: File not found
Sat Jun 27 15:19:01 2015 daemon.notice netifd: NetKeeper (20938): /usr/sbin/pppd: Couldn't load plugin sxplugin.so
Sat Jun 27 15:19:01 2015 daemon.err pppd[20938]: Couldn't load plugin sxplugin.so
Sat Jun 27 15:19:01 2015 daemon.notice netifd: Interface 'NetKeeper' is now down

lycycycycyc, 2015-06-27:
这样是除了什么问题？

lycycycycyc, 2015-07-10:
有人小米路由mini成功的么?为什么我的怎么弄都不行

SuperBain, 2015-07-14:
我也是和你一样的问题，日志内容也差不多，请问你找到解决方法了吗？

lycycycycyc, 2015-07-14:
应该是编译时候出的问题，路由器还在我寝室，我放假回家了，就没再研究了

lycycycycyc, 2015-07-14:
编译时报错，environment variable "STAGING_DIR" not defined

miao1007, 2015-07-15:
湖北的有人编译好了 http://www.right.com.cn/forum/forum.php?mod=redirect&goto=findpost&ptid=141979&pid=1061829

lycycycycyc, 2015-09-19:
问题已经解决，当初是因为gcc找错了所以放进路由里面才会出错

closed 2015-09-19 by lycycycycyc