miao1007/Openwrt-NetKeeper #13: 提供一个南昌地区算法.希望能集成到当中

xseven007, 2015-06-27:
https://github.com/xseven007/Netkeeper
Win环境的源码在这里.之前尝试过用闪迅的源码进行修改来移植,后来懒得搞了.希望你有时间能搞进去

miao1007, 2015-06-27:
是“nanchang3.0”吗，源码好像很旧，不知道还能不能用

xseven007, 2015-06-27:
不是3.0,是2.5.这个算法是针对南昌各个高校的星空极速的.虽然源码旧,但是依然能用.

miao1007, 2015-06-27:
代码看了，是可以复用的，也就是说，我们只要替换了radius，就可以了

```
if (ver == 18)
	{
		RADIUS = _T("nanchang3.0");
		LR = _T("\r0");
	}
	else
	{
		RADIUS = _T("radius");
		LR = _T("\r\n");
	}
```

这两个具体版本可以提供一下吗

xseven007, 2015-06-27:
星空极速2.5v29?你是说那个程序的版本吗?

miao1007, 2015-06-27:
我刚刚把南昌的代码（"nanchang3.0"）也加进去了，算法应该不会错，因为那个pc版的是通过重邮版修改的。各位南昌的同学如果有时间就试一下吧。

xseven007, 2015-06-27:
PC版那个算法是2.5早期版本的dll文件里面弄出来的,南昌地区特别之处在于家庭宽带和高校宽带的拨号器版本不一样,家庭版多是3.0.之前抓包验证的时候也是花了不少时间

lknife, 2015-06-28:
http://dx.nc163.cn/xiaoyuan.html
南昌校园网的星空极速下载，貌似和你新加的版本不同

xseven007, 2015-06-29:
V29不是强制要求安装的,安装到V18就可以了..

miao1007, 2015-06-29:
那么多学校 😓 一个个解包太费时间了，有radius的发我吧，反正电信基本不会换算法了

realityone, 2015-07-13:
V29 也是`nanchang3.0`
但是因为南昌的 prefix 一直是`\r0`

xseven007, 2015-07-15:
反正南昌地区是通用的,因为从写开发文档到毕业2年后,我那个替代程序依然能在南昌多数高校正常使用..