miao1007/Openwrt-NetKeeper #14: 如何从原版exe客户端得到radius的值

jacywy, 2015-07-12:
找了好久没找到，有相关的文章吗，谢谢

miao1007, 2015-07-12:
脱壳，反编译

realityone, 2015-07-13:
对于大部分`netkeeper.exe`运行起来后用 LordPE 直接 dump 出来
在`TimeFlag`附近会找到一个加密的字符串，解出来就是对应的 shareKey
心跳密钥同理

jacywy, 2015-08-20:
以前我们这加密后的用户名只要不用客户端登录生成新用户名，这个用户名就一直能用，我就用你这个前面加回车换行符的功能，加密部分删掉了，用客户端生成的用户名，用了大半年了。结果今天不行了，也变成短时间有效了（大概一分钟），每次要用客户端的用户名，再迅速填到路由器里，好麻烦。
不懂脱壳反编译，希望能帮忙找出radius的值（新疆），非常感谢！
这是安装包：http://pan.baidu.com/s/1mg7t2Bm 密码: nu1w
这是安装后的：http://pan.baidu.com/s/1dDcz9IT 密码: t9hu

realityone, 2015-08-21:
shareKey = `qhtel@xiaoyuanyi`

jacywy, 2015-08-21:
感谢realityone，现在可以使用了，青海也许也是这个值，因为以前安装老版本exe的时候是标题显示青海的，从qh应该也能证实。
还有我们这貌似有些用户需要短信验证，就是登录后会收到条短信验证码，在客户端里输入，不输入3分钟后会自动下线。我有段时间也需要，后来又不需要了，不知道怎么回事。那时我就在路由器里用php写了个程序，模拟的短信请求和验证验证码。我把过程写出来，留给有需要的人，能整合进插件里更好了，输入验证码可以写个luci界面。
通信地址：61.128.123.116，端口8099，协议udp
向服务器请求发送短信给手机：<op><accountInfo login='手机号' checkCode='aa6b32a9d7df036b1fefd3a265bfb3ed' id='1418474298' /></op>
验证验证码：<op><validateAccount login='手机号' checkCode='aa6b32a9d7df036b1fefd3a265bfb3ed' validateCode=‘验证码’ id='1418471404' /></op>

miao1007, 2015-08-22:
非常感谢，如果你目前是可用的话，请pull request ，补上你的code <https://github.com/miao1007/Openwrt-NetKeeper/blob/master/src/sxplugin.c#L39>

jacywy, 2015-08-22:
看见已经添加青海支持了，我在新疆，可以用，只是根据客户端标题和qh推测青海也可以用，没测试过。

miao1007, 2015-08-23:
恩？你是新疆，用的青海的key？

jacywy, 2015-08-23:
恩，我这（SHZU）确实是qhtel@xiaoyuanyi