miao1007/Openwrt-NetKeeper #15: 无法加载sxplugin.so？

SuperBain, 2015-07-14:
首先非常感谢你的劳动成果，让我们有了拜托校园网客户端限制的曙光。
我安装你的教程编译了sxplugin.so，也给自己的小米路由器Mini刷上了最新的PandoraBox。
sxplugin.so也已经放在了指定位置。
不过我貌似还是不能使用路由器拨号。
请问这是为什么呢？
最后几行日志如下：

Tue Jul 14 22:35:44 2015 daemon.notice netifd: Interface 'NetKeeper' is setting up now
Tue Jul 14 22:35:44 2015 daemon.info pppd[3045]: Plugin rp-pppoe.so loaded.
Tue Jul 14 22:35:44 2015 daemon.info pppd[3045]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Tue Jul 14 22:35:44 2015 daemon.notice netifd: NetKeeper (3045): /usr/sbin/pppd: File not found
Tue Jul 14 22:35:44 2015 daemon.err pppd[3045]: File not found
Tue Jul 14 22:35:44 2015 daemon.notice netifd: NetKeeper (3045): /usr/sbin/pppd: Couldn't load plugin sxplugin.so
Tue Jul 14 22:35:44 2015 daemon.err pppd[3045]: Couldn't load plugin sxplugin.so
Tue Jul 14 22:35:44 2015 daemon.notice netifd: Interface 'NetKeeper' is now down

感激不尽！

miao1007, 2015-07-15:
把2.4.7改成2.4.8

zbczbczbc123456, 2015-07-15:
呃，请问你指的是路由器系统的pppd版本要升级到2.4.8，还是说要用2.4.8版本的pppd编译sxplugin.so呢？
而且这个的最新版本貌似就是2.4.7？
……

miao1007, 2015-07-15:
把源码的2.4.7改成2.4.8

zbczbczbc123456, 2015-07-16:
我试了一下这里某同学编译好的小米mini可用的E信的sxplugin.so，
http://www.right.com.cn/forum/forum.php?mod=redirect&goto=findpost&ptid=141979&pid=1061829
从路由器日志看，应该是可以正常加载sxplugin.so了(不过我是重庆，所以还是不能登陆)
所以我觉得应该是我编译的问题……
编译过程中报错的主要内容应该是mipsel-openwrt-linux-musl-gcc.bin: warning: environment variable 'STAGING_DIR' not defined
请问前辈知道这是什么原因吗？
非常感谢！

以下是完整的编译过程提示：
bain@bain-Parallels-Virtual-Platform:~/Openwrt-NetKeeper-master/src$ make
/home/bain/OpenWrt-Toolchain-ramips-mt7620_gcc-4.8-linaro_musl-1.1.10.Linux-x86_64/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_musl-1.1.10/bin/mipsel-openwrt-linux-gcc -Os -Wall sxplugin.c -fPIC -I/home/bain/OpenWrt-Toolchain-ramips-mt7620_gcc-4.8-linaro_musl-1.1.10.Linux-x86_64/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_musl-1.1.10/include -shared -o sxplugin.so
mipsel-openwrt-linux-musl-gcc.bin: warning: environment variable 'STAGING_DIR' not defined
sxplugin.c: In function 'getPIN':
sxplugin.c:41:5: warning: pointer targets in passing argument 1 of 'strcpy' differ in signedness [-Wpointer-sign]
     strcpy(RADIUS, "xianxinli1radius");
     ^
In file included from sxplugin.c:4:0:
/home/bain/OpenWrt-Toolchain-ramips-mt7620_gcc-4.8-linaro_musl-1.1.10.Linux-x86_64/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_musl-1.1.10/include/string.h:31:7: note: expected 'char * __restrict__' but argument is of type 'byte *'
 char *strcpy (char *__restrict, const char *__restrict);
       ^
sxplugin.c:53:5: warning: pointer targets in passing argument 1 of 'strlen' differ in signedness [-Wpointer-sign]
     byte* beforeMD5=malloc(strlen(timeByte)+strlen(userName)+strlen(RADIUS)+1);
     ^
In file included from sxplugin.c:4:0:
/home/bain/OpenWrt-Toolchain-ramips-mt7620_gcc-4.8-linaro_musl-1.1.10.Linux-x86_64/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_musl-1.1.10/include/string.h:52:8: note: expected 'const char *' but argument is of type 'byte *'
 size_t strlen (const char *);
        ^
sxplugin.c:53:5: warning: pointer targets in passing argument 1 of 'strlen' differ in signedness [-Wpointer-sign]
     byte* beforeMD5=malloc(strlen(timeByte)+strlen(userName)+strlen(RADIUS)+1);
     ^
In file included from sxplugin.c:4:0:
/home/bain/OpenWrt-Toolchain-ramips-mt7620_gcc-4.8-linaro_musl-1.1.10.Linux-x86_64/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_musl-1.1.10/include/string.h:52:8: note: expected 'const char *' but argument is of type 'byte *'
 size_t strlen (const char *);
        ^
sxplugin.c:53:5: warning: pointer targets in passing argument 1 of 'strlen' differ in signedness [-Wpointer-sign]
     byte* beforeMD5=malloc(strlen(timeByte)+strlen(userName)+strlen(RADIUS)+1);
     ^
In file included from sxplugin.c:4:0:
/home/bain/OpenWrt-Toolchain-ramips-mt7620_gcc-4.8-linaro_musl-1.1.10.Linux-x86_64/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_musl-1.1.10/include/string.h:52:8: note: expected 'const char *' but argument is of type 'byte *'
 size_t strlen (const char *);
        ^
sxplugin.c:58:5: warning: pointer targets in passing argument 1 of 'strcspn' differ in signedness [-Wpointer-sign]
     memcpy(beforeMD5 + 4 , userName , strcspn(userName,"@"));
     ^
In file included from sxplugin.c:4:0:
/home/bain/OpenWrt-Toolchain-ramips-mt7620_gcc-4.8-linaro_musl-1.1.10.Linux-x86_64/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_musl-1.1.10/include/string.h:46:8: note: expected 'const char *' but argument is of type 'byte *'
 size_t strcspn (const char *, const char *);
        ^
sxplugin.c:61:5: warning: pointer targets in passing argument 1 of 'strcat' differ in signedness [-Wpointer-sign]
     strcat(beforeMD5,RADIUS);//string_copy
     ^
In file included from sxplugin.c:4:0:
/home/bain/OpenWrt-Toolchain-ramips-mt7620_gcc-4.8-linaro_musl-1.1.10.Linux-x86_64/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_musl-1.1.10/include/string.h:34:7: note: expected 'char * __restrict__' but argument is of type 'byte *'
 char *strcat (char *__restrict, const char *__restrict);
       ^
sxplugin.c:61:5: warning: pointer targets in passing argument 2 of 'strcat' differ in signedness [-Wpointer-sign]
     strcat(beforeMD5,RADIUS);//string_copy
     ^
In file included from sxplugin.c:4:0:
/home/bain/OpenWrt-Toolchain-ramips-mt7620_gcc-4.8-linaro_musl-1.1.10.Linux-x86_64/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_musl-1.1.10/include/string.h:34:7: note: expected 'const char * __restrict__' but argument is of type 'byte *'
 char *strcat (char *__restrict, const char *__restrict);
       ^
sxplugin.c:63:5: warning: pointer targets in passing argument 1 of 'strlen' differ in signedness [-Wpointer-sign]
     info("4.length=<%d>",strlen(beforeMD5));
     ^
In file included from sxplugin.c:4:0:
/home/bain/OpenWrt-Toolchain-ramips-mt7620_gcc-4.8-linaro_musl-1.1.10.Linux-x86_64/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_musl-1.1.10/include/string.h:52:8: note: expected 'const char *' but argument is of type 'byte *'
 size_t strlen (const char *);
        ^
sxplugin.c:68:5: warning: pointer targets in passing argument 1 of 'strlen' differ in signedness [-Wpointer-sign]
[... 24 lines cut ...]
/home/bain/OpenWrt-Toolchain-ramips-mt7620_gcc-4.8-linaro_musl-1.1.10.Linux-x86_64/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_musl-1.1.10/include/string.h:31:7: note: expected 'const char * __restrict__' but argument is of type 'byte *'
 char *strcpy (char *__restrict, const char *__restrict);
       ^
sxplugin.c:18:11: warning: unused variable 'j' [-Wunused-variable]
     int i,j;
           ^
sxplugin.c: In function 'pap_modifyusername':
[... cut ...]
/home/bain/OpenWrt-Toolchain-ramips-mt7620_gcc-4.8-linaro_musl-1.1.10.Linux-x86_64/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_musl-1.1.10/include/string.h:31:7: note: expected 'const char * __restrict__' but argument is of type 'byte *'
 char *strcpy (char *__restrict, const char *__restrict);
       ^
sxplugin.c:124:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
mipsel-openwrt-linux-musl-gcc.bin: warning: environment variable 'STAGING_DIR' not defined
mipsel-openwrt-linux-musl-gcc.bin: warning: environment variable 'STAGING_DIR' not defined

mdluo, 2015-08-01:
遇到同样的问题，HG255D，固件是 PandoraBox 14.09 r460 ，使用 https://downloads.openwrt.org/snapshots/trunk/ramips/rt305x/OpenWrt-Toolchain-ramips-rt305x_gcc-4.8-linaro_musl-1.1.10.Linux-x86_64.tar.bz2 里的 mipsel-openwrt-linux-gcc 编译。改过 sxplugin.c 里的 pppd_version[] = "2.4.7"; 为 2.4.8 后仍然出现相同的报错，都是 Couldn't load plugin sxplugin.so 。不知能否给个可用的 sxplugin.so 文件，谢谢！

mdluo, 2015-08-02:
昨天折腾到凌晨3点终于成功了，改成了这个地址的 GCC：https://downloads.openwrt.org/barrier_breaker/14.07/ramips/rt305x/OpenWrt-Toolchain-ramips-for-mipsel_24kec%2bdsp-gcc-4.8-linaro_uClibc-0.9.33.2.tar.bz2，然后 sxplugin.c 里的 pppd_version[] 仍保持 2.4.7。只不过要注意新建的 NETKEEPER 网络接口防火墙设置要分配给 WAN 区域，而且 WAN 接口保存 DHCP 客户端

miao1007, 2015-08-03:
应该是GCC的问题，我也不太确定，如果有现成的so库，直接用HexEdit修改版本或者radius即可