miao1007/Openwrt-NetKeeper #20: Timeout waiting for PADS packets

saginardo, 2015-09-05:
大神好；我是用编译好的插件，按照教程进行设置，但是报错，无法连接网络，用的是openwrt，试了几个版本还是这个错误；radius使用的是ChongQing Netkeeper(0094): xianxinli1radius，经过验证这个radius是能够进行拨号的，我在电脑是能够使用这个进行拨号。

看错误提示应该是无法连接到pppoe，检查网线/内网设置；但是我的重邮内网是正常的；
类型: dhcp
地址: 172.18.134.63
子网掩码: 255.255.252.0
网关: 172.18.132.1
DNS 1: 61.128.128.68
DNS 2: 202.202.32.33
DNS 3: 202.202.32.34

下面是日志：
Sat Sep  5 10:21:15 2015 daemon.notice netifd: Interface 'NetKeeper' is setting up now
Sat Sep  5 10:21:15 2015 daemon.notice netifd: NetKeeper (8365): uci: Entry not found
Sat Sep  5 10:21:15 2015 daemon.notice netifd: NetKeeper (8365): uci: Entry not found
Sat Sep  5 10:21:15 2015 daemon.info pppd[8377]: Plugin rp-pppoe.so loaded.
Sat Sep  5 10:21:15 2015 daemon.info pppd[8377]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Sat Sep  5 10:21:15 2015 daemon.info pppd[8377]: Plugin sxplugin.so loaded.
Sat Sep  5 10:21:15 2015 daemon.info pppd[8377]: sxplugin : init
Sat Sep  5 10:21:15 2015 daemon.info pppd[8377]: -------------------------------------
Sat Sep  5 10:21:15 2015 daemon.info pppd[8377]: timenow(Hex)=%ßx
Sat Sep  5 10:21:15 2015 daemon.info pppd[8377]: Begin : beforeMD5
Sat Sep  5 10:21:15 2015 daemon.info pppd[8377]: 1.<.?  ?w>
Sat Sep  5 10:21:15 2015 daemon.info pppd[8377]: 2.<.?0614827>
Sat Sep  5 10:21:15 2015 daemon.info pppd[8377]: 3.<.?0614827xianxinli1radius>
Sat Sep  5 10:21:15 2015 daemon.info pppd[8377]: 4.length=<27>
Sat Sep  5 10:21:15 2015 daemon.info pppd[8377]: End : beforeMD5
Sat Sep  5 10:21:15 2015 daemon.info pppd[8377]: Begin : afterMD5
Sat Sep  5 10:21:15 2015 daemon.info pppd[8377]: 1.MD5use_1=< d>
Sat Sep  5 10:21:15 2015 daemon.info pppd[8377]: 2.MD5use_2=< 4>
Sat Sep  5 10:21:15 2015 daemon.info pppd[8377]: End : afterMD5
Sat Sep  5 10:21:15 2015 daemon.info pppd[8377]: -------------------------------------
Sat Sep  5 10:21:15 2015 daemon.info pppd[8377]: sxplugin : user  is <
1]YSYO d40614827@cqupt> 
Sat Sep  5 10:21:15 2015 daemon.info pppd[8377]: sxplugin : passwd loaded
Sat Sep  5 10:21:15 2015 daemon.notice pppd[8377]: pppd 2.4.7 started by root, uid 0
Sat Sep  5 10:21:31 2015 daemon.warn pppd[8377]: Timeout waiting for PADS packets
Sat Sep  5 10:21:31 2015 daemon.err pppd[8377]: Unable to complete PPPoE Discovery
Sat Sep  5 10:21:31 2015 daemon.info pppd[8377]: Exit.
Sat Sep  5 10:21:31 2015 daemon.notice netifd: Interface 'NetKeeper' is now down

希望大神看到能帮帮忙，不尽感激

miao1007, 2015-09-05:
自行配置静态路由表

Jeffrey95, 2015-09-08:
应该是重邮pppoe拨号要内网验证？先设置静态路由让你可以通过路由器脸上内网，然后就可以拨号成功了。找了好久解决这个问题，之前一直重刷都不管用。发现是内网问题后，又纠结了很久为什么我的这条网线插上之后不能直接连上内网，但可以用netkeeper连上外网。可能是硬件有点问题吧，换了室友那个位置的网线插上就好了。

milescr, 2015-10-04:
我配置了静态路由表也没能连上，但是我把netkeeper的物理接口改成了eth1就能连上了，结果访问不了内网= =

closed 2021-05-01 by saginardo