miao1007/Openwrt-NetKeeper #22: 已经升级radius 方式

jingpg, 2015-09-08:
新客户端已经改了radius,貌似是随机值了.

reusu, 2015-09-08:
如果现在还能播的上去 但是会定时断开的话 是不是心跳的问题 如果是radius问题应该无法通过验证吧

miao1007, 2015-09-08:
怎么看出来的？

reusu, 2015-09-08:
请问下链接上之后不定时2-10分钟断开
是心跳问题么
请问如何解决

miao1007, 2015-09-08:
心跳或者PPPoE的LCP配置问题

reusu, 2015-09-08:
pppd的日志显示LCP terminated by peer
心跳的话需要怎么实现
使用那个安卓客户端么

reusu, 2015-09-08:
[image]
图示如上 感谢回答

miao1007, 2015-09-08:
建议换一个刷机包，如果用官方的还是有这个问题，就是电信改了

reusu, 2015-09-08:
用的是官方的14.07 然后pppd+plugins 拨号正常 但是会自动断。。。。。。
我看您在readme中有提到过心跳机制
请问和这个有关么
如果有关该怎么实现心跳

miao1007, 2015-09-08:
心跳要定时发送套接字来实现，这个折腾太费事了。。。

reusu, 2015-09-08:
您确定自动断开是心跳没维持导致的结果么
如果是的话我去看看安卓端的实现

realityone, 2015-09-08:
拨号完成后 Wireshark 抓包 10 分钟。
~~然后贴上来我看下~~

Sg4Dylan, 2015-09-08:
Android版本有心跳
程序及反编译后的project放在zip压缩包里
http://1drv.ms/1Kv02bJ

realityone, 2015-09-08:
拿官方 Netkeeper 拨号完成后抓包挂个 10 分钟左右
Android 的心跳对这个可能没什么研究意义

Sg4Dylan, 2015-09-08:
室友不间断打LOL断网不能，让学弟抓了十分钟 = =
还是这里 http://1drv.ms/1Kv02bJ

realityone, 2015-09-08:
你们的心跳主要包是发到 `117.21.209.186    heart.fsjpt.jx` 的 443 端口的一个 UDP。现在我看见的是 120s 一次，不过这个其实是服务端控制的，在心跳开始前由另外一个包控制频率。

```
0000   48 52 33 30 05 00 00 00 b0 71 4e 2d 64 71 1e 3f  HR30.....qN-dq.?
0010   49 91 c2 69 6c 97 24 56 e3 ac e0 f4 c6 26 3a 42  I..il.$V.....&:B
0020   ea ef d6 92 10 d0 c8 fa 22 32 7b bd 1a f4 21 16  ........"2{...!.
0030   54 05 5e 65 a2 35 82 69 25 58 20 40 c4 8f 03 07  T.^e.5.i%X @....
0040   31 fd 6d 7f 1b 64 50 88 ef 79 4e 86 fe c7 c6 2d  1.m..dP..yN....-
0050   53 43 43 f4 ed a3 d0 6c 4f c1 3d 7e 95 2b d0 a5  SCC....lO.=~.+..
0060   fb 07 69 76 05 a4 7f 09 a8 72 9a fe 2f 60 b7 2b  ..iv.....r../`.+
0070   38 04 56 1c 61 da 78 cd 93 1f e0 c6 d2 ee ce b3  8.V.a.x.........
0080   c4 57 ce 3d 90 1c bb cb 94 31 c0 ae ce 43 6e 80  .W.=.....1...Cn.
0090   ad 85 dd 79 9d 5e 49 9e e2 1a e2 c4 b4 80 f2 80  ...y.^I.........
00a0   a1 95 d1 e7 20 dd eb 7c 8e b4 5a 43 bb 08 c6 73  .... ..|..ZC...s
00b0   7c b9 83 1b 80 b3 bc 15 97                       |........
```

完整的一个包差不多就是这样 `HR` 是魔术字，`30` 应该是心跳包版本，0x05 是包类型，之后是四个字节包长（只包含后面的数据）。

之后一块 `0x71 - 0x97` 的部分就是心跳包的主内容，AES128/ECB/PKCS5Padding，密钥是 `wanglibinwanglei`，不含初始向量。

密钥来源于

```
    # .data:0044B0FC 00000021 C c0d3c068f22f97b567af4eb521d3c022
    # .data:0044AE84 00000021 C c0d3c068f2e9c0b5215b86769810016b
```

是一个信利自造算法，逆出来不是什么问题，在我的 repo 上你们也能找到，其他学校大部分密钥也都是这么保护的。

对于那一块心跳数据，解开来大概就是 

```
TYPE=HEARTBEAT&USER_NAME=********&PASSWORD=********&IP=10.8.0.3&MAC=00%2D00%2D00%2D00%2D00%2D00&DRIVER=1&VERSION_NUMBER=2.5.0016v32&HOOK=&IP2=10.8.0.3
```

最后可能会有几个凑位字符，其实内容很清晰了，别的学校的包大概结构也是这个样子，一次 UDP 后服务器可能会返回一个包，解开来有一个 KEY 参数，就是下次要带上的参数，也有地方是不带的，因为你们有了 AES_KEY 之后怎么研究都比较方便了。

其他还有几个类似的包，到 444 或者 445 之类的，以前也有菊苣想全部解开，不够后来证明其他包都是没意义的之后也都懒掉了，基本上就是这个包是有意义的。

大部分学校情况就是这样，想调试的话可以直接断在几个加密算法的常量上。

浙江地区的心跳已经全部都换了，如果你们有用的话就拿去套一套加解密，其实都不难。

感觉就像是把之前菊苣们的话又传递了一遍，就酱。

Sg4Dylan, 2015-09-08:
Android的心跳前天已经看过了，现在忙课设不大想开这个坑，课设完了再折腾(毕竟心跳检查不如某些寝室楼层最快两分钟一断线，可以安心用路由)。
能否给一份早先研究过的菊苣们留下的讨论和相关代码(传送门就行

realityone, 2015-09-09:
https://github.com/nowind/sx_pi
可以追着 commit 看最早时候的样子，密钥是肯定不一样的，过程都差不多

Sg4Dylan, 2015-09-09:
那就这样，不明白地方还请指教～

reusu, 2015-09-09:
请问下密钥的逆算是在哪个repo里_(´ཀ`」 ∠)_

realityone, 2015-09-09:
Anti_teNelgniS/blob/master/src/XLStringEnc.py

reusu, 2015-09-09:
very thx

reusu, 2015-09-09:
江西昨天用plugins还能拨号成功
今天就验证失败了
返回[30|0]to many access
求原因

reusu, 2015-09-09:
[image]
今天刚写完一个心跳实现准备测试下
发现连机会都不给了
直接auth fail

realityone, 2015-09-09:
换官方 Netkeeper 试试
一般情况下 shareKey 是不会动态变化的

reusu, 2015-09-09:
找到原因了
openwrt的mac地址被ban了
可能是多次没有保活就当作黑户处理了
换了一个mac地址就正常了
保活软件也是正常跑  超过15分钟不断了
感谢@realityone

Sg4Dylan, 2015-09-09:
too many access 解决很简单的，直接拔下wan的线，五分钟插回去就可以了
主要是你前一次未正确挂断

reusu, 2015-09-09:
关于too many这点 因为是朋友让我帮忙处理 本身对闪讯也不了解 开始工作也没环境测试 所以比较无知请谅解

Sg4Dylan, 2015-09-09:
看楼上的拨号账户应该是隔壁学校的（
可以发一份源码看看么，这边可以顺便协助测试

reusu, 2015-09-09:
保活程序还是账号生成程序呢  
用到现在没断保活应该是成功了

Sg4Dylan, 2015-09-09:
帐号生成的用sxplugin就可以了，看看心跳的程序

reusu, 2015-09-09:
因为这边学校是EAZY MODE（无动态key）
所以我就随便写了个比较死板的东西
发包内容都是一样的 见笑了
https://github.com/reusu/NetKeeperHeartbeat_JXS

Sg4Dylan, 2015-09-09:
作为一个日常C、Python & Shell 的渣渣
看到Java感到了一丝的忧伤
Thx
[image]

reusu, 2015-09-09:
不管怎样非常感谢您们维护的sxplugin

miao1007, 2015-09-09:
换上okio的库吧，这个看的太蛋疼了

reusu, 2015-09-09:
关于语言这事儿
我也可以说python的代码
都是边看边google的吗 ._.

sunflyer, 2015-09-12:
@realityone 按照那个StringEnc解出来这个 key不是还有一个 wangl0jiangxi876 么。。。。。。。。。 = =

realityone, 2015-09-12:
我也不知道那东西干嘛的
可能是用来加密别的包的
也有可能根本没用就是编译的时候进来的

xseven007, 2015-09-26:
在mipsel设备上做心跳就是挖大坑。再说江西那边看起来也没打算要搞心跳的样子

reusu, 2015-09-28:
对楼上 - 我朋友的学校 江西的 已经是需要心跳维持才可以正常上网了 不然pppoe拨号之后10分钟必断 测试路由是WNDR4300 跑心跳资源还是很富裕的

realityone, 2015-09-28:
你们有兴趣继续开发的话我发个 pr 上来

sunflyer, 2015-09-28:
@realityone 我已经在Simple Netkeeper Windows加入了重庆和江西的心跳支持了，江西心跳还特么内网IP也是感人，也难怪之前发心跳还掉线。。。然而重庆的mac客户端更新后那个upx壳完全当没有壳。。。。。。

btw ，今天的github怎么回事，这么卡

Sg4Dylan, 2015-09-28:
大约有4成的有线网拨号用户受心跳影响，绝大多数是使用TP/Mercury，有SimpleNetkeeper的心跳支持便是极好的。只是在重灾区，使用官方netkeeper的都会因为多线程下载被断网警告，不知道上openwrt做心跳会怎样。

sunflyer, 2015-09-28:
@Sg4Dylan 路由器用特殊3就好了，simple的心跳支持对江西只有本地拨号有效。因为江西校园网是内网ip，我还有bug没改完。用路由器再用客户端发心跳其实还是没特殊拨号模式3那么方便

xseven007, 2015-09-28:
@reusu 好吧，估计是我out了，之前做NetKeeper win版本的时候并没有加心跳，反正代码有几年了，懒得回头去维护了

Sg4Dylan, 2015-09-28:
校园网PPPoE得到的是不封任何端口的公网IP。由客户端发心跳得做端口映射，又是个大坑。

sunflyer, 2015-09-28:
@Sg4Dylan  不对，如果你们获取到的IP是100开头的则全部是内网IP，我这边有几个江西的同学测试过了返回的都是100开头的内网IP，所以之前直接获取外网IP发心跳的逻辑完全行不通。
另外，客户端发心跳不需要端口映射啊，你实在是担心的话开个50000+的端口来发心跳就好了，服务器没有回复包的

realityone, 2015-09-28:
openwrt 的话用 `getifaddrs` 拿到 ip 之后 `bind` 掉就行了