miao1007/Openwrt-NetKeeper #23: 首先谢谢你所做的一切，出现了一些问题麻烦帮我看一看

latte-z, 2015-09-09:
我的路由是miwifimini，编译的so可以正常使用，也能正常转换出系统日志中类似的账号。但是在passwd loaded之后，会出现 Connection terminated -> Modem hangup -> exit  ，后续还有CHAP authentication failed:[code 13]:Not special client的提示。

我是南昌电信的校园网，用的是v32的星空极速，编译前确认修改了TODO CODE

具体如图所示，希望能帮我看看，非常感谢！

[image]

miao1007, 2015-09-09:
[Uploading QQ图片20150909200138.jpg…]()......

latte-z, 2015-09-09:
Sorry 之前上传图片的时候出现了网络故障

miao1007, 2015-09-09:
应该是帐号没有正确生成，看下时间是否精确

latte-z, 2015-09-09:
好的，我再试试，现在我也是比较头疼，因为无法拨号就不能同步时间。之前几次靠着 date " " 强制调整时间能完成拨号，但是接收了几十个数据包之后就自动断开了。

miao1007, 2015-09-09:
不知道你们学校内网有没有ntp服务器

latte-z, 2015-09-09:
这个估计是没有的....我再自己想想办法，非常感谢，不打扰了

realityone, 2015-09-09:
用浏览器时间同步

miao1007, 2015-09-09:
扫描 123端口 试试

latte-z, 2015-09-09:
谢谢~刚才我已经用windows自带的服务开了个NTP服务器，不过奇怪的是之前偶尔能拨通，现在已经彻底失败了- - 不知道校园网还加了什么黑科技，总之感谢帮助

lknife, 2015-09-10:
我这也是这样，会自动断开，看个视频都看不完。。

latte-z, 2015-09-10:
研究了一下，南昌电信的校园网有心跳检测，要把路由改成原来用的特殊3+克隆mac才有用

lknife, 2015-09-11:
那你和我的还不一样，我的可以直接拨号成功，但是用着用着就会突然断开，比如视频，加载了十来兆就断开，刷新一下网页又有网，这样反复循环，玩游戏也是一样

qq734694920, 2016-09-23:
我觉得扫描到了123端口的话怎么同步时间呢，只有一个ppope连接外网