miao1007/Openwrt-NetKeeper #32: 求指导重邮怎么上内网

s752550916, 2015-09-23:
能拨号上外网，但是上不了内网，能不能给个相关教程？

saginardo, 2015-09-23:
同是大重邮的，请教下你是怎样拨上号 的；我连接wan用的dhcp，然后能够通过路由器上内网；拨号时候按照教程，时间已同步，netkeeper已经添加到wan防火墙，但是还是不能拨号，直接提示：Unable to complete PPPoE Discovery；求指导，是不是需要配置什么静态路由表什么的。

saginardo, 2015-09-23:
还有时候会提示：pap authentication failed

saginardo, 2015-09-23:
Sat Sep 5 10:21:15 2015 daemon.notice netifd: Interface 'NetKeeper' is setting up now
Sat Sep 5 10:21:15 2015 daemon.notice netifd: NetKeeper (8365): uci: Entry not found
Sat Sep 5 10:21:15 2015 daemon.notice netifd: NetKeeper (8365): uci: Entry not found
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: Plugin rp-pppoe.so loaded.
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: Plugin sxplugin.so loaded.
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: sxplugin : init
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: -------------------------------------
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: timenow(Hex)=%ßx
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: Begin : beforeMD5
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: 1.<.? ?w>
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: 2.<.?0614827>
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: 3.<.?0614827xianxinli1radius>
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: 4.length=<27>
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: End : beforeMD5
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: Begin : afterMD5
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: 1.MD5use_1=< d>
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: 2.MD5use_2=< 4>
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: End : afterMD5
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: -------------------------------------
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: sxplugin : user is <
1]YSYO d40614827@cqupt> 
Sat Sep 5 10:21:15 2015 daemon.info pppd[8377]: sxplugin : passwd loaded
Sat Sep 5 10:21:15 2015 daemon.notice pppd[8377]: pppd 2.4.7 started by root, uid 0
Sat Sep 5 10:21:31 2015 daemon.warn pppd[8377]: Timeout waiting for PADS packets
Sat Sep 5 10:21:31 2015 daemon.err pppd[8377]: Unable to complete PPPoE Discovery
Sat Sep 5 10:21:31 2015 daemon.info pppd[8377]: Exit.
Sat Sep 5 10:21:31 2015 daemon.notice netifd: Interface 'NetKeeper' is now down

saginardo, 2015-09-23:
wan口
类型: dhcp
地址: 172.18.134.63
子网掩码: 255.255.252.0
网关: 172.18.132.1
DNS 1: 61.128.128.68
DNS 2: 202.202.32.33
DNS 3: 202.202.32.34

s752550916, 2015-09-23:
我也就只是按照教程来的，我也是小白，不懂你的问题。

s752550916, 2015-09-24:
@saginardo 谢谢了，上内网了。我系统日志是这样，太详细的我也不懂，看你看一下能不能参考。
Thu Sep 24 19:50:07 2015 daemon.notice netifd: Interface 'NetKeeper' is setting up now
Thu Sep 24 19:50:07 2015 daemon.notice netifd: NetKeeper (3676): uci: Entry not found
Thu Sep 24 19:50:07 2015 daemon.info pppd[3692]: Plugin rp-pppoe.so loaded.
Thu Sep 24 19:50:07 2015 daemon.info pppd[3692]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.6
Thu Sep 24 19:50:07 2015 daemon.info pppd[3692]: Plugin sxplugin.so loaded.
Thu Sep 24 19:50:07 2015 daemon.info pppd[3692]: sxplugin : init
Thu Sep 24 19:50:07 2015 daemon.info pppd[3692]: -------------------------------------
Thu Sep 24 19:50:07 2015 daemon.info pppd[3692]: timenow(Hex)=%ßx
Thu Sep 24 19:50:07 2015 daemon.info pppd[3692]: Begin : beforeMD5
Thu Sep 24 19:50:07 2015 daemon.info pppd[3692]: 1.<3?I>
Thu Sep 24 19:50:07 2015 daemon.info pppd[3692]: 2.<3?I189****0250w?A&wP`8w????H!?>
Thu Sep 24 19:50:07 2015 daemon.info pppd[3692]: 3.<3?I189****0250xianxinli1radius>
Thu Sep 24 19:50:07 2015 daemon.info pppd[3692]: 4.length=<31>
Thu Sep 24 19:50:07 2015 daemon.info pppd[3692]: End : beforeMD5
Thu Sep 24 19:50:07 2015 daemon.info pppd[3692]: Begin : afterMD5
Thu Sep 24 19:50:07 2015 daemon.info pppd[3692]: 1.MD5use_1=< 3>
Thu Sep 24 19:50:07 2015 daemon.info pppd[3692]: 2.MD5use_2=< 7>
Thu Sep 24 19:50:07 2015 daemon.info pppd[3692]: End : afterMD5
Thu Sep 24 19:50:07 2015 daemon.info pppd[3692]: -------------------------------------
Thu Sep 24 19:50:07 2015 daemon.info pppd[3692]: sxplugin : user  is <
HTR0M 37189****0250@cqupt> 
Thu Sep 24 19:50:07 2015 daemon.info pppd[3692]: sxplugin : passwd loaded
Thu Sep 24 19:50:07 2015 daemon.notice pppd[3692]: pppd 2.4.6 started by root, uid 0
Thu Sep 24 19:50:07 2015 daemon.info pppd[3692]: PPP session is 8004
Thu Sep 24 19:50:07 2015 daemon.warn pppd[3692]: Connected to 00:18:82:1e:b1:97 via interface eth0.2
Thu Sep 24 19:50:07 2015 daemon.info pppd[3692]: Using interface pppoe-NetKeeper
Thu Sep 24 19:50:07 2015 daemon.notice pppd[3692]: Connect: pppoe-NetKeeper <--> eth0.2
Thu Sep 24 19:50:07 2015 daemon.info pppd[3692]: syncppp not active
Thu Sep 24 19:50:08 2015 daemon.info pppd[3692]: Remote message: 2001
Thu Sep 24 19:50:08 2015 daemon.notice pppd[3692]: PAP authentication succeeded
Thu Sep 24 19:50:08 2015 daemon.notice pppd[3692]: peer from calling number 00:18:82:1E:B1:97 authorized
Thu Sep 24 19:50:08 2015 daemon.notice pppd[3692]: local  IP address 113.250.155.51
Thu Sep 24 19:50:08 2015 daemon.notice pppd[3692]: remote IP address 113.250.152.1
Thu Sep 24 19:50:08 2015 daemon.notice pppd[3692]: primary   DNS address 61.128.128.68
Thu Sep 24 19:50:08 2015 daemon.notice pppd[3692]: secondary DNS address 61.128.192.68
Thu Sep 24 19:50:08 2015 daemon.notice netifd: Network device 'pppoe-NetKeeper' link is up
Thu Sep 24 19:50:08 2015 daemon.notice netifd: Interface 'NetKeeper' is now up
Thu Sep 24 19:50:08 2015 user.notice firewall: Reloading firewall due to ifup of NetKeeper (pppoe-NetKeeper)
Thu Sep 24 19:50:09 2015 daemon.info dnsmasq[3628]: exiting on receipt of SIGTERM
Thu Sep 24 19:50:09 2015 daemon.info dnsmasq[3770]: started, version 2.71 cachesize 150
Thu Sep 24 19:50:09 2015 daemon.info dnsmasq[3770]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-auth no-DNSSEC
Thu Sep 24 19:50:09 2015 daemon.info dnsmasq-dhcp[3770]: DHCP, IP range 192.168.1.100 -- 192.168.1.249, lease time 12h
Thu Sep 24 19:50:09 2015 daemon.info dnsmasq[3770]: using local addresses only for domain lan
Thu Sep 24 19:50:09 2015 daemon.info dnsmasq[3770]: reading /tmp/resolv.conf.auto
Thu Sep 24 19:50:09 2015 daemon.info dnsmasq[3770]: using local addresses only for domain lan
Thu Sep 24 19:50:09 2015 daemon.info dnsmasq[3770]: using nameserver 61.128.128.68#53
Thu Sep 24 19:50:09 2015 daemon.info dnsmasq[3770]: using nameserver 61.128.192.68#53
Thu Sep 24 19:50:09 2015 daemon.info dnsmasq[3770]: using nameserver 61.128.128.68#53
Thu Sep 24 19:50:09 2015 daemon.info dnsmasq[3770]: using nameserver 202.202.32.33#53
Thu Sep 24 19:50:09 2015 daemon.info dnsmasq[3770]: read /etc/hosts - 1 addresses
Thu Sep 24 19:50:09 2015 daemon.info dnsmasq[3770]: read /tmp/hosts/dhcp - 0 addresses
Thu Sep 24 19:50:09 2015 daemon.info dnsmasq[3770]: read /tmp/hosts/odhcpd - 1 addresses
Thu Sep 24 19:50:09 2015 daemon.info dnsmasq-dhcp[3770]: read /etc/ethers - 0 addresses
Thu Sep 24 19:50:16 2015 daemon.warn dnsmasq[3770]: possible DNS-rebind attack detected: cwzx.cqupt.edu.cn

saginardo, 2015-09-24:
你用的是pppd 2.4.6  我的是pppd2.4.7  不知道是不是这个原因。。。但是作者不是说用的2.4.7么。。。困惑中

saginardo, 2015-09-24:
另：我用simple netkeeper能够是FAST的一款路由器用上了；能不能发一份sxplugin.so到我的邮箱 谢谢  我看看是不是我编译的有问题  saginardo@163.com

saginardo, 2015-09-24:
还有个你的寝室网线插到wan口 用的是不是DHCP自动获取的内网地址，没有进行其他的配置么？

milescr, 2015-10-04:
最后怎么解决的？没有看懂

saginardo, 2015-10-04:
编辑/etc/config/network  添加如下路由表，只需要修改你的option gateway （ 这里的网关是我寝室的网关（172.18.132.1），你的肯定不一样，你可以看看你的路由器用dhcp获取内网ip时候用的是什么网关），其他默认就可以同时内外网：
PS：netkeeper用的好像就是和wan一个接口，在我的路由器上wan和netkeeper都是用的eth0.2这个接口，你的估计是eth1这个接口

config route
	option interface 'wan'
	option target '202.202.32.0'
	option netmask '255.255.240.0'
	option gateway '172.18.132.1'
	option metric '20'

config route
	option interface 'wan'
	option target '172.16.0.0'
	option netmask '255.240.0.0'
	option gateway '172.18.132.1'
	option metric '20'

config route
	option interface 'wan'
	option target '192.168.0.0'
	option netmask '255.255.0.0'
	option gateway '172.18.132.1'
	option metric '20'

config route
	option interface 'wan'
	option target '211.83.208.0'
	option netmask '255.255.240.0'
	option gateway '172.18.132.1'
	option metric '20'

config route
	option interface 'wan'
	option target '219.153.62.64'
	option netmask '255.255.255.192'
	option gateway '172.18.132.1'
	option metric '20'

config route
	option interface 'wan'
	option target '222.177.140.0'
	option netmask '255.255.255.128'
	option gateway '172.18.132.1'
	option metric '20'

config route
	option interface 'wan'
	option target '10.10.10.0'
	option netmask '255.255.255.0'
	option gateway '172.18.132.1'
	option metric '20'

milescr, 2015-10-05:
@saginardo 谢谢！连上内网了！

kuri-leo, 2015-10-24:
@saginardo 学长你好，我也是重邮的，麻烦请问一下怎么设置时间同步？我之前设置了路由表后可以重启后自动设置时间，但是过了几天就不能同步了。。。麻烦想请教一下

saginardo, 2015-10-24:
@kuri-leo 重邮的NTP服务器是 202.202.43.198 202.202.43.198 ；把这个填在openwrt顶部 系统选项卡下的第一项 系统设置中的时间同步设置 替换掉openwrt的默认ntp服务器；按理说，能够上内网，路由器重启也会自动同步时间

kuri-leo, 2015-10-24:
@saginardo 谢谢，刚刚试了下可以自动同步了。我也在纳闷，前几天都好好的，然后就不行了。。。。

ousun, 2015-10-27:
赞一个！

Citron504, 2015-11-08:
@saginardo 我是把netkeeper设置到了原来wan口用的eth0.1上面才拨得上号（这时候wan口就被挤掉了 ， 后来我干脆删掉了...）...然后我设置的路由表都是对netkeeper设置的  但是换掉网关后  还是不能访问内网  然后我发现掩码也不一样...把掩码也换掉了  还是上不了内网啊

saginardo, 2015-11-08:
把netkeeper接口设置到eth0.1 拨号成功后挤掉内网是 正常的，而且openwrt里面显示的ipv4
链接也只有PPPOE-netkeeper这个；所以这个时候才需要添加路由表，以此来描述内网数据包的流向；wan口不需要删除；成功拨上号以后，路由表显示的应该就是wan口的数据流向；
还有就是只需要换网关 ，掩码不能改（255.255.252.0）是固定的，跃点最好是20
40等，需要大于0（PPPOE—netkeeper跃点是0）；同时应该注意PPPOE-netkeeper的防火墙规则应该和wan口一致。
同时我看新版的openwrt-netkeeper作者好像已经用了自动配置的方法来帮忙配置路由器；make用于生成对应so文件，make
upload用于上传并设置路由器网络链接，你可以重新编译试试，注意一些编译参数的修改，应该就没问题了。
至此，好像我都没啥需要补充的，希望你能配置成功。
我在附上我拨号成功的截图吧

Citron504, 2015-11-08:
@saginardo   好的谢谢，按照你上面的设置是不是可以访问所有内网的地址了？ 还有我可以在Luci下面的静态路由加上上面的地址吧  觉得是防火墙的设置没有和netkeeper的设置一样 回去试试

Citron504, 2015-11-08:
@saginardo 昨晚回来试了一下已经可以了   跃点设置为40 掩码改了回来 还手动加了两个内网地址.......可以使用的时候可以在接口看到netkeeper和内网都获取了ip  但是今天早上起来好像又不行了  有时候内网直接没有ip  或者两边都有ip  内网上不了  外网慢得几乎没速度

Citron504, 2015-11-08:
@saginardo 多试了几次..比如在路由表界面按了几次保存  给内网按了几次连接再连外网   竟然又好了

closed 2015-11-24 by s752550916

sunflowerorange, 2017-04-14:
@saginardo 感谢 解决了这么久的困惑 重邮移通

orange62, 2018-06-13:
@saginardo 麻烦问下 老毛子能不能用这种方法连内网？

saginardo, 2018-06-15:
@orange62 我记得老毛子固件设置wan里面，有个选项可以关闭自动获取Main IP，然后自己手动填入就行。

orange62, 2018-06-16:
@saginardo 非常感谢 设置成功了

orange62, 2018-06-16:
@saginardo 非常感谢 添加了静态路由就行了