miao1007/Openwrt-NetKeeper #40: make 的时候出错

killgoo, 2015-10-03:
make之后显示如下：
kelo@kelo-VirtualBox:~/netkeeper/Openwrt-NetKeeper-master/src$ make
/home/kelo/netkeeper/OpenWrt-Toolchain-atheros-for-mips_mips32-gcc-4.8-linaro_uClibc-0.9.33.2/toolchain-mips_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/bin/mips-openwrt-linux-gcc: 行 82: /home/kelo/netkeeper/OpenWrt-Toolchain-atheros-for-mips_mips32-gcc-4.8-linaro_uClibc-0.9.33.2/toolchain-mips_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/bin/mips-openwrt-linux-uclibc-gcc.bin: cannot execute binary file: 可执行文件格式错误
/home/kelo/netkeeper/OpenWrt-Toolchain-atheros-for-mips_mips32-gcc-4.8-linaro_uClibc-0.9.33.2/toolchain-mips_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/bin/mips-openwrt-linux-gcc: 行 82: /home/kelo/netkeeper/OpenWrt-Toolchain-atheros-for-mips_mips32-gcc-4.8-linaro_uClibc-0.9.33.2/toolchain-mips_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/bin/mips-openwrt-linux-uclibc-gcc.bin: 成功

makefile:34: recipe for target 'chongqing_sxplugin' failed

make: *** [chongqing_sxplugin] Error 126

kelo@kelo-VirtualBox:~/netkeeper/Openwrt-NetKeeper-master/src$ 

请解释，我是新手对编译了解的不多，请见谅

lycycycycyc, 2015-10-04:
把你的makefile贴出来看看

killgoo, 2015-10-10:
已经弄好了，谢谢，原因是使用了不正确的linux系统版本

prettys, 2016-09-11:
同  ，换了高版本好了  。  但出现警告  环境没有定义

closed 2016-09-11 by miao1007