miao1007/Openwrt-NetKeeper #43: 武汉e信路由器是360C301

qlingxing, 2015-10-10:
Thu Oct  8 18:12:09 2015 daemon.info pppd[3739]: sxplugin : passwd loaded
Thu Oct  8 18:12:09 2015 daemon.notice pppd[3739]: pppd 2.4.7 started by root, uid 0
Thu Oct  8 18:12:24 2015 daemon.warn pppd[3739]: Timeout waiting for PADO packets
Thu Oct  8 18:12:24 2015 daemon.err pppd[3739]: Unable to complete PPPoE Discovery
Thu Oct  8 18:12:24 2015 daemon.info pppd[3739]: Exit.
Thu Oct  8 18:12:24 2015 daemon.notice netifd: Interface 'NetKeeper' is now down
Thu Oct  8 18:12:24 2015 daemon.notice netifd: Interface 'NetKeeper' is disabled
Thu Oct  8 18:12:24 2015 daemon.notice netifd: Interface 'NetKeeper' is enabled
Thu Oct  8 18:12:24 2015 daemon.notice netifd: Interface 'NetKeeper' is setting up now
Thu Oct  8 18:12:24 2015 daemon.info pppd[3779]: Plugin rp-pppoe.so loaded.
Thu Oct  8 18:12:24 2015 daemon.info pppd[3779]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
这个怎么解决啊

zcw1993, 2016-09-17:
你搞定没    同型号，求发固件知道哈

miao1007, 2016-09-23:
```
Thu Oct 8 18:12:24 2015 daemon.warn pppd[3739]: Timeout waiting for PADO packets
```

你的包没有发出去，需要看看你内网、路由表是否设置有误

zcw1993, 2016-09-23:
大大，我找到原因了，今天晚上搞定了，终于可以愉快的玩耍了，谢谢

qlingxing, 2016-09-24:
这么厉害 怎么解决的啊

zcw1993, 2016-09-24:
你加我QQ153612405  我给你给个库，然后教你操作，我也是昨天才搞好的。现在用起来很不错，我们都是一样的路由，这个我会了