miao1007/Openwrt-NetKeeper #44: MT7620N 武汉E信  登陆问题

Vicrime, 2015-10-14:
武汉E信     MT7620N

编译时
Warning: Unable to locate ipset utility, disabling ipset support
Warning: Section @zone[1] (wan) cannot resolve device of network 'netkeeper'

系统日志
Wed Oct 14 17:33:17 2015 daemon.notice netifd: Interface 'netkeeper' is now down
Wed Oct 14 17:33:17 2015 daemon.notice netifd: Interface 'netkeeper' is setting up now
Wed Oct 14 17:33:17 2015 daemon.info pppd[4059]: Plugin rp-pppoe.so loaded.
Wed Oct 14 17:33:17 2015 daemon.info pppd[4059]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Wed Oct 14 17:33:17 2015 daemon.info pppd[4059]: Plugin wuhan_sxplugin.so loaded.
Wed Oct 14 17:33:17 2015 daemon.info pppd[4059]: sxplugin : init
Wed Oct 14 17:33:17 2015 daemon.info pppd[4059]: -------------------------------------
Wed Oct 14 17:33:17 2015 daemon.info pppd[4059]: timenow(Hex)=%ßx
Wed Oct 14 17:33:17 2015 daemon.info pppd[4059]: Begin : beforeMD5
Wed Oct 14 17:33:17 2015 daemon.info pppd[4059]: 1.<99?>
Wed Oct 14 17:33:17 2015 daemon.info pppd[4059]: 2.<99?1862674240>
Wed Oct 14 17:33:17 2015 daemon.info pppd[4059]: 3.<99?1862674240hubtxinli01>
Wed Oct 14 17:33:17 2015 daemon.info pppd[4059]: 4.length=<25>
Wed Oct 14 17:33:17 2015 daemon.info pppd[4059]: End : beforeMD5
Wed Oct 14 17:33:17 2015 daemon.info pppd[4059]: Begin : afterMD5
Wed Oct 14 17:33:17 2015 daemon.info pppd[4059]: 1.MD5use_1=< d>
Wed Oct 14 17:33:17 2015 daemon.info pppd[4059]: 2.MD5use_2=< 0>
Wed Oct 14 17:33:17 2015 daemon.info pppd[4059]: End : afterMD5
Wed Oct 14 17:33:17 2015 daemon.info pppd[4059]: -------------------------------------
Wed Oct 14 17:33:17 2015 daemon.info pppd[4059]: sxplugin : user  is <
`Z3 [ d01862674240> 
Wed Oct 14 17:33:17 2015 daemon.info pppd[4059]: sxplugin : passwd loaded
Wed Oct 14 17:33:17 2015 daemon.notice pppd[4059]: pppd 2.4.7 started by root, uid 0
Wed Oct 14 17:33:17 2015 daemon.info pppd[4059]: PPP session is 19376
Wed Oct 14 17:33:17 2015 daemon.warn pppd[4059]: Connected to 00:46:4b:71:8a:a1 via interface eth0.2
Wed Oct 14 17:33:17 2015 daemon.info pppd[4059]: Using interface pppoe-netkeeper
Wed Oct 14 17:33:17 2015 daemon.notice pppd[4059]: Connect: pppoe-netkeeper <--> eth0.2
Wed Oct 14 17:33:17 2015 daemon.info pppd[4059]: Remote message: 24;User(1862674240) not found(database authen forbiden), Request Deny by hb.cn
Wed Oct 14 17:33:17 2015 daemon.err pppd[4059]: PAP authentication failed
Wed Oct 14 17:33:17 2015 daemon.notice pppd[4059]: Connection terminated.
Wed Oct 14 17:33:17 2015 daemon.info pppd[4059]: Exit.

miao1007, 2015-10-14:
```
Remote message: 24;User(1862674240) not found(database authen forbiden), Request Deny by hb.cn
```

这感觉应该算错账号了？试试[这里](https://sunflyer.cn/archives/8)的算号器来调试下？

@sunflyer 可否加一个基于特定timestamp的算号调试功能呢，输入timestamp来计算出真正账号从而配合嵌入式设备上debug的时间来调试？

sunflyer, 2015-10-14:
需要的话留个邮箱给我，我晚上发给你一份。

Vicrime, 2015-10-14:
@miao1007 账号木有错。
10次中大概只有1次拨号成功。

Vicrime, 2015-10-14:
@miao1007  那两个Warning 是否可以直接无视？？？

sunflyer, 2015-10-14:
@miao1007 已发送到你的gmail

sunflyer, 2015-10-14:
@miao1007 。。。。被退回来了，说文件太大（我勒个去总共就9K的附件也算大）。。。。。。我传百度网盘。http://pan.baidu.com/s/1qW0381e , http://pan.baidu.com/s/1qW0381e

miao1007, 2015-10-14:
@sunflyer thanks!其他同学也用可以用这个进行调试。

@Vicrime  这两个warining没有见过，貌似是在运行我的openwrt的sh脚本的警告，因为我是按照重庆配置的。你可以自行用鼠标配置防火墙。