miao1007/Openwrt-NetKeeper #47: 编译make的时候出错求协助

xxzbg, 2015-10-19:
admini@admini-virtual-machine:~$ cd Openwrt-NetKeeper/src/
admini@admini-virtual-machine:~/Openwrt-NetKeeper/src$ make all
make: *** 没有规则可以创建目标“all”。 停止。
admini@admini-virtual-machine:~/Openwrt-NetKeeper/src$ 

//附上makefile
//是MT7620N  

#download latest GCC in https://github.com/miao1007/Openwrt-NetKeeper/wiki#2-%E5%A6%82%E4%BD%95%E4%B8%8B%E8%BD%BDgcc
#This is a demo for MTK7620A

#TODO:填写gcc中bin与include所在的目录（根目录）
LOCATION=/home/admini/OpenWrt-Toolchain-ramips-for-mipsel_24kec+dsp-gcc-4.8-linaro_uClibc-0.9.33.2/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_uClibc-0.9.33.2
#TODO:请注意不同型号(mips/mipsel)路由器此处gcc文件名可能略有不同
CC=$(LOCATION)/bin/mipsel-openwrt-linux-gcc
#TODO:可能需要修改pppd的版本，取决于路由器的/usr/lib/pppd/，目前主流的14.09是2.4.7
PPPD_VER=2.4.5
CFLAGS=-Os -Werror -I$(LOCATION)/include -fPIC -DPPPOE_VER='"${PPPD_VER}"'

# TODO: change RADIUS
# 
# ChongQing Netkeeper: cqxinliradius002
# ChongQing Netkeeper(0094): xianxinli1radius
# WuHan E xin: hubtxinli01
# Hangzhou(Tested on HDU): singlenet01
# NanChangV18: nanchang3.0
# NanChangV12~V17: radius
# NanChangV29: nanchang3.0
# NanChangV32: jiangxi4.0
# QingHai: qhtel@xiaoyuanyi
# hebei: hebeicncxinli002
# ShanDong Mobile : shandongmobile13

# TODO: Change PREFIX1
# 
# NanChangV32: '1'
# Others: '\n'

all:chongqing_sxplugin wuhan_sxplugin hangzhou_sxplugin nanchang_sxplugin qinghai_sxplugin shandongmobile_sxplugin hebei_sxplugin shandongmobile_sxplugin hainan_sxplugin wo_sxplugin
	@find *.so
	@echo "拨号组件编译成功，注意修改confnetwork.sh的sxplugin名称为本省的名称，以及拨号帐号与密码!，然后运行make upload"

#you can also upload only your province's so file
upload:
	@echo "正在上传拨号组件，请输入路由器ssh密码"
	scp *.so root@192.168.1.1:/usr/lib/pppd/${PPPD_VER}/
	@echo "正在上传网络配置脚本，请输入路由器ssh密码"
	scp confnetwork.sh root@192.168.1.1:/tmp/
	@echo "正在登录路由器，请在输入路由器ssh密码后，运行sh /tmp/confnetwork.sh"
	ssh root@192.168.1.1

chongqing_sxplugin:
	@$(CC) $(CFLAGS) sxplugin.c -DRADIUS='"xianxinli1radius"'  -DPREFIX1="'\n'" -shared -o $@.so

wuhan_sxplugin:
	@$(CC) $(CFLAGS) sxplugin.c -DRADIUS='"hubtxinli01"' -DPREFIX1="'\n'" -shared -o $@.so

hangzhou_sxplugin:
	@$(CC) $(CFLAGS) sxplugin.c -DRADIUS='"singlenet01"' -DPREFIX1="'\n'" -shared -o $@.so

nanchang_sxplugin:
	@$(CC) $(CFLAGS) sxplugin.c -DRADIUS='"jiangxi4.0"' -DPREFIX1="'1'" -shared -o $@.so

qinghai_sxplugin:
	@$(CC) $(CFLAGS) sxplugin.c -DRADIUS='"qhtel@xiaoyuanyi"' -DPREFIX1="'\n'" -shared -o $@.so

shandongmobile_sxplugin:
	@$(CC) $(CFLAGS) sxplugin.c -DRADIUS='"shandongmobile13"' -DPREFIX1="'\n'" -shared -o $@.so

hebei_sxplugin:
	@$(CC) $(CFLAGS) sxplugin.c -DRADIUS='"hebeicncxinli002"' -DPREFIX1="'\n'" -shared -o $@.so

hainan_sxplugin:
	@$(CC) $(CFLAGS) hainan_sxplugin.c -shared -o $@.so

wo_sxolugin:
	@$(CC) $(CFLAGS) wo_sxplugin.c -shared -o $@.so

#clean *.so 
clean:
	rm *.so

xxzbg, 2015-10-19:
系统从15换到14也不行   使用的是Ubuntu kylin 14.04 LTS

lycycycycyc, 2015-10-19:
建议在history里面找以前的makefile，这个新版makefile用不了，我也没找到问题所在

lycycycycyc, 2015-10-19:
建议在history里面找以前的代码用，这个新版makefile用不了，我也没找到问题所在

xxzbg, 2015-10-19:
我用了以前的代码和别人编译成功的makefile都不行，直接在make行就错误了，是不是少安了什么东西。我都开始怀疑。但是openwrt的makeconfig进的去，不知道是什么原因，

miao1007, 2015-10-19:
改好了

xxzbg, 2015-10-19:
那我明天尝试一下，实在是受不了了，还有，大大，只要能进去openwrt的makeconfig说明环境是没问题的对吧？

xxzbg, 2015-10-20:
搞定了  make的时候过去了  也生成了.so 文件  但是有个警告提示
//mipsel-openwrt-linux-uclibc-gcc.bin: warning: environment variable 'STAGING_DIR' not defined
这个会有什么影响吗？

lycycycycyc, 2015-10-20:
这是正常情况

prettys, 2016-09-10:
@cuyl  能解释下原因吗？

cuyl, 2016-09-11:
字面意思，环境变量'STAGING_DIR'未定义

prettys, 2016-09-11:
能说下怎么解决吗？

cuyl, 2016-09-11:
配一下环境变量就解决了，请善用搜索引擎

closed 2016-09-11 by miao1007