miao1007/Openwrt-NetKeeper #48: 南昌，两三天前开始，每21分钟断一次

lknife, 2015-10-23:
星空极速直连拨号就没问题。
之前都正常拨号，但是从两三天前开始，每21分钟断一次。

用的是release里编译好的：
https://github.com/miao1007/Openwrt-NetKeeper/releases/download/1.0/sxplugin_mtk7620a_14.04.zip

路由器固件也重装过，完全按照之前的没问题的方法操作，刷好固件以后没有进行其他操作，直接上传拨号插件然后拨号。

路由器型号  极路由2
固件版本	    PandoraBox R2 14.09 / LuCI Trunk (0.12+svn-r1024)
内核版本 	    3.14.44

系统日志：
Fri Oct 23 15:01:16 2015 daemon.info pppd[3668]: sxplugin : passwd loaded
Fri Oct 23 15:01:16 2015 daemon.notice pppd[3668]: pppd 2.4.7 started by root, uid 0
Fri Oct 23 15:01:16 2015 daemon.info pppd[3668]: PPP session is 18643
Fri Oct 23 15:01:16 2015 daemon.warn pppd[3668]: Connected to e0:24:7f:b7:77:36 via interface eth0.2
Fri Oct 23 15:01:16 2015 daemon.info pppd[3668]: Using interface pppoe-wan
Fri Oct 23 15:01:16 2015 daemon.notice pppd[3668]: Connect: pppoe-wan <--> eth0.2
Fri Oct 23 15:01:19 2015 daemon.info pppd[3668]: syncppp not active
Fri Oct 23 15:01:19 2015 daemon.info pppd[3668]: CHAP authentication succeeded: Authentication success,Welcome!
Fri Oct 23 15:01:19 2015 daemon.notice pppd[3668]: CHAP authentication succeeded
Fri Oct 23 15:01:19 2015 daemon.notice pppd[3668]: peer from calling number E0:24:7F:B7:77:36 authorized
Fri Oct 23 15:01:19 2015 daemon.notice pppd[3668]: local  LL address fe80::94ed:44a7:36b1:597c
Fri Oct 23 15:01:19 2015 daemon.notice pppd[3668]: remote LL address fe80::e224:7fff:feb7:7736
Fri Oct 23 15:01:19 2015 daemon.notice pppd[3668]: local  IP address 100.74.50.227
Fri Oct 23 15:01:19 2015 daemon.notice pppd[3668]: remote IP address 100.74.32.1
Fri Oct 23 15:01:19 2015 daemon.notice pppd[3668]: primary   DNS address 202.101.224.69
Fri Oct 23 15:01:19 2015 daemon.notice pppd[3668]: secondary DNS address 202.101.226.69
Fri Oct 23 15:01:19 2015 daemon.notice netifd: Network device 'pppoe-wan' link is up
Fri Oct 23 15:01:19 2015 daemon.notice netifd: Interface 'wan' is now up
Fri Oct 23 15:01:19 2015 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Fri Oct 23 15:01:21 2015 daemon.info dnsmasq[3322]: reading /tmp/resolv.conf.auto
Fri Oct 23 15:01:21 2015 daemon.info dnsmasq[3322]: using local addresses only for domain lan
Fri Oct 23 15:01:21 2015 daemon.info dnsmasq[3322]: using nameserver 202.101.224.69#53
Fri Oct 23 15:01:21 2015 daemon.info dnsmasq[3322]: using nameserver 202.101.226.69#53
Fri Oct 23 15:01:21 2015 daemon.notice miniupnpd[3365]: shutting down MiniUPnPd
Fri Oct 23 15:01:21 2015 daemon.err miniupnpd[3365]: sendto(udp_shutdown=-1): Bad file descriptor
Fri Oct 23 15:01:21 2015 daemon.err miniupnpd[3365]: sendto(udp_shutdown=-1): Bad file descriptor
Fri Oct 23 15:01:21 2015 daemon.err miniupnpd[3365]: sendto(udp_shutdown=-1): Bad file descriptor
Fri Oct 23 15:01:21 2015 daemon.err miniupnpd[3365]: sendto(udp_shutdown=-1): Bad file descriptor
Fri Oct 23 15:01:21 2015 daemon.err miniupnpd[3365]: sendto(udp_shutdown=-1): Bad file descriptor
Fri Oct 23 15:01:21 2015 daemon.err miniupnpd[3365]: sendto(udp_shutdown=-1): Bad file descriptor
Fri Oct 23 15:01:21 2015 daemon.err miniupnpd[3365]: sendto(udp_shutdown=-1): Bad file descriptor
Fri Oct 23 15:01:21 2015 daemon.err miniupnpd[3365]: sendto(udp_shutdown=-1): Bad file descriptor
Fri Oct 23 15:01:21 2015 daemon.err miniupnpd[3365]: sendto(udp_shutdown=-1): Bad file descriptor
Fri Oct 23 15:01:21 2015 daemon.err miniupnpd[3365]: sendto(udp_shutdown=-1): Bad file descriptor
Fri Oct 23 15:01:21 2015 daemon.err miniupnpd[3365]: sendto(udp_shutdown=-1): Bad file descriptor
Fri Oct 23 15:01:21 2015 daemon.err miniupnpd[3365]: Failed to broadcast good-bye notifications
Fri Oct 23 15:01:22 2015 daemon.notice miniupnpd[4023]: HTTP listening on port 5000
Fri Oct 23 15:01:22 2015 daemon.warn miniupnpd[4023]: no HTTP IPv6 address, disabling IPv6
Fri Oct 23 15:01:22 2015 daemon.notice miniupnpd[4023]: Listening for NAT-PMP/PCP traffic on port 5351
Fri Oct 23 15:01:24 2015 daemon.info dnsmasq-dhcp[3322]: DHCPDISCOVER(br-lan) b0:d5:9d:2d:e4:61 
Fri Oct 23 15:01:24 2015 daemon.info dnsmasq-dhcp[3322]: DHCPOFFER(br-lan) 192.168.1.189 b0:d5:9d:2d:e4:61 
Fri Oct 23 15:01:24 2015 daemon.info dnsmasq-dhcp[3322]: DHCPREQUEST(br-lan) 192.168.1.189 b0:d5:9d:2d:e4:61 
Fri Oct 23 15:01:24 2015 daemon.info dnsmasq-dhcp[3322]: DHCPACK(br-lan) 192.168.1.189 b0:d5:9d:2d:e4:61 XL_MINER_9882
Fri Oct 23 15:01:58 2015 authpriv.info dropbear[3504]: Exit (root): Exited normally
Fri Oct 23 15:01:58 2015 authpriv.info dropbear[3375]: Exit (root): Exited normally
Fri Oct 23 15:01:58 2015 cron.err crond[2251]: time disparity of 196834 minutes detected
Fri Oct 23 15:02:26 2015 daemon.warn odhcpd[1671]: DHCPV6 SOLICIT IA_NA from 000100011d7a11a8f0def1bfbe9d on br-lan: no addresses available 
Fri Oct 23 15:02:27 2015 daemon.warn odhcpd[1671]: DHCPV6 SOLICIT IA_NA from 000100011d7a11a8f0def1bfbe9d on br-lan: no addresses available 
Fri Oct 23 15:02:27 2015 daemon.warn odhcpd[1671]: DHCPV6 SOLICIT IA_NA from 000100011d7a11a8f0def1bfbe9d on br-lan: no addresses available 
Fri Oct 23 15:02:27 2015 daemon.warn odhcpd[1671]: DHCPV6 SOLICIT IA_NA from 000100011d7a11a8f0def1bfbe9d on br-lan: no addresses available 
Fri Oct 23 15:02:27 2015 daemon.warn odhcpd[1671]: DHCPV6 SOLICIT IA_NA from 000100011d7a11a8f0def1bfbe9d on br-lan: no addresses available 
Fri Oct 23 15:02:27 2015 daemon.warn odhcpd[1671]: DHCPV6 SOLICIT IA_NA from 000100011d7a11a8f0def1bfbe9d on br-lan: no addresses available 
Fri Oct 23 15:02:27 2015 daemon.warn odhcpd[1671]: DHCPV6 SOLICIT IA_NA from 000100011d7a11a8f0def1bfbe9d on br-lan: no addresses available 
Fri Oct 23 15:02:27 2015 daemon.warn odhcpd[1671]: DHCPV6 SOLICIT IA_NA from 000100011d7a11a8f0def1bfbe9d on br-lan: no addresses available 
Fri Oct 23 15:03:32 2015 daemon.info dnsmasq-dhcp[3322]: DHCPRELEASE(br-lan) 192.168.1.189 b0:d5:9d:2d:e4:61 
Fri Oct 23 15:03:34 2015 kern.info kernel: [  434.440000] eth0: port 4 link down
Fri Oct 23 15:03:34 2015 user.notice switch: ACTION=DOWN  PORT=4  SPEED=0 DUPLEX=0
[... 103 lines cut ...]
Fri Oct 23 15:22:50 2015 daemon.notice pppd[8749]: remote IP address 100.74.0.1
Fri Oct 23 15:22:50 2015 daemon.notice pppd[8749]: primary   DNS address 202.101.224.69
Fri Oct 23 15:22:50 2015 daemon.notice pppd[8749]: secondary DNS address 202.101.226.69
Fri Oct 23 15:22:50 2015 daemon.notice netifd: Network device 'pppoe-wan' link is up
Fri Oct 23 15:22:50 2015 daemon.notice netifd: Interface 'wan' is now up
Fri Oct 23 15:22:51 2015 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Fri Oct 23 15:22:52 2015 daemon.info dnsmasq[3322]: reading /tmp/resolv.conf.auto
Fri Oct 23 15:22:52 2015 daemon.info dnsmasq[3322]: using local addresses only for domain lan
Fri Oct 23 15:22:52 2015 daemon.info dnsmasq[3322]: using nameserver 202.101.224.69#53
Fri Oct 23 15:22:52 2015 daemon.info dnsmasq[3322]: using nameserver 202.101.226.69#53
Fri Oct 23 15:22:53 2015 daemon.notice miniupnpd[4023]: shutting down MiniUPnPd
Fri Oct 23 15:22:53 2015 daemon.err miniupnpd[4023]: sendto(udp_shutdown=-1): Bad file descriptor
Fri Oct 23 15:22:53 2015 daemon.err miniupnpd[4023]: sendto(udp_shutdown=-1): Bad file descriptor
Fri Oct 23 15:22:53 2015 daemon.err miniupnpd[4023]: sendto(udp_shutdown=-1): Bad file descriptor
Fri Oct 23 15:22:53 2015 daemon.err miniupnpd[4023]: sendto(udp_shutdown=-1): Bad file descriptor
Fri Oct 23 15:22:53 2015 daemon.err miniupnpd[4023]: sendto(udp_shutdown=-1): Bad file descriptor
Fri Oct 23 15:22:53 2015 daemon.err miniupnpd[4023]: sendto(udp_shutdown=-1): Bad file descriptor
Fri Oct 23 15:22:53 2015 daemon.err miniupnpd[4023]: sendto(udp_shutdown=-1): Bad file descriptor
Fri Oct 23 15:22:53 2015 daemon.err miniupnpd[4023]: sendto(udp_shutdown=-1): Bad file descriptor
Fri Oct 23 15:22:53 2015 daemon.err miniupnpd[4023]: sendto(udp_shutdown=-1): Bad file descriptor
Fri Oct 23 15:22:53 2015 daemon.err miniupnpd[4023]: sendto(udp_shutdown=-1): Bad file descriptor
Fri Oct 23 15:22:54 2015 daemon.notice miniupnpd[9110]: HTTP listening on port 5000
Fri Oct 23 15:22:54 2015 daemon.warn miniupnpd[9110]: no HTTP IPv6 address, disabling IPv6
Fri Oct 23 15:22:54 2015 daemon.notice miniupnpd[9110]: Listening for NAT-PMP/PCP traffic on port 5351
Fri Oct 23 15:22:55 2015 daemon.err miniupnpd[9110]: upnp_event_recv: recv(): Connection reset by peer

miao1007, 2015-10-23:
好像是开心跳了，如果不是心跳问题，那就是固件的问题

lknife, 2015-10-23:
这个固件我一直在用

lknife, 2015-10-23:
这个固件我一直在用，前面一个月都是正常的。
这个问题能解决吗？

sunflyer, 2015-10-23:
DaoNet目录下面是江西的OP心跳插件，编译了试试

smounives, 2015-10-23:
@sunflyer 怎么编译的啊？直接make报错了
mips-openwrt-linux-gcc  -o main.o main.c -g -Wall -Wimplicit-function-declaration -I /home/ubuntu/workspace/OpenWrt-SDK/staging_dir/include/  -c
make: mips-openwrt-linux-gcc: Command not found
make: *** [main.o] Error 127

sunflyer, 2015-10-24:
你的gcc位置错了，另外，那个是手误

smounives, 2015-10-24:
Makefile里的CC = mips-openwrt-linux-gcc 要自己改吗？

sunflyer, 2015-10-25:
你需要把mbedtls这个库包含进来。（丢到include里面）

Whatnamegood, 2015-10-26:
mbedtls首先是要编译吗？
如果要编译的话是直接make install吗？
这里的”include“ 是指/usr下的  还是sdk下的？

sunflyer, 2015-10-26:
sdk

Whatnamegood, 2015-10-26:
这样。那么请问要这么编译进sdk？

Whatnamegood, 2015-10-26:
要写Makefile文件吗？

Whatnamegood, 2015-10-26:
@sunflyer

smounives, 2015-10-26:
完全搞不定啊QAQ @sunflyer 可不可以帮忙编译一份mt7620a的心跳插件

Whatnamegood, 2015-10-26:
@smounives 不用这样吧     还要现成的

closed 2016-03-25 by lknife