miao1007/Openwrt-NetKeeper #49: 编译心跳插件出错了

CHeukFung, 2015-10-24:
编译daonet里面打插件失败，错误信息是：
/home/cheukfung/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/bin/mips-openwrt-linux-gcc  -o DaoNet main.o frame.o netutils.o netkeeper.o /home/cheukfung/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/lib/libmbedcrypto.a -g -Wall -Wimplicit-function-declaration -I /home/cheukfung/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/include/ 
mips-openwrt-linux-gcc: error: /home/cheukfung/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/lib/libmbedcrypto.a: No such file or directory
make: *** [DaoNet] 错误 1
是库的原因吗，希望能解决，谢谢。

xseven007, 2015-10-25:
这个是mbed TLS 的library文件，并不是openssl的。你可以考虑编译一个mbedtls。

CheukFung, 2015-10-27:
谢谢你的回复。我用openwrt编译的SDK编译了mbedtls，然后成功编译出DaoNet，可是在openwrt上面不能运行，只显示not found，我是否遗漏了什么？

xseven007, 2015-10-30:
编译的时候是否使用了动态库？

CheukFung, 2015-10-31:
应该没有，我不是很懂。能详细点吗？或者你可以编译一个ar71的心跳插件给我吗？谢谢。。现在20分钟必断网

Whatnamegood, 2015-11-01:
@CheukFung 你是怎么编译mbedtls的？  有的是哪个版本哦？

CheukFung, 2015-11-02:
@Whatnamegood 用下载的gcc编译mbedtls，然后放到lib文件夹

Whatnamegood, 2015-11-03:
@CheukFung 怎么用下载GCC编译呢？直接设置环境变量吗？

Whatnamegood, 2015-11-03:
[root@localhost DaoNet]# make
mips-openwrt-linux-gcc  -o DaoNet main.o frame.o netutils.o netkeeper.o /home/li/Open/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib/libmbedcrypto.a -g -Wall -Wimplicit-function-declaration -I /home/li/Open/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include/
/home/li/Open/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/bin/../lib/gcc/mips-openwrt-linux-uclibc/4.8.3/../../../../mips-openwrt-linux-uclibc/bin/ld: /home/li/Open/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib/libmbedcrypto.a(aes.o): Relocations in generic ELF (EM: 62)
/home/li/Open/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/bin/../lib/gcc/mips-openwrt-linux-uclibc/4.8.3/../../../../mips-openwrt-linux-uclibc/bin/ld: /home/li/Open/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib/libmbedcrypto.a(aes.o): Relocations in generic ELF (EM: 62)
/home/li/Open/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib/libmbedcrypto.a: error adding symbols: File in wrong format
collect2: error: ld returned 1 exit status
make: *** [DaoNet] 错误 1
[root@localhost DaoNet]#

Whatnamegood, 2015-11-03:
这里是说libmbedcrypto.a 文件类型错误    那也就是说编译时没有用到交叉编译的gcc

Whatnamegood, 2015-11-07:
@CheukFung 你好

CheukFung, 2015-11-07:
@Whatnamegood 修改mbedtls下library下的makefile，添加CC=xxxx，xxxx为你下载的gcc路径，然后make就可以了。

Whatnamegood, 2015-11-10:
@CheukFung 谢谢你   我已经成功编译了   可是要怎么使用呢？
 root@OpenWrt:~# DaoNet
WARRNING: Support for ShanXi Netkeeper temporary.
Missing arguments.
Usage: daonet [OPTIONS]
Options:
  --username,  -u:      Username.
  --password,  -p:      Password.
  --ipaddress, -i:      IP Address.
  --version,   -v:      Version.
  --aeskey,    -k:      Aes key.
  --interval,  -I:      Interval.
  --target,    -t:      Target.
  --port,      -P:      Port.
  --help,      -h:      Help.
root@OpenWrt:~#

这里是要填写其配置文件？  还是要在编译是把账号信息写入到源码在编译？  或者接选项-u -p 然后接参数执行呢？

sunflyer, 2015-11-10:
直接带参数执行。注意ip地址是你拨号获得的ip

Whatnamegood, 2015-11-10:
能给个实例吗？

Whatnamegood, 2015-11-10:
@sunflyer

sunflyer, 2015-11-10:
DaoNet -u <宽带账号> -p <密码> -i <你拨号获得的IP地址> -v 2.5.0016v32 -k wanglibinwanglei -I 120 -t 117.21.209.186 -P 443

Whatnamegood, 2015-11-10:
谢谢！   这里的-k 的key 是固定吗   是不是各个地区的都不同？

Whatnamegood, 2015-11-10:
王立彬王雷？

Whatnamegood, 2015-11-10:
root@OpenWrt:~# ./daonet.sh
root@OpenWrt:~# WARRNING: Support for ShanXi Netkeeper temporary.
INFO: Send packet succeed.
ERROR: Wait for packet timeout.
INFO: Wait 120 seconds.

这样是正常的吗？ ERROR: Wait for packet timeout.

sunflyer, 2015-11-10:
没事，江西没有回复包，正常

smounives, 2015-11-19:
@sunflyer 湖北E信可以使用这个心跳插件吗？需不需要改什么？

sunflyer, 2015-11-19:
湖北不能

smounives, 2015-11-19:
QAQ 难道只能直接插网线用您的SimpleNetkeeper了吗

sunflyer, 2015-11-19:
最快下周出op支持吧，看情况

smounives, 2015-11-19:
多谢！

theflash2017, 2017-04-05:
您好，请问您有空吗？我想编译一个daonet但是并不会，您能指导一下吗？

closed 2023-03-22 by Cheukfung