miao1007/Openwrt-NetKeeper #53: 路由器时间同步问题

gancaocao, 2015-11-02:
我的路由器没有硬件时钟，无法保存时间，但是netkeeper认证需要时间是正确的 ，请问，怎么才能让时间自动同步呢

lycycycycyc, 2015-11-02:
那就只能用工具设置路由了

lycycycycyc, 2015-11-02:
http://www.nktools.win/
这个也许对你有用

miao1007, 2015-11-02:
自己找学校的ntp即可

~~还有不要发那种用E语言写的闭源垃圾软件广告~~

lycycycycyc, 2015-11-02:
只是为了解决问题，没必要带那么大偏见

lycycycycyc, 2015-11-02:
你看看这些开源代码，还不是被人一声不吭拿去赚钱，不要盲目迷信开源好不好

0x5e, 2015-11-02:
不需要时间正确的,时间戳用1970年的用着也没啥问题..只是用作随机数

gancaocao, 2015-11-03:
有的路由器可以时间不同步，我试过另一个时间不对也能连上，但是我这个就提示这个：
Remote message: Pin code error,please check your time##Pin code error,please check your time

lycycycycyc, 2015-11-28:
这是我写的湖北同步时间的办法，uci部分暂时还没测试其他省应该都有对应的网址吧？
timesyn.sh
```
#!/bin/sh
uci set network.timesyn='interface'
uci set network.timesyn.proto='pppoe'
uci set network.timesyn.username='wuhan'
uci set network.timesyn.password='wuhan'
uci set network.timesyn.metric='0'
uci commit time
ifup time
date -u "`curl -sI kd.wuhan.net.cn | awk '/Date:/{gsub(/Jan/,"01",$4);gsub(/Feb/,"02",$4);gsub(/Mar/,"03",$4);gsub(/Apr/,"04",$4);gsub(/May/,"05",$4);gsub(/Jun/,"06",$4);gsub(/Jul/,"07",$4);gsub(/Aug/,"08",$4);gsub(/Sep/,"09",$4);gsub(/Oct/,"10",$4);gsub(/Nov/,"11",$4);gsub(/Dec/,"11",$4);print $5"-"$4"-"$3" "$6;}'`"

```

a1379478560, 2016-09-22:
@cuyl 这位哥们帮我大忙了

closed 2019-03-03 by gancaocao