miao1007/Openwrt-NetKeeper #55: 请问RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7是什么错误？

kuri-leo, 2015-11-02:
出现了以下错误。。。路由器是MW305W

Wed Oct  1 14:51:48 2014 daemon.notice netifd: Interface 'NetKeeper' is enabled
Wed Oct  1 14:51:48 2014 daemon.notice netifd: Interface 'NetKeeper' has link connectivity 
Wed Oct  1 14:51:48 2014 daemon.notice netifd: Interface 'NetKeeper' is setting up now
Wed Oct  1 14:51:48 2014 daemon.notice netifd: Interface 'NetKeeper' is now down
Wed Oct  1 14:51:48 2014 daemon.notice netifd: Interface 'NetKeeper' is setting up now
Wed Oct  1 14:51:48 2014 daemon.info pppd[1472]: Plugin rp-pppoe.so loaded.
Wed Oct  1 14:51:48 2014 daemon.info pppd[1472]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Wed Oct  1 14:51:48 2014 daemon.notice netifd: NetKeeper (1472): /usr/sbin/pppd: '/usr/lib/pppd/2.4.7/sxplugin.so' is not an ELF executable for MIPS
Wed Oct  1 14:51:48 2014 daemon.notice netifd: NetKeeper (1472): /usr/sbin/pppd: File not found
Wed Oct  1 14:51:48 2014 daemon.err pppd[1472]: File not found
Wed Oct  1 14:51:48 2014 daemon.notice netifd: NetKeeper (1472): /usr/sbin/pppd: Couldn't load plugin sxplugin.so
Wed Oct  1 14:51:48 2014 daemon.err pppd[1472]: Couldn't load plugin sxplugin.so
Wed Oct  1 14:51:48 2014 daemon.notice netifd: Interface 'NetKeeper' is now down

miao1007, 2015-11-02:
编译时把makefile的2.4.7改成2.4.8即可

kuri-leo, 2015-11-03:
@miao1007 学长麻烦了，我改成了2.4.8还是之前那个问题，并且他说插件是2.4.8的，不支持2.4.7 
Wed Oct  1 23:00:06 2014 daemon.notice netifd: Interface 'NetKeeper' has link connectivity loss
Wed Oct  1 23:00:06 2014 daemon.notice netifd: Interface 'NetKeeper' has link connectivity 
Tue Nov  3 21:46:14 2015 daemon.notice netifd: Interface 'NetKeeper' is setting up now
Tue Nov  3 21:46:14 2015 daemon.info pppd[1581]: Plugin rp-pppoe.so loaded.
Tue Nov  3 21:46:14 2015 daemon.info pppd[1581]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Tue Nov  3 21:46:14 2015 daemon.err pppd[1581]: Plugin sxplugin.so is for pppd version 2.4.8, this is 2.4.7
Tue Nov  3 21:46:14 2015 daemon.notice netifd: NetKeeper (1581): /usr/sbin/pppd: Plugin sxplugin.so is for pppd version 2.4.8, this is 2.4.7
Tue Nov  3 21:46:14 2015 daemon.notice netifd: Interface 'NetKeeper' is now down

sunflyer, 2015-11-03:
第一次错应该是编译格式错了，拥错了编译工具导致的，第二次就是真的版本错误，应该是2.4.7

kuri-leo, 2015-11-03:
@miao1007 谢谢学长帮忙！问题已经解决了！
原因貌似是Ubuntu/GCC有问题造成原来编译的插件有问题，重新开虚拟机编译了一个就解决了

kuri-leo, 2015-11-03:
@sunflyer 感谢，貌似是工具链出问题了，重新编译了一次就好了，谢谢关心！

closed 2015-11-03 by kuri-leo