miao1007/Openwrt-NetKeeper #56: 能否支持广东广州的Young网络客户端

matthew728960, 2015-11-05:
我把客户端上传了，请帮忙，谢谢！（打开时请将png改为zip）
https://cloud.githubusercontent.com/assets/8460881/10959876/e80fb610-83b8-11e5-9339-c17c92895885.png
如果可以说一下怎么手动上传就更好了，而不是make upload，毕竟不是每个人都有Ubuntu。感谢

sunflyer, 2015-11-05:
这个似乎并不是Netkeeper系列，

matthew728960, 2015-11-05:
这样啊，那你知不知道这种有什么方法登陆呢，谢谢！！

milkyfox, 2015-11-19:
@sunflyer 
请问这个是否为 Netkeeper 系列
http://www.wandoujia.com/apps/com.xinli.portalclient

sunflyer, 2015-11-19:
这个是，你是哪个地区的

milkyfox, 2015-11-20:
@sunflyer 山东，电信校园网。连上 WiFi 会弹出这个网址来下载客户端
http://219.146.3.54:8080
这个可以添加支持吗？

sunflyer, 2015-11-20:
山东电信的radius我之前解了一次似乎和陕西的是一样的，你试下看

milkyfox, 2015-11-20:
@sunflyer 谢谢，我试试

milkyfox, 2015-11-20:
@sunflyer 请问 radius 是 sh_do@xiaoyuan01 吗？

sh_xi@xiaoyuan01 >> sh_do@xiaoyuan01

sunflyer, 2015-11-20:
不是，就直接和陕西一样的，不用改

milkyfox, 2015-11-20:
@sunflyer 似乎不是 sh_xi@xiaoyuan01
电脑上用天翼客户端拨号和用 SimpleNetkeeper_Win 算出来的账号不一样
[image]
手机号前的两位不一样

sunflyer, 2015-11-20:
shdOx!a0yuan01lz

你试下是不是这个？

milkyfox, 2015-11-21:
@sunflyer 手机号前两位还是不一样...

sunflyer, 2015-11-21:
你直接复制过去，注意不要多了空格。最好是用测试功能的自定义radius试试
前缀\r\n

milkyfox, 2015-11-21:
@sunflyer 
[image]
[image]
radius 没有空格

sunflyer, 2015-11-21:
鸭血，这尼玛什么情况。。。dump出来的确是这个啊，你用的3.7.3么

milkyfox, 2015-11-21:
@sunflyer 是 3.7.3

sunflyer, 2015-11-21:
然而我从3.7.3拿到的就是这个。。。回去再看看吧。。。

milkyfox, 2015-11-21:
@sunflyer 呃，有两个 3.7.3 版本，10 月 17 号那个我输上密码显示密码错误
[image]

这个是 8 月 17 号的版本，可以登录
http://jianguoyun.com/p/DXus_kIQwobeBRi7kQ8
可否再试试

sunflyer, 2015-11-21:
我开始是从10月哪个版本抓的。。。等下试试8月的

milkyfox, 2015-11-21:
@sunflyer 谢谢！

sunflyer, 2015-11-21:
radius = 560Ox!a0yuanOlIz
怎么都是些莫名其妙的玩意儿。。。。。

milkyfox, 2015-11-21:
@sunflyer 
[image]
 谢谢！！

sunflyer, 2015-11-21:
那么这个radius是对的咯

milkyfox, 2015-11-22:
@sunflyer 嗯，是对的

closed 2015-11-22 by miao1007