miao1007/Openwrt-NetKeeper #58: 河北地区拨号问题

opUserZangYing, 2015-11-17:
编译正常，上传正常
pppd是2.4.5已改
但是无法联网，有什么办法看原因吗？
是否需要把 \n 改成 \r\n？

sunflyer, 2015-11-17:
把系统日志发上来

opUserZangYing, 2015-11-17:
我错了……make的时候有一个STAGING-DIR未设置……还有pppd是直接在MAKEfile里改吗？

sunflyer, 2015-11-17:
不是。。。我是说op的系统日志
另外ppp版本因该是makefile或者c文件里面，有的话就改咯

opUserZangYing, 2015-11-18:
Nov 17 18:25:03 OpenWrt kern.debug kernel: [    0.000000] env[1]: initrd_start=0x00000000
Nov 17 18:25:03 OpenWrt kern.debug kernel: [    0.000000] env[2]: initrd_size=0x0
Nov 17 18:25:03 OpenWrt kern.debug kernel: [    0.000000] env[3]: flash_start=0xBF000000
Nov 17 18:25:03 OpenWrt kern.debug kernel: [    0.000000] env[4]: flash_size=0x400000
Nov 17 18:25:03 OpenWrt kern.info kernel: [    0.000000] bootconsole [early0] enabled
Nov 17 18:25:03 OpenWrt kern.info kernel: [    0.000000] CPU revision is: 0001964c (MIPS 24KEc)
Nov 17 18:25:03 OpenWrt kern.info kernel: [    0.000000] Ralink RT3052 id:1 rev:3 running at 384.00 MHz
Nov 17 18:25:03 OpenWrt kern.info kernel: [    0.000000] Determined physical RAM map:
Nov 17 18:25:03 OpenWrt kern.info kernel: [    0.000000]  memory: 02000000 @ 00000000 (usable)
Nov 17 18:25:03 OpenWrt kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Nov 17 18:25:03 OpenWrt kern.warn kernel: [    0.000000] Zone PFN ranges:
Nov 17 18:25:03 OpenWrt kern.warn kernel: [    0.000000]   Normal   0x00000000 -> 0x00002000
Nov 17 18:25:03 OpenWrt kern.warn kernel: [    0.000000] Movable zone start PFN for each node
Nov 17 18:25:03 OpenWrt kern.warn kernel: [    0.000000] Early memory PFN ranges
Nov 17 18:25:03 OpenWrt kern.warn kernel: [    0.000000]     0: 0x00000000 -> 0x00002000
Nov 17 18:25:03 OpenWrt kern.debug kernel: [    0.000000] On node 0 totalpages: 8192
Nov 17 18:25:03 OpenWrt kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat 802547a0, node_mem_map 81000000
Nov 17 18:25:03 OpenWrt kern.debug kernel: [    0.000000]   Normal zone: 64 pages used for memmap
Nov 17 18:25:03 OpenWrt kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Nov 17 18:25:03 OpenWrt kern.debug kernel: [    0.000000]   Normal zone: 8128 pages, LIFO batch:0
Nov 17 18:25:03 OpenWrt kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Nov 17 18:25:03 OpenWrt kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 
Nov 17 18:25:03 OpenWrt kern.warn kernel: [    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 8128
Nov 17 18:25:03 OpenWrt kern.notice kernel: [    0.000000] Kernel command line:  board=DIR-615-D console=ttyS1,57600 mtdparts=physmap-flash.0:192k(u-boot)ro,64k(devdata)ro,64k(devconf)ro,896k(kernel),2880k(rootfs),3776k@0x50000(firmware) rootfstype=squashfs,jffs2
Nov 17 18:25:03 OpenWrt kern.info kernel: [    0.000000] PID hash table entries: 128 (order: -3, 512 bytes)
Nov 17 18:25:03 OpenWrt kern.info kernel: [    0.000000] Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
Nov 17 18:25:03 OpenWrt kern.info kernel: [    0.360000] msgmni has been set to 57
Nov 17 18:25:03 OpenWrt kern.info kernel: [    0.360000] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 254)
Nov 17 18:25:03 OpenWrt kern.info kernel: [    0.380000] io scheduler noop registered
Nov 17 18:25:03 OpenWrt kern.info kernel: [    0.390000] io scheduler deadline registered (default)
Nov 17 18:25:03 OpenWrt kern.info kernel: [    0.400000] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
Nov 17 18:25:03 OpenWrt kern.info kernel: [    0.410000] serial8250: ttyS0 at MMIO 0x10000500 (irq = 13) is a 16550A
Nov 17 18:25:03 OpenWrt kern.info kernel: [    0.430000] serial8250: ttyS1 at MMIO 0x10000c00 (irq = 20) is a 16550A
Nov 17 18:25:03 OpenWrt kern.notice kernel: [    0.450000] physmap platform flash device: 00800000 at bf000000
Nov 17 18:25:03 OpenWrt kern.info kernel: [    0.460000] physmap-flash.0: Found 1 x16 devices at 0x0 in 16-bit bank. Manufacturer ID 0x000001 Chip ID 0x0022f9
Nov 17 18:25:03 OpenWrt kern.debug kernel: [    0.480000] physmap-flash.0: Found an alias at 0x400000 for the chip at 0x0
Nov 17 18:25:03 OpenWrt kern.info kernel: [    0.480000] Amd/Fujitsu Extended Query Table at 0x0040
Nov 17 18:25:03 OpenWrt kern.info kernel: [    0.490000]   Amd/Fujitsu Extended Query version 1.1.
Nov 17 18:25:03 OpenWrt kern.notice kernel: [    0.500000] number of CFI chips: 1
Nov 17 18:25:03 OpenWrt kern.notice kernel: [    0.510000] 6 cmdlinepart partitions found on MTD device physmap-flash.0
Nov 17 18:25:03 OpenWrt kern.notice kernel: [    0.520000] Creating 6 MTD partitions on "physmap-flash.0":
Nov 17 18:25:03 OpenWrt kern.notice kernel: [    0.540000] 0x000000000000-0x000000030000 : "u-boot"
Nov 17 18:25:03 OpenWrt kern.notice kernel: [    0.550000] 0x000000030000-0x000000040000 : "devdata"
Nov 17 18:25:03 OpenWrt kern.notice kernel: [    0.560000] 0x000000040000-0x000000050000 : "devconf"
Nov 17 18:25:03 OpenWrt kern.notice kernel: [    0.570000] 0x000000050000-0x000000130000 : "kernel"
Nov 17 18:25:03 OpenWrt kern.notice kernel: [    0.590000] 0x000000130000-0x000000400000 : "rootfs"
Nov 17 18:25:03 OpenWrt kern.notice kernel: [    0.600000] mtd: partition "rootfs" set to be root filesystem
Nov 17 18:25:03 OpenWrt kern.info kernel: [    0.610000] mtd: partition "rootfs_data" created automatically, ofs=2F0000, len=110000 
Nov 17 18:25:03 OpenWrt kern.notice kernel: [    0.630000] 0x0000002f0000-0x000000400000 : "rootfs_data"
Nov 17 18:25:03 OpenWrt kern.notice kernel: [    0.640000] 0x000000050000-0x000000400000 : "firmware"
Nov 17 18:25:03 OpenWrt kern.info kernel: [    0.660000] ramips-wdt ramips-wdt: timeout value must be 0 < timeout <= 33, using 33
Nov 17 18:25:03 OpenWrt kern.info kernel: [    0.680000] TCP cubic registered
[... cut ...]
Nov 18 10:29:14 OpenWrt daemon.info dnsmasq-dhcp[1318]: DHCPDISCOVER(br-lan) 6c:71:d9:0e:a1:7b 
Nov 18 10:29:14 OpenWrt daemon.info dnsmasq-dhcp[1318]: DHCPOFFER(br-lan) 192.168.1.105 6c:71:d9:0e:a1:7b 
Nov 18 10:29:14 OpenWrt daemon.info dnsmasq-dhcp[1318]: DHCPREQUEST(br-lan) 192.168.1.105 6c:71:d9:0e:a1:7b 
Nov 18 10:29:14 OpenWrt daemon.info dnsmasq-dhcp[1318]: DHCPACK(br-lan) 192.168.1.105 6c:71:d9:0e:a1:7b ToBeACracker-PC
Nov 18 10:29:30 OpenWrt daemon.info hostapd: wlan0: STA 6c:71:d9:0e:a1:7b WPA: group key handshake completed (RSN)

sunflyer, 2015-11-18:
你的日志里面没有任何拨号记录。

opUserZangYing, 2015-11-18:
/发呆 我一会再查查，顺便：你喜欢虎纹鲨鱼嘛/斜眼

opUserZangYing, 2015-11-18:
3Q----刷了个15.05的OP再试就好了----

closed 2015-11-18 by opUserZangYing

hong0980, 2017-10-29:
求个ppp2.4.5.的源码，谢谢！hong0980@qq.com