miao1007/Openwrt-NetKeeper #74: 你好，关于Netkeeper重庆区账号加密算法的一些问题想请教一下

yyx290799684, 2016-04-01:
最近被Netkeeper弄的蓝屏给搞烦死了，于是想自己写一个WPF平台Netkeeper，在看您那张sxplugin.so的流程图，有几地方我没怎么看懂。
一个是TimeByFive后面的Bitwise();得到的TimeByte[4]. 这一步具体是怎么做呢？
还一个是UnixStamp下面的TimeHash/Bitwise();，这一步是对这个时间戳求Hash吗？以及PIN27是什么意思呢？
谢谢

miao1007, 2016-04-01:
1. 首先获得一个Unix时间戳，然后除以5，然后进行一些位运算，最后与帐号混到一起进行md5运算，最后取出前2位
2. pin27是一个数组，它也是通过时间运算出来的

你要是想移植的话，建议直接把这个包装为一个库即可。如果写好了，告诉我，我把它也添加到项目readme中

yyx290799684, 2016-04-01:
我按照sxplugin.c这个文件中的代码对账号进行加密，我看到有一行注释   
 //beforeMD5={time encryption(4)}+{user name(11)}+{RADIUS}+'\0';default length in ChongQing is 31
这里说重庆地区的要用到一个RADIUS，上下文中并没有RADIUS这个变量。
我在不理会这条注释的条件下，用加密所得的账号，在程序中调用cmd，使用rasdial命令进行拨号，但是总是报密码错误。我猜可能有两个问题，一个是那行注释的问题，账号错了；还有可能是由于加密后的账号中有换行符以及空格，我用双引号将其括起来，但是控制台不认这种指令。
那个.so文件在windows下也用不了。搞不定的话也只能用路由器了

miao1007, 2016-04-01:
这个在makefile中，通过 `-D` 参数定义的

yyx290799684, 2016-04-01:
我现在能用程序生成账号了，和在linux里面跑出来的一样 但是还是没法用，感觉是windows的拨号命令不能支持这么复杂的账号格式。我先不弄这个了，谢谢你了

miao1007, 2016-04-01:
你传递的是ASIC编码的字符串，怎么可能不支持呢？据我所知，有人早就用QT写出来windows下的app了，放弃太早了

yyx290799684, 2016-04-01:
嗯。。有道理，我再挣扎挣扎

0x5e, 2016-04-01:
@yyx290799684 命令行传不来换行回车符,可以用win32的api,好像叫RasDial

yyx290799684, 2016-04-02:
@miao1007 您好。我又遇到了一个问题，我看到您在 “http://www.right.com.cn/forum/thread-141979-1-1.html” 这个帖子里，有一张图，我把时间戳改为图中的0x5374cc15。您进行运算后的结果显示为“^p??”，而我用c++和c#跑出来的结果都不是这个值。
[image]

@0x5e 我现在用的就是string arg = string.Format("rasdial {0} \"{1}\" {2}", connectionName, user, pass); 感觉问题还是出在加密运算上

miao1007, 2016-04-02:
试试这个文章，算法都是一样的，在C上这样计算肯定没有问题

要是cpp的话，可以参考 http://www.purpleroc.com/html/507231.html

yyx290799684, 2016-04-02:
已经能计算出结果了，也是正确的，但是用RasDial命令确实无法拨号。只能通过系统底层的方式去拨号。我在网上找到一个，基本是它自己重新去实现了拨号的过程，但是只有极低的概率能够拨上，其余都卡在验证密码的地方（不说密码错，就一直验证）。

sunflyer, 2016-04-06:
直接C语言去写不就好了么 - -  
RasDialA()
你直接命令行传递时没用的，换行符会导致参数错误。