miao1007/Openwrt-NetKeeper #76: Remote message: 29;

nimasi, 2016-05-06:
`
Fri May  6 12:46:10 2016 daemon.info pppd[8049]: PPP session is 9458
Fri May  6 12:46:10 2016 daemon.warn pppd[8049]: Connected to 00:46:4b:71:8a:98 via interface eth1
Fri May  6 12:46:10 2016 daemon.info pppd[8049]: Using interface pppoe-wan
Fri May  6 12:46:10 2016 daemon.notice pppd[8049]: Connect: pppoe-wan <--> eth1
Fri May  6 12:46:10 2016 kern.info kernel: [  438.820000] pppoe-wan: renamed from ppp0
Fri May  6 12:46:10 2016 daemon.info pppd[8049]: syncppp not active
Fri May  6 12:46:10 2016 daemon.info pppd[8049]: Remote message: 29;User(************)'s Authen Attrib(ai-Service-Password: NAS is 8c31e8fd6208625a, Radius is 7e2a07ffaae1fc0f, not match) Check Error, Request Deny by hb.cn
Fri May  6 12:46:10 2016 daemon.err pppd[8049]: PAP authentication failed
Fri May  6 12:46:10 2016 daemon.notice pppd[8049]: Modem hangup
Fri May  6 12:46:10 2016 daemon.notice pppd[8049]: Connection terminated.
Fri May  6 12:46:10 2016 daemon.info pppd[8049]: Sent PADT
Fri May  6 12:46:10 2016 daemon.info pppd[8049]: Exit.
`
*************************************
`
Fri May  6 12:46:10 2016 daemon.info pppd[8049]: PPP session is 9458
Fri May  6 12:46:10 2016 daemon.warn pppd[8049]: Connected to 00:46:4b:71:8a:98 via interface eth1
Fri May  6 12:46:10 2016 daemon.info pppd[8049]: Using interface pppoe-wan
Fri May  6 12:46:10 2016 daemon.notice pppd[8049]: Connect: pppoe-wan <--> eth1
Fri May  6 12:46:10 2016 kern.info kernel: [  438.820000] pppoe-wan: renamed from ppp0
Fri May  6 12:46:10 2016 daemon.info pppd[8049]: syncppp not active
Fri May  6 12:46:10 2016 daemon.info pppd[8049]: Remote message: 29;User(****************)'s Authen Attrib(ai-Service-Password: NAS is 8c31e8fd6208625a, Radius is 7e2a07ffaae1fc0f, not match) Check Error, Request Deny by hb.cn
Fri May  6 12:46:10 2016 daemon.err pppd[8049]: PAP authentication failed
Fri May  6 12:46:10 2016 daemon.notice pppd[8049]: Modem hangup
Fri May  6 12:46:10 2016 daemon.notice pppd[8049]: Connection terminated.
Fri May  6 12:46:10 2016 daemon.info pppd[8049]: Sent PADT
Fri May  6 12:46:10 2016 daemon.info pppd[8049]: Exit.
`
**************************************
`
Fri May  6 12:46:21 2016 daemon.info pppd[8155]: syncppp not active
Fri May  6 12:46:21 2016 daemon.info pppd[8155]: Remote message: 29;User(**************)'s Authen Attrib(ai-Service-Password: NAS is 8c31e8fd6208625a, Radius is 7e2a07ffaae1fc0f, not match) Check Error, Request Deny by hb.cn
Fri May  6 12:46:21 2016 daemon.err pppd[8155]: PAP authentication failed
`
*************************************
`Fri May  6 12:46:50 2016 daemon.info pppd[8594]: syncppp not active
Fri May  6 12:46:50 2016 daemon.info pppd[8594]: Remote message: 29;User(18186471312@hkd)'s Authen Attrib(ai-Service-Password: NAS is 8c31e8fd6208625a, Radius is 7e2a07ffaae1fc0f, not match) Check Error, Request Deny by hb.cn
Fri May  6 12:46:50 2016 daemon.err pppd[8594]: PAP authentication failed
`
----------------------------------------------------------------------------------------------------
一直出现Remote message: 29;这种的提示，
看日志，算号的流程貌似是没问题的，时间也同步过了。

后来偶尔出现
Remote message: 18;User(****************) Checking LoginUser RBillServer, return Called Number Not Allowed, Request Deny by hb.cn  
这种的。

请问大大能看出是什么原因导致的吗？路由是qca9533的mw305。

sunflyer, 2016-05-07:
这东西就是电信自己计费系统BUG，过段时间就好了

nimasi, 2016-05-07:
@sunflyer 额好吧，谢啦

closed 2016-05-11 by nimasi