miao1007/Openwrt-NetKeeper #77: 武汉E信的IPv6拨号有可能破解吗？

smounives, 2016-05-18:
用E信官方客户端（2.5.0084）可以进行IPv6拨号，但是只支持32位Windows，这个IPv6无限速，能跑满物理带宽，要是能破解就好了_(:3」∠)_

miao1007, 2016-05-18:
如果目前在ipv4上能够拨号的话，那么理论上是可以的。

需要在openwrt的ppp中配置ipv6相关即可

smounives, 2016-05-18:
貌似不一样，电脑上的E信客户端用IPv6拨号成功之后没有PPPoE连接，直接在以太网配置好了IP，不明其原理_(:3」∠)_

miao1007, 2016-05-18:
用实体机或者虚拟机进行ipv6拨号，用wireshark打开某个网卡的监控，然后看抓包用的协议与内容

smounives, 2016-05-18:
麻烦您帮忙看一下，我没找到在哪个步骤进行拨号了，抓的是有线网卡，从点击IPv6拨号开始到拨号完成获取到IPv6地址结束。
https://mega.nz/#!exVFUQha!mU8k0_f6xw6wTuXLMoX14DfOXS5O4yR0dOj-MEhhU3M

miao1007, 2016-05-18:
发个安装包吧，里面UDP消息太多了

sunflyer, 2016-05-18:
ipv6拨号的时候账号应该加上@ds_lite作为标识，然后WAN口改成Dual-Stack Lite，
这样一来获取到的是私网IPV4+一个IPV6/64

smounives, 2016-05-18:
@sunflyer 您好，WAN口改成Dual-Stack Lite之后DS-Lite AFTR address改写什么呢？
@miao1007 这里是安装包：http://share.weiyun.com/c9e0f405066295cce3e6005435cba1af

smounives, 2016-05-19:
@sunflyer 加上@ds_lite后缀之后并不能拨号
`Thu May 19 09:26:24 2016 daemon.info pppd[16002]: sxplugin : user  is <
%SSEM a717762407447@ds_lite> 
Thu May 19 09:26:24 2016 daemon.info pppd[16002]: PPP session is 17124
Thu May 19 09:26:24 2016 daemon.warn pppd[16002]: Connected to 54:89:98:75:76:83 via interface eth0.2
Thu May 19 09:26:24 2016 daemon.info pppd[16002]: Using interface pppoe-wan6
Thu May 19 09:26:24 2016 daemon.notice pppd[16002]: Connect: pppoe-wan6 <--> eth0.2
Thu May 19 09:26:24 2016 daemon.info pppd[16002]: syncppp not active
Thu May 19 09:26:27 2016 daemon.info pppd[16002]: syncppp not active
Thu May 19 09:26:30 2016 daemon.info pppd[16002]: syncppp not active
Thu May 19 09:26:30 2016 daemon.info pppd[16002]: Remote message: 12;Roaming User(17762407447@ds_lite) comm with Remote Radius Error, To 202.97.3.18, Timeout 6008ms Error, Request Deny by hb.cn
Thu May 19 09:26:30 2016 daemon.err pppd[16002]: PAP authentication failed
`

miao1007, 2016-05-19:
看了抓包，应该是ppp over ipv6

既然武汉e信官网已经说了是ds_lite双协议栈，建议参考op配置，并配置**wan6**

我建议直接换15.05的系统，原生支持ds_lite，并配置

参考配置：
* https://wiki.openwrt.org/doc/uci/network6#dual-stack_lite_tunnel_ds-lite_ipv4_in_ipv6
* http://www.right.com.cn/forum/thread-167562-1-1.html
* https://tools.ietf.org/html/rfc4241

smounives, 2016-05-19:
@miao1007 谢谢回复，您发的这三个链接我在发issues之前就已经看过了，但是还是有些疑问，首先按照恩山那位楼主说的账号后面加@ds_lite 并不能成功拨上号，大概是我用的PandoraBox的固件原因，等下换官方15.05.1试试，然后DS-Lite AFTR address这个地址我该怎样获取到呢？

miao1007, 2016-05-19:
关于固件：

不建议用国产的软固件，第一是包管理上经常出现奇葩问题，暗坑多，第二是跑满百兆让人没信心，如果有更高的需求话，建议换netgear 3800。

关于拨号：

目前没有用过ipv6环境，这个只能自己慢慢挖坑了。

关于AFTR:

see https://www.wireshark.org/docs/dfref/d/dhcpv6.html

elemaster, 2016-05-19:
@smounives 我用E信客户端IPV6现在无法拨号成功？你是XP系统吗？
op固件 接口wan6 我设置成6to4可以获取到IPV6地址打开IPV6网站 但是IPV6的网速很慢

smounives, 2016-05-19:
@elemaster 我是用的室友的Windows 7 32-bit直接插网线拨号，6to4能获取到IPv6我知道，但是原生IPv6最关键的是不限速，IPv4拨号的限速的
@miao1007 我刷过了Openwrt官方源的15.05.1和最新的Trunk版都是一样的错误日志，加@ds_lite就没法成功拨号，不晓得哪里不对，另外wireshark抓了几次包分析也没找到AFTR地址，DHCPv6的包就只看到一两个而且毫无可用信息，快要放弃了QAQ

elemaster, 2016-05-19:
@smounives QAQ我回去安装个win7试一下 我之前一直IPV6拨号不能 用最新版的E信可以拨成功是吧

smounives, 2016-05-19:
@elemaster 不是最新的，最新版4.0的直接把IPv6拨号的功能都给删了，用我上面发的那个可以

mcgivefenngg, 2016-05-20:
[image]
我学校有人卖可以ipv6拨号的路由器拨号的确实是dual stake
[image]
账号也要加后缀，原作者说过把ipv4封装在ipv6通道里进行传输的，具体我也没研究出来
[image]

miao1007, 2016-05-20:
> 把ipv4封装在ipv6通道里

就是ds_lite

smounives, 2016-05-20:
@mcgivefenngg 您好，您也是用的E信吗？可以告诉我你的DS-Lite下AFTR地址地址是什么吗？

mcgivefenngg, 2016-05-20:
@smounives 
http://pan.baidu.com/s/1mi1tQvm
network文件在这，貌似真的忘了打码。。

smounives, 2016-05-20:
@mcgivefenngg 实在是太太太太太感谢你了！帮了大忙了！

smounives, 2016-05-20:
感谢楼上的各位，总结一下：武汉E信可以使用@exin_ds_lite 后缀进行PPPoE拨号获得原生IPv6，然后增加一个然后WAN口改成Dual-Stack Lite模式，DS-Lite AFTR 地址为：240e:d:1000::ffff:1

miao1007, 2016-05-20:
感谢各位，另帐号、抓包等敏感信息已经删除

nimasi, 2016-05-21:
@smounives   能把你pppoe和Dual-Stack接口的设置给看一下吗，我试了下，加@exin_ds_lite，Remote信息提示用户没找到之类的

smounives, 2016-05-21:
@nimasi 把/etc/config/network 修改成如下即可：

config interface 'wan'
	option ifname 'eth0.2'
	option proto 'pppoe'
	option pppd_options 'plugin sxplugin.so'
	option username 'number@exin_ds_lite'
	option password 'password'
	option ipv6 '1'

config interface 'wan6'
	option proto 'dhcpv6'
	option ifname '@wan'
	option reqaddress 'try'
	option reqprefix 'auto'

config interface 'dslite'
	option proto 'dslite'
	option peeraddr '240e:d:1000::ffff:1'

nimasi, 2016-05-21:
@smounives  谢啦，看上面的图dslite应该要设置和wan一个防火墙里面吧，可是还是不行，提示 `Remote message: 24;User(17771839629) not found(database authen forbiden), Request Deny by hb.cn`，另外请问dslite接口里面Tunnel Link要设置吗？

elemaster, 2016-05-21:
你这提示是认证都没过吧…用客户端ipv6能拨上号吗

nimasi, 2016-05-21:
电脑都是64位的，ipv6说系统不支持没试，我觉得可能是@exin_ds_lite这个后缀我的帐号用不了。。。另外问下dslite那个接口里面的高级设置的Tunnel Link要勾选接口么？

elemaster, 2016-05-21:
不用

nimasi, 2016-05-24:
```
Tue May 24 04:44:16 2016 daemon.info pppd[2071]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Tue May 24 04:44:16 2016 daemon.info pppd[2071]: Plugin sxplugin.so loaded.
Tue May 24 04:44:16 2016 daemon.info pppd[2071]: sxplugin : init
Tue May 24 04:44:16 2016 daemon.info pppd[2071]: -------------------------------------
Tue May 24 04:44:16 2016 daemon.info pppd[2071]: timenow(Hex)=%?x
Tue May 24 04:44:16 2016 daemon.info pppd[2071]: Begin : beforeMD5
Tue May 24 04:44:16 2016 daemon.info pppd[2071]: 1.<s??>
Tue May 24 04:44:16 2016 daemon.info pppd[2071]: 2.<s??02*********>
Tue May 24 04:44:16 2016 daemon.info pppd[2071]: 3.<s??02**********70hubtxinli01>
Tue May 24 04:44:16 2016 daemon.info pppd[2071]: 4.length=<26>
Tue May 24 04:44:16 2016 daemon.info pppd[2071]: End : beforeMD5
Tue May 24 04:44:16 2016 daemon.info pppd[2071]: Begin : afterMD5
Tue May 24 04:44:16 2016 daemon.info pppd[2071]: 1.MD5use_1=< a>
Tue May 24 04:44:16 2016 daemon.info pppd[2071]: 2.MD5use_2=< e>
Tue May 24 04:44:16 2016 daemon.info pppd[2071]: End : afterMD5
Tue May 24 04:44:16 2016 daemon.info pppd[2071]: -------------------------------------
Tue May 24 04:44:16 2016 daemon.info pppd[2071]: sxplugin : user  is <
XZR4< ae02********0@exin_ds_lite> 
Tue May 24 04:44:16 2016 daemon.info pppd[2071]: sxplugin : passwd loaded
Tue May 24 04:44:16 2016 daemon.notice pppd[2071]: pppd 2.4.7 started by root, uid 0
Tue May 24 04:44:16 2016 daemon.info pppd[2071]: PPP session is 6821
Tue May 24 04:44:16 2016 daemon.warn pppd[2071]: Connected to 00:46:4b:71:8a:98 via interface eth0.2
Tue May 24 04:44:16 2016 kern.info kernel: [  268.620000] pppoe-wan: renamed from ppp0
Tue May 24 04:44:16 2016 daemon.info pppd[2071]: Using interface pppoe-wan
Tue May 24 04:44:16 2016 daemon.notice pppd[2071]: Connect: pppoe-wan <--> eth0.2
Tue May 24 04:44:16 2016 daemon.info pppd[2071]: Remote message: 0;User(02********) Authenticate OK, Request Accept by hb.cn
Tue May 24 04:44:16 2016 daemon.notice pppd[2071]: PAP authentication succeeded
Tue May 24 04:44:16 2016 daemon.notice pppd[2071]: peer from calling number 00:46:4B:71:8A:98 authorized
Tue May 24 04:44:16 2016 daemon.notice pppd[2071]: local  LL address fe80::f4eb:98c7:1640:2777
Tue May 24 04:44:16 2016 daemon.notice pppd[2071]: remote LL address fe80::0225:9eff:fe81:6477
Tue May 24 04:44:16 2016 daemon.notice netifd: Network device 'pppoe-wan' link is up
Tue May 24 04:44:16 2016 daemon.notice netifd: Interface 'wan6' is enabled
Tue May 24 04:44:16 2016 daemon.notice netifd: Interface 'wan6' is setting up now
Tue May 24 04:44:16 2016 daemon.notice netifd: Interface 'wan' is now up
Tue May 24 04:44:17 2016 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Tue May 24 04:44:17 2016 daemon.notice netifd: Interface 'dslite' is setting up now
Tue May 24 04:44:17 2016 daemon.notice netifd: Interface 'wan6' is now up
Tue May 24 04:44:17 2016 daemon.notice netifd: Interface 'dslite' is now up
Tue May 24 04:44:17 2016 daemon.notice netifd: IP tunnel 'ds-dslite' link is up
Tue May 24 04:44:18 2016 user.notice firewall: Reloading firewall due to ifup of wan6 (pppoe-wan)
Tue May 24 04:44:18 2016 daemon.warn odhcpd[787]: A default route is present but there is no public prefix on br-lan thus we don't announce a default route!
Tue May 24 04:44:19 2016 user.notice firewall: Reloading firewall due to ifup of dslite (ds-dslite)

```

@elemaster  @smounives  
我用同学的账号尝试，认证成功了，他那种是纯手机号，没有后缀的那种，而且连上后wan虽然有收发，但是连不上网，网页也没法打开，请问从上面的日志能看出什么问题吗，

smounives, 2016-05-24:
@nimasi wan6接口下的 ifname 要改成@wan

nimasi, 2016-05-24:
@smounives    wan6的ifname已经改成了@wan的。。。。

elemaster, 2016-05-24:
@nimasi 请设置DNS

nimasi, 2016-05-24:
@elemaster  额T◡T，请问该如何设置DNS。。。。

smounives, 2016-05-24:
@nimasi Lan口下IPv6设置，或者dnsmasq自己手动配置，或者其他诸如ChinaDNS、Pcap_DNSProxy、dnscrypt-proxy等等，善用搜索引擎

nimasi, 2016-05-24:
@smounives   谢谢 ，明天找同学再试试

nimasi, 2016-05-25:
调皮的电信啊，今天不知道什么鬼。。。。`Remote message: 26;User(02792763170)'s service aDSLPPPoEAuthen(1001) is Locked, Request Deny by hb.cn`   以后有时间再试吧，先谢谢各位了

closed 2016-06-04 by smounives

nimasi, 2016-06-13:
试了下在lan口添加了dns不行，不知道是不是设置不对，但是添加一个为DHCP协议的接口，和wan在同一个防火墙区域里面后就可以了

fy1789, 2016-10-14:
谢谢分享， 10月份后武汉e信的ipv6应该限速了。现在只有的原来的20M宽带了。 
   请问下楼主， 有什么方法可以把ipv6分配给lan口让电脑和wifi能够访问ipv6资源。  折腾了两天，现在客户端可以获取ipv6地址，但还是无法访问ipv6资源。实在没办法，特此求助。谢谢

smounives, 2016-10-14:
我已经毕业了，没环境测试，最简单的你可以试试NAT虽然不推荐，Openwrt 15.05自带的IPv6有点问题，我当初用的时候也是有时候能拿到Global IPv6 有时候ping一下网关才能拿到路由，Bug很多，现在没用Openwrt了，有些东西不记得了