miao1007/Openwrt-NetKeeper #86: 资源帖，请贴出你们所在地区的客户端下载官网，以供反编译分析

1558204459, 2016-06-20:


sunflyer, 2016-06-20:
http://img.cq.ct10000.com/static/upload/2014/11/11/windows589.rar
重庆电信4.7.9.589.
还请帮忙。目前已知变动部分为帐号前2位。详细信息在wiki（不保证准确）

BackTrackCRoot, 2016-06-22:
要安卓端还是PC端？PC端是最新的VMP保护。

1558204459, 2016-06-22:
PC和Android都贴上

mcgivefenngg, 2016-06-26:
kd.wuhan.net.cn
武汉的，所有版本都在这个网站下载

milkyfox, 2016-07-26:
http://219.146.3.50/clientdownloadall.html
山东电信

brainbush, 2016-09-06:
http://117.21.211.239/apps/NKSetup4.7.9.589.exe

Kolakona, 2016-09-10:
http://www.114school.cn/xytypt/typt/download/download.html
浙江电信

sunflyer, 2016-09-10:
@Kolakona 浙江电信此次更新没有改动，2.5继续使用

hehewocao, 2016-09-15:
大神能不能搞广东天翼呢？

qq734694920, 2016-09-23:
http://223.99.248.20/山东移动

AmonKing, 2016-10-03:
广东电信天翼 暂时没有发现支持的
http://enet.10000.gd.cn:10001/qs/index.jsp

qq734694920, 2017-03-15:
山东翼讯
http://pan.baidu.com/s/1qYdLkvE
同样没有发现支持的

1shanyue1, 2017-03-24:
河北，不过一直在用第三个重庆的
www.icampus.ren

sunflyer, 2017-03-24:
此贴终结。目前没什么必要了，VMP3.X保护的Netkeeper按照目前的情况不是高玩基本没有反编译的可能。

zxy4ever, 2017-04-22:
重庆地区今年虽然更新为了NKSetup4.7.9.589，但其实其拨号原理还是和之前的NKSetup2.5完全一样的，因为老版客户端依然可以正常使用。
http://dnxz2.vhdown.com:8045/soft/1204/WIN8NKSetup64_veryhuo.com.rar

disaccept, 2017-05-06:
http://218.77.121.111/DownloadXml/ClientPkgs/3ad1f9ad-4f96-46f5-b86b-1e88fb76afd9/HNCV2.2.6.17032201.exe
万恶的湖南电信

disaccept, 2017-05-06:
pc：http://218.77.121.111/DownloadXml/ClientPkgs/3ad1f9ad-4f96-46f5-b86b-1e88fb76afd9/HNCV2.2.6.17032201.exe
Android：http://218.77.121.111/DownloadXml/ClientPkgs/907255ff-eebf-4eca-89ef-22fd462d109f/HNCV3.3.24.03061b.apk

sjg1993, 2017-05-26:
重庆电信，最近刚升级客户端，彻底取缔了老版本客户端
下载地址:http://cq.189.cn/college/home/resource.htm