miao1007/Openwrt-NetKeeper #90: 我在执行sh文件的时候出错了，请问下这是怎么回事

DevilShow, 2016-07-03:
Sun Jul  3 11:33:27 2016 daemon.notice netifd: Interface 'netkeeper' has link connectivity loss
Sun Jul  3 11:33:27 2016 daemon.notice netifd: Interface 'netkeeper' has link connectivity 
Sun Jul  3 11:33:27 2016 daemon.notice netifd: Interface 'lan' is now down
Sun Jul  3 11:33:27 2016 kern.info kernel: [ 3263.540000] br-lan: port 2(wlan0) entered disabled state
Sun Jul  3 11:33:27 2016 kern.info kernel: [ 3263.540000] br-lan: port 1(eth0) entered disabled state
Sun Jul  3 11:33:27 2016 kern.info kernel: [ 3263.550000] device eth0 left promiscuous mode
Sun Jul  3 11:33:27 2016 kern.info kernel: [ 3263.560000] br-lan: port 1(eth0) entered disabled state
Sun Jul  3 11:33:27 2016 kern.info kernel: [ 3263.570000] eth0: link down
Sun Jul  3 11:33:27 2016 kern.info kernel: [ 3263.570000] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
Sun Jul  3 11:33:27 2016 kern.info kernel: [ 3263.590000] device wlan0 left promiscuous mode
Sun Jul  3 11:33:27 2016 kern.info kernel: [ 3263.590000] br-lan: port 2(wlan0) entered disabled state
Sun Jul  3 11:33:27 2016 daemon.notice netifd: Interface 'lan' is disabled
Sun Jul  3 11:33:27 2016 daemon.notice netifd: Network device 'eth0' link is down
Sun Jul  3 11:33:27 2016 daemon.notice netifd: VLAN 'eth0.2' link is down
Sun Jul  3 11:33:27 2016 daemon.notice netifd: Interface 'netkeeper' has link connectivity loss
Sun Jul  3 11:33:27 2016 daemon.notice netifd: Bridge 'br-lan' link is down
Sun Jul  3 11:33:27 2016 daemon.notice netifd: Interface 'lan' has link connectivity loss
Sun Jul  3 11:33:27 2016 daemon.notice netifd: Interface 'loopback' is now down
Sun Jul  3 11:33:27 2016 daemon.notice netifd: Interface 'loopback' is disabled
Sun Jul  3 11:33:27 2016 daemon.notice netifd: Network device 'lo' link is down
Sun Jul  3 11:33:27 2016 daemon.notice netifd: Interface 'loopback' has link connectivity loss
Sun Jul  3 11:33:31 2016 kern.info kernel: [ 3267.570000] device eth0 entered promiscuous mode
Sun Jul  3 11:33:31 2016 kern.info kernel: [ 3267.580000] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
Sun Jul  3 11:33:31 2016 daemon.notice netifd: Interface 'lan' is enabled
Sun Jul  3 11:33:31 2016 daemon.notice netifd: Interface 'lan' is setting up now
Sun Jul  3 11:33:31 2016 daemon.notice netifd: Interface 'lan' is now up
Sun Jul  3 11:33:31 2016 daemon.notice netifd: Interface 'loopback' is enabled
Sun Jul  3 11:33:31 2016 daemon.notice netifd: Interface 'loopback' is setting up now
Sun Jul  3 11:33:31 2016 daemon.notice netifd: Interface 'loopback' is now up
Sun Jul  3 11:33:31 2016 kern.info kernel: [ 3267.630000] IPv6: ADDRCONF(NETDEV_UP): eth0.2: link is not ready
Sun Jul  3 11:33:31 2016 daemon.notice netifd: Interface 'netkeeper' is enabled
Sun Jul  3 11:33:31 2016 daemon.notice netifd: Network device 'lo' link is up
Sun Jul  3 11:33:31 2016 daemon.notice netifd: Interface 'loopback' has link connectivity 
Sun Jul  3 11:33:31 2016 daemon.info dnsmasq[1074]: read /etc/hosts - 1 addresses
Sun Jul  3 11:33:31 2016 daemon.info dnsmasq[1074]: read /tmp/hosts/odhcpd - 0 addresses
Sun Jul  3 11:33:31 2016 daemon.info dnsmasq[1074]: read /tmp/hosts/dhcp - 1 addresses
Sun Jul  3 11:33:31 2016 daemon.info dnsmasq-dhcp[1074]: read /etc/ethers - 0 addresses
Sun Jul  3 11:33:31 2016 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Sun Jul  3 11:33:32 2016 kern.info kernel: [ 3268.930000] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Sun Jul  3 11:33:32 2016 daemon.notice netifd: radio0 (7624): Configuration file: /var/run/hostapd-phy0.conf
Sun Jul  3 11:33:32 2016 kern.info kernel: [ 3268.970000] device wlan0 entered promiscuous mode
Sun Jul  3 11:33:32 2016 kern.info kernel: [ 3268.970000] br-lan: port 2(wlan0) entered forwarding state
Sun Jul  3 11:33:32 2016 kern.info kernel: [ 3268.980000] br-lan: port 2(wlan0) entered forwarding state
Sun Jul  3 11:33:32 2016 daemon.notice netifd: Bridge 'br-lan' link is up
Sun Jul  3 11:33:32 2016 daemon.notice netifd: Interface 'lan' has link connectivity 
Sun Jul  3 11:33:32 2016 kern.info kernel: [ 3268.990000] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
Sun Jul  3 11:33:32 2016 daemon.notice netifd: radio0 (7624): wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Sun Jul  3 11:33:32 2016 daemon.notice netifd: radio0 (7624): Using interface wlan0 with hwaddr e0:05:c5:2a:50:54 and ssid "holishit!"
Sun Jul  3 11:33:33 2016 kern.info kernel: [ 3269.540000] br-lan: port 2(wlan0) entered disabled state
Sun Jul  3 11:33:33 2016 kern.info kernel: [ 3269.670000] eth0: link up (1000Mbps/Full duplex)
Sun Jul  3 11:33:33 2016 kern.info kernel: [ 3269.670000] br-lan: port 1(eth0) entered forwarding state
Sun Jul  3 11:33:33 2016 kern.info kernel: [ 3269.680000] br-lan: port 1(eth0) entered forwarding state
Sun Jul  3 11:33:33 2016 daemon.notice netifd: Network device 'eth0' link is up
Sun Jul  3 11:33:33 2016 daemon.notice netifd: VLAN 'eth0.2' link is up
Sun Jul  3 11:33:33 2016 daemon.notice netifd: Interface 'netkeeper' has link connectivity 
Sun Jul  3 11:33:33 2016 daemon.notice netifd: Interface 'netkeeper' is setting up now
Sun Jul  3 11:33:33 2016 kern.info kernel: [ 3269.710000] IPv6: ADDRCONF(NETDEV_CHANGE): eth0.2: link becomes ready
Sun Jul  3 11:33:34 2016 daemon.info pppd[7772]: Plugin rp-pppoe.so loaded.
Sun Jul  3 11:33:34 2016 daemon.info pppd[7772]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Sun Jul  3 11:33:34 2016 daemon.notice netifd: netkeeper (7772): /usr/sbin/pppd: '/usr/lib/pppd/2.4.7/chongqing_sxplugin.so' is not an ELF executable for MIPS
Sun Jul  3 11:33:34 2016 daemon.notice netifd: netkeeper (7772): /usr/sbin/pppd: File not found
Sun Jul  3 11:33:34 2016 daemon.err pppd[7772]: File not found
Sun Jul  3 11:33:34 2016 daemon.notice netifd: netkeeper (7772): /usr/sbin/pppd: Couldn't load plugin chongqing_sxplugin.so
Sun Jul  3 11:33:34 2016 daemon.err pppd[7772]: Couldn't load plugin chongqing_sxplugin.so
Sun Jul  3 11:33:34 2016 daemon.warn odhcpd[775]: DHCPV6 SOLICIT IA_NA from 000100011d96aecbb82a72b5587f on br-lan: ok fd18:bfe4:4525::cfb/128 
Sun Jul  3 11:33:34 2016 kern.info kernel: [ 3270.770000] br-lan: port 2(wlan0) entered forwarding state
Sun Jul  3 11:33:34 2016 kern.info kernel: [ 3270.770000] br-lan: port 2(wlan0) entered forwarding state
Sun Jul  3 11:33:34 2016 kern.info kernel: [ 3270.780000] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Sun Jul  3 11:33:34 2016 daemon.notice netifd: Interface 'netkeeper' is now down
Sun Jul  3 11:33:34 2016 daemon.notice netifd: Interface 'netkeeper' is disabled
Sun Jul  3 11:33:34 2016 daemon.notice netifd: radio0 (7624): wlan0: interface state COUNTRY_UPDATE->ENABLED
Sun Jul  3 11:33:34 2016 daemon.notice netifd: radio0 (7624): wlan0: AP-ENABLED 
Sun Jul  3 11:33:34 2016 daemon.warn odhcpd[775]: DHCPV6 SOLICIT IA_NA from 000100011d96aecbb82a72b5587f on br-lan: ok fd18:bfe4:4525::cfb/128 
Sun Jul  3 11:33:35 2016 daemon.notice netifd: Network device 'wlan0' link is up
Sun Jul  3 11:33:35 2016 kern.info kernel: [ 3271.680000] br-lan: port 1(eth0) entered forwarding state
Sun Jul  3 11:33:35 2016 daemon.warn odhcpd[775]: DHCPV6 REQUEST IA_NA from 000100011d96aecbb82a72b5587f on br-lan: ok fd18:bfe4:4525::cfb/128 
Sun Jul  3 11:33:36 2016 kern.info kernel: [ 3272.770000] br-lan: port 2(wlan0) entered forwarding state

这是系统日志
[image]
这个是我在winscp上执行confnetwork.sh时的报错，请问下这个问题怎么解决啊

sunflyer, 2016-07-03:
/usr/sbin/pppd: '/usr/lib/pppd/2.4.7/chongqing_sxplugin.so' is not an ELF executable for MIPS
Sun Jul 3 11:33:34 2016 daemon.notice netifd: netkeeper (7772): /usr/sbin/pppd: File not found
Sun Jul 3 11:33:34 2016 daemon.err pppd[7772]: File not found
Sun Jul 3 11:33:34 2016 daemon.notice netifd: netkeeper (7772): /usr/sbin/pppd: Couldn't load plugin chongqing_sxplugin.so
Sun Jul 3 11:33:34 2016 daemon.err pppd[7772]: Couldn't load plugin chongqing_sxplugin.so

你用错插件版本了。年轻人，你不看你路由器CPU型号的？

sunflyer, 2016-07-03:
另外这个脚本对应的接口是WAN而不是NETKEEPER，你没有WAN这个接口所以报错

miao1007, 2016-07-03:
同楼上，你可以把NETKEEPER接口换成wan口

DevilShow, 2016-07-07:
这个该用哪个插件啊。。本人才接触这个，请大神们指教指教