miao1007/Openwrt-NetKeeper #93: 运行第二个指令报错

3721915, 2016-09-07:
学校有卖破解路由器的居然一个要300.于是想自己弄
前面一切顺利到了执行sh /tmp/confnetwork.sh这个的时候报错
/tmp$ chmod a+x /tmp/confnetwork.sh
/tmp$ sh /tmp/confnetwork.sh
/tmp/confnetwork.sh: line 1: 锘?!bin/sh: not found
uci: Entry not found
uci: Entry not found
Warning: Unable to locate ipset utility, disabling ipset support
Warning: Section @zone[1] (wan) cannot resolve device of network 'wan'
 * Flushing IPv4 filter table
 * Flushing IPv4 nat table
 * Flushing IPv4 mangle table
 * Flushing IPv4 raw table
 * Flushing IPv6 filter table
 * Flushing IPv6 mangle table
 * Flushing IPv6 raw table
 * Flushing conntrack table ...
 * Populating IPv4 filter table
   * Zone 'lan'
   * Zone 'wan'
   * Rule 'Allow-DHCP-Renew'
   * Rule 'Allow-Ping'
   * Forward 'lan' -> 'wan'
 * Populating IPv4 nat table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv4 mangle table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv4 raw table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv6 filter table
   * Zone 'lan'
   * Zone 'wan'
   * Rule 'Allow-DHCPv6'
   * Rule 'Allow-ICMPv6-Input'
   * Rule 'Allow-ICMPv6-Forward'
   * Forward 'lan' -> 'wan'
 * Populating IPv6 mangle table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv6 raw table
   * Zone 'lan'
   * Zone 'wan'
 * Flushing conntrack: fe80::28c6:8eff:fec2:b0a
 * Set tcp_ecn to off
 * Set tcp_syncookies to on
 * Set tcp_window_scaling to on
 * Running script '/etc/firewall.user'
'radio0' is disabled
'radio1' is disabled
'radio0' is disabled
'radio1' is disabled
'radio0' is disabled
'radio1' is disabled

下面是confnetwork.sh 请问是这样写吗我是河北联通
#!bin/sh
#If /etc/config/network file has preconfig wan sittings then delete it.
uci delete network.wan
uci delete network.wan6
uci commit network
uci set network.netkeeper=interface
uci set network.netkeeper.ifname=eth0.2
uci set network.netkeeper.macaddr=aabbccddeeff
uci set network.netkeeper.proto=pppoe
#TODO:set your province's plugin name
uci set network.netkeeper.pppd_options='hebei_sxplugin.so'
#TODO:set pppoe account and password
uci set network.netkeeper.username=031105343350@campus 
uci set network.netkeeper.password=3721915
uci set network.netkeeper.metric='0'
uci commit network
#set firewall                                    
uci set firewall.@zone[1].network='wan netkeeper' 
uci commit firewall
/etc/init.d/firewall restart
/etc/init.d/network reload
/etc/init.d/network restart

miao1007, 2016-09-07:
没有sh? 那就直接运行 `./confnetwork.sh` 吧

3721915, 2016-09-07:
@miao1007 直接第二个指令运行./confnetwork.sh吗

SaltNego, 2016-09-30:
其实是编码问题，改一下就好了

thealivens, 2021-11-07:
请问你怎么解决的

miao1007, 2021-11-07:
如果是用windows上传的话，可能是编码问题，可以用 dos2unix 命令转换

closed 2021-11-07 by miao1007