miao1007/Openwrt-NetKeeper #94: Couldn't load plugin wuhan_sxplugin.so

InCommingMessage, 2016-09-07:
Wed Aug 31 20:30:44 2016 daemon.notice netifd: wan (8231): udhcpc (v1.24.2) started
Wed Aug 31 20:30:44 2016 daemon.notice netifd: wan (8231): Sending discover...
Wed Aug 31 20:30:47 2016 daemon.notice netifd: wan (8231): Sending discover...
Wed Aug 31 20:30:50 2016 daemon.notice netifd: wan (8231): Sending discover...
Wed Aug 31 20:30:57 2016 daemon.notice netifd: Interface 'netkeeper' is enabled
Wed Aug 31 20:30:57 2016 daemon.notice netifd: Interface 'netkeeper' has link connectivity 
Wed Aug 31 20:30:57 2016 daemon.notice netifd: Interface 'netkeeper' is setting up now
Wed Aug 31 20:30:57 2016 daemon.notice netifd: Interface 'netkeeper' is now down
Wed Aug 31 20:30:57 2016 daemon.notice netifd: Interface 'netkeeper' is setting up now
Wed Aug 31 20:30:57 2016 daemon.info pppd[8465]: Plugin rp-pppoe.so loaded.
Wed Aug 31 20:30:57 2016 daemon.info pppd[8465]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Wed Aug 31 20:30:57 2016 daemon.err pppd[8465]: Error relocating /usr/lib/pppd/2.4.7/wuhan_sxplugin.so: unsupported relocation type 3
Wed Aug 31 20:30:57 2016 daemon.notice netifd: netkeeper (8465): /usr/sbin/pppd: Error relocating /usr/lib/pppd/2.4.7/wuhan_sxplugin.so: unsupported relocation type 3
Wed Aug 31 20:30:57 2016 daemon.err pppd[8465]: Couldn't load plugin wuhan_sxplugin.so
Wed Aug 31 20:30:57 2016 daemon.notice netifd: netkeeper (8465): /usr/sbin/pppd: Couldn't load plugin wuhan_sxplugin.so
Wed Aug 31 20:30:57 2016 daemon.notice netifd: Interface 'netkeeper' is now down
======================================================
在树莓派3B上使用貌似无法加载so库......求救

sunflyer, 2016-09-08:
所以你是否重新编译了？
树莓派是arm架构，但是路由器一般是mips，肯定不能直接用的

InCommingMessage, 2016-09-08:
求教。。。应该用哪个Toolchain进行编译啊。。。

closed 2016-09-11 by miao1007