miao1007/Openwrt-NetKeeper #96: 网件 WNDR3800 ，已刷openwrt15.05，重庆电信netkeeper（已Star）

cherishfall, 2016-09-09:
自购网件 WNDR3800 ，已刷openwrt 15.05，重庆邮电大学学生，校园电信宽带，win10系统。
大致按照教程http://tieba.baidu.com/p/4500434644?see_lz=1
我的操作如下：
1）下载winscp，vscode并安装；下载confnetwork.sh，sxplugin-ar71xx-14.04文件夹中chongqing_sxplugin.so
2)打开winscp，连接路由器（主机名192.168.1.1，端口22，用户名和密码都是root），将chongqing_sxplugin.so文件拖拽到 /usr/lib/pppd/2.4.7/目录下，并对chongqing_sxplugin.so文件左击——属性——权限——八进制，将0644改为0755
3）用vscode打开confnetwork.sh，代码如下：
`#!bin/sh
#If /etc/config/network file has preconfig wan sittings then delete it.
uci delete network.wan
uci delete network.wan6
uci commit network
#this is a sample for Netkeeper in chongqing on OpenWRT
uci set network.netkeeper=interface
uci set network.netkeeper.ifname=eth0.2
uci set network.netkeeper.macaddr=aabbccddeeff
uci set network.netkeeper.proto=pppoe
#TODO:set your province's plugin name
uci set network.netkeeper.pppd_options='plugin chongqing_sxplugin.so'
#TODO:set pppoe account and password
uci set network.netkeeper.username=17772413437@cqupt
uci set network.netkeeper.password=057757
uci set network.netkeeper.metric='0'
uci commit network
#set firewall                                    
uci set firewall.@zone[1].network='wan netkeeper' 
uci commit firewall
/etc/init.d/firewall restart
/etc/init.d/network reload
/etc/init.d/network restart`
并将保存后的confnetwork.sh文件拖拽到/tmp/ 目录
4）打开winscp带的终端
运行chmod a+x/tmp/confnetwork.sh命令，出现如下错误窗口提示：
/tmp$ chmod a+x/tmp/confnetwork.sh
BusyBox v1.23.2 (2015-07-25 15:09:46 CEST) multi-call binary.

Usage: chmod [-R] MODE[,MODE]... FILE...

Each MODE is one or more of the letters ugoa, one of the
symbols +-= and one or more of the letters rwxst

	-R	Recurse
5)运行sh /tmp/confnetwork.sh命令，输出如下：
/tmp$ sh /tmp/confnetwork.sh
Warning: Unable to locate ipset utility, disabling ipset support
Warning: Section @zone[1] (wan) cannot resolve device of network 'netkeeper'
 * Flushing IPv4 filter table
 * Flushing IPv4 nat table
 * Flushing IPv4 mangle table
 * Flushing IPv4 raw table
 * Flushing IPv6 filter table
 * Flushing IPv6 mangle table
 * Flushing IPv6 raw table
 * Flushing conntrack table ...
 * Populating IPv4 filter table
   * Zone 'lan'
   * Zone 'wan'
   * Rule 'Allow-DHCP-Renew'
   * Rule 'Allow-Ping'
   * Rule 'Allow-IGMP'
   * Rule #7
   * Rule #8
   * Forward 'lan' -> 'wan'
 * Populating IPv4 nat table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv4 mangle table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv4 raw table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv6 filter table
   * Zone 'lan'
   * Zone 'wan'
   * Rule 'Allow-DHCPv6'
   * Rule 'Allow-MLD'
   * Rule 'Allow-ICMPv6-Input'
   * Rule 'Allow-ICMPv6-Forward'
   * Rule #7
   * Rule #8
   * Forward 'lan' -> 'wan'
 * Populating IPv6 mangle table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv6 raw table
   * Zone 'lan'
   * Zone 'wan'
 * Set tcp_ecn to off
 * Set tcp_syncookies to on
 * Set tcp_window_scaling to on
 * Running script '/etc/firewall.user'
'radio0' is disabled
'radio1' is disabled
'radio0' is disabled
'radio1' is disabled
6）IE浏览器地址栏输入192.168.1.1，用户名密码输入root，进入openwrt [luci界面——接口——出现如下图片：](http://img2.ph.126.net/7CdtROc8CS9BWWctcJmrRg==/6631785746793964304.png)
wan口，lan口连接正常，账号插在电脑网线插口netkeeper拨号成功正常联网。
请问如何操作才能正确破解netkeeper呢？实现路由开wifi，路由上网呢？

miao1007, 2016-09-11:
重邮的需要配置一个wan，一个netkeeper，它们的metric一个是0，一个是正数，它们在一个防火墙中。

你可以按照脚本的内容去鼠标配置，具体我忘记了 >_>

cappuccinoqi, 2016-10-14:
@cherishfall 我略有经验，如果需要帮助 我留个企鹅617674934

xiaogegelc, 2016-11-07:
现在你那个还能用吗？我这里十分钟掉一次 重医

cherishfall, 2016-11-07:
我不用了，我们学校里有人在用，你的路由器要是支持 simple netkeeper

https://simplenetkeeper.com，你可以去看看。

Fallen

cappuccinoqi, 2016-11-07:
重邮的还能用，

cherishfall, 2016-11-07:
你是一个问句么？你去重庆邮电大学的咸鱼鱼塘，还有人在出租卖这个。有刷潘多拉成功的，有用那个simple netkeeper成功的，openwrt没看到。

Fallen

cappuccinoqi, 2016-11-07:
openwrT和Pandora差不多的

cherishfall, 2016-11-07:
我知道，只是我用网件刷openwrt没成功过，K2刷潘多拉一次成功，我也不知道原因在哪。

Fallen

cappuccinoqi, 2016-11-07:
配置一样么？刷同一个固件试试。

cherishfall, 2016-11-07:
不一样，处理器不一样，k2好像不能刷openwrt.网件那个不能刷潘多拉。

Fallen

cappuccinoqi, 2016-11-07:
@cherishfall  理论上open和Pandora都可以用netkeeper的，你刷的固件可能是缺少某些组件

cherishfall, 2016-11-07:
起先我刷的是明月的，后来换成openwrt官网的，都没用。

Fallen

cappuccinoqi, 2016-11-07:
@cherishfall 这样就比较尴尬了

cherishfall, 2016-11-07:
对，很尴尬，去撸了K2

Fallen

cappuccinoqi, 2016-11-07:
k1 k2都没问题  我都上车了

cherishfall, 2016-11-07:
我安全下车了，质量很差，五六十的货，不用了，对wifi没有多大需求。

Fallen

cappuccinoqi, 2016-11-07:
拿来玩玩而已，应付一下也还将就

kuri-leo, 2017-02-15:
同学，你的netkeeper账号和密码都暴露了0w0

caiqiqi, 2017-02-21:
@miao1007 
WAN口必须要修改成172的内网地址才能连上netkeeper吗？如果用DHCP，只能得到 192.168.0.32/24这样的地址。另外WAN接口和NETKEEPER接口的metric值必须一个为0，另一个为正数(1，2都可以吗？)