miao1007/Openwrt-NetKeeper #98: 执行sh出现的问题

LosReturn, 2016-09-15:
uci: Entry not found
uci: Entry not found
Warning: Section @zone[1] (wan) cannot resolve device of network 'wan'

SaltNego, 2016-09-30:
我也是求教

mentospcg, 2017-03-06:
这个问题一直没有解决吗？我最近也在玩这个也没有解决，感觉.so库都没有运行起来

miao1007, 2017-03-06:
修改你的外网network，改名叫做wan

mentospcg, 2017-03-07:
十分感激 @miao1007 ，这个问题暂时解决了。但是这个脚本我还是没有正常运行完成，麻烦继续帮忙解决一下。下面是log信息：
root@OpenWrt:/# sh ./confnetwork.sh
Warning: Unable to locate ipset utility, disabling ipset support
 * Flushing IPv4 filter table
 * Flushing IPv4 nat table
 * Flushing IPv4 mangle table
 * Flushing IPv4 raw table
 * Flushing IPv6 filter table
 * Flushing IPv6 mangle table
 * Flushing IPv6 raw table
 * Flushing conntrack table ...
 * Populating IPv4 filter table
   * Zone 'lan'
   * Zone 'wan'
   * Rule 'Allow-DHCP-Renew'
   * Rule 'Allow-Ping'
   * Forward 'lan' -> 'wan'
 * Populating IPv4 nat table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv4 mangle table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv4 raw table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv6 filter table
   * Zone 'lan'
   * Zone 'wan'
   * Rule 'Allow-DHCPv6'
   * Rule 'Allow-ICMPv6-Input'
   * Rule 'Allow-ICMPv6-Forward'
   * Forward 'lan' -> 'wan'
 * Populating IPv6 mangle table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv6 raw table
   * Zone 'lan'
   * Zone 'wan'
 * Set tcp_ecn to off
 * Set tcp_syncookies to on
 * Set tcp_window_scaling to on
 * Running script '/etc/firewall.user'
[  262.920000] rt305x-esw 10110000.esw: link changed 0x00
[  263.290000] br-lan: port 2(wlan0) entered disabled state
[  263.300000] br-lan: port 1(eth0.1) entered disabled state
[  263.330000] device eth0.1 left promiscuous mode
[  263.330000] device eth0 left promiscuous mode
[  263.340000] br-lan: port 1(eth0.1) entered disabled state
[  263.370000] IPv6: ADDRCONF(NETDEV_UP): eth0.1: link is not ready
[  263.410000] device wlan0 left promiscuous mode
[  263.410000] br-lan: port 2(wlan0) entered disabled state
**Command failed: Not found**
root@OpenWrt:/# [  266.640000] rt305x-esw 10110000.esw: link changed 0x08
[  268.090000] device eth0.1 entered promiscuous mode

调试了一下发现应该是
/etc/init.d/network restart   这句出现的问题

mentospcg, 2017-03-07:
我用WinSCP执行了一下脚本命令
错误截图如下：
[image]
对了，我是用的RT5350的一块开发板在玩，版本信息如下：
Linux version 3.10.36 (f403tech@f403tech-pc) (gcc version 4.8.3 (OpenWrt/Linaro GCC 4.8-2014.04 r40773) ) #3 Sat Mar 4 15:20:46 CST 2017

NETLizard, 2017-03-08:
你可以直接更改network文件

mentospcg, 2017-03-09:
我查看了network文件里面的配置应该是没有问题的。现在就是在接口那点连接后并没有什么反应，怎样才知道加载了.so文件呢

root@OpenWrt:/# vi etc/config/network 

config interface 'loopback'
        option ifname 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd6f:4a87:66b4::/48'

config interface 'lan'
        option ifname 'eth0.1'
        option force_link '1'
        option type 'bridge'
        option proto 'static'
        option ipaddr '172.22.141.199'
        option netmask '255.255.252.0'
        option ip6assign '60'

config interface 'wan'
        option ifname 'eth0.2'
        option proto 'dhcp'

config switch
        option name 'rt305x'
        option reset '1'
        option enable_vlan '1'

config switch_vlan
        option device 'rt305x'
        option vlan '1'
        option ports '0 1 2 3  6t'

config switch_vlan
        option device 'rt305x'
        option vlan '2'
        option ports '4  6t'

config interface 'netkeeper'
        option ifname 'eth0.2'
        option macaddr 'aabbccddeeff'
        option proto 'pppoe'
        option pppd_options 'plugin chongqing_sxplugin.so'
        option username '189xxxxxxxx'
        option password 'xxxxxx'
        option metric '0'

[image]