miao1007/Openwrt-NetKeeper #100: 我们学校是有的宿舍能用有的不能，什么情况

qq734694920, 2016-09-23:
系统日志已经上传：http://pan.baidu.com/s/1eSbowNo
校园安卓版客户端：http://pan.baidu.com/s/1ctD3iu

qq734694920, 2016-09-26:
应该是您所说的静态路由问题，但是应该怎么配置静态路由呢
我这里总是显示Timeout waiting for PADS packets
原来还可以，前几天电拔了一下就不行了

InCommingMessage, 2016-09-26:
Timeout waiting for PADS packets
要么是时间不对->同步时间
要么是PIN码不对->找客服

另外拔电并不会保存配置，只有重启才会保存配置

qq734694920, 2016-09-26:
这种不是要配置静态路由么，应该怎么配置？

qq734694920, 2016-09-26:
Mon Sep 26 17:05:09 2016 daemon.notice netifd: Interface 'netkeeper' is now down
Mon Sep 26 17:05:09 2016 daemon.notice netifd: Interface 'netkeeper' is disabled
Mon Sep 26 17:05:09 2016 daemon.notice netifd: Interface 'netkeeper' is enabled
Mon Sep 26 17:05:09 2016 daemon.notice netifd: Interface 'netkeeper' is setting up now
Mon Sep 26 17:05:09 2016 daemon.notice netifd: VLAN 'eth0.2' link is down
Mon Sep 26 17:05:09 2016 daemon.notice netifd: Interface 'netkeeper' has link connectivity loss
Mon Sep 26 17:05:09 2016 daemon.notice netifd: VLAN 'eth0.2' link is up
Mon Sep 26 17:05:09 2016 daemon.notice netifd: Interface 'netkeeper' has link connectivity 
Mon Sep 26 17:05:09 2016 daemon.info pppd[3696]: Plugin rp-pppoe.so loaded.
Mon Sep 26 17:05:09 2016 daemon.info pppd[3696]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Mon Sep 26 17:05:09 2016 daemon.info pppd[3696]: Plugin sxplugin.so loaded.
Mon Sep 26 17:05:09 2016 daemon.info pppd[3696]: sxplugin : init
Mon Sep 26 17:05:09 2016 daemon.info pppd[3696]: -------------------------------------
Mon Sep 26 17:05:09 2016 daemon.info pppd[3696]: timenow(Hex)=%ßx
Mon Sep 26 17:05:09 2016 daemon.info pppd[3696]: Begin : beforeMD5
Mon Sep 26 17:05:09 2016 daemon.info pppd[3696]: 1.<??>
Mon Sep 26 17:05:09 2016 daemon.info pppd[3696]: 2.<??13854131095>
Mon Sep 26 17:05:09 2016 daemon.info pppd[3696]: 3.<??13854131095shandongmobile13>
Mon Sep 26 17:05:09 2016 daemon.info pppd[3696]: 4.length=<31>
Mon Sep 26 17:05:09 2016 daemon.info pppd[3696]: End : beforeMD5
Mon Sep 26 17:05:09 2016 daemon.info pppd[3696]: Begin : afterMD5
Mon Sep 26 17:05:09 2016 daemon.info pppd[3696]: 1.MD5use_1=< d>
Mon Sep 26 17:05:09 2016 daemon.info pppd[3696]: 2.MD5use_2=< 2>
Mon Sep 26 17:05:09 2016 daemon.info pppd[3696]: End : afterMD5
Mon Sep 26 17:05:09 2016 daemon.info pppd[3696]: -------------------------------------
Mon Sep 26 17:05:09 2016 daemon.info pppd[3696]: sxplugin : user  is <
FT"8^ d213854131095> 
Mon Sep 26 17:05:09 2016 daemon.info pppd[3696]: sxplugin : passwd loaded
Mon Sep 26 17:05:09 2016 daemon.notice pppd[3696]: pppd 2.4.7 started by root, uid 0
Mon Sep 26 17:05:09 2016 daemon.info pppd[3696]: PPP session is 3003
Mon Sep 26 17:05:09 2016 daemon.warn pppd[3696]: Connected to 64:3e:8c:41:f2:bb via interface eth0.2
Mon Sep 26 17:05:09 2016 daemon.info pppd[3696]: Using interface pppoe-netkeeper
Mon Sep 26 17:05:09 2016 daemon.notice pppd[3696]: Connect: pppoe-netkeeper <--> eth0.2
Mon Sep 26 17:05:12 2016 daemon.info pppd[3696]: syncppp not active
Mon Sep 26 17:05:13 2016 daemon.info pppd[3696]: Terminating on signal 15
Mon Sep 26 17:05:13 2016 daemon.notice pppd[3696]: Connection terminated.
Mon Sep 26 17:05:13 2016 daemon.info pppd[3696]: Exit.

miao1007, 2016-09-26:
网上找的：

Metric 是用于路由决策的一个参数，这里你只需要知道，当两个 Interfaces 都配置成 Default Gateway 的时候，通过 Metric 参数来决定哪个的优先级更高。

你现在要让你的PPPoE的网口的meric为0，而另一个网口的metric设置大一点，比如100

qq734694920, 2016-09-26:
谢谢，虽然听不懂，但是还是很感谢你 @miao1007