miao1007/Openwrt-NetKeeper #104: 新人请求帮助，网件 WNDR3800，明月固件，执行sh时出错。

Cumlnbus, 2016-10-31:
刚入门openwrt，不知如何解决，希望各位指教。
插件用的是hangzhou_sxplugin.so，错误信息如下：
/tmp$ chmod a+x /tmp/confnetwork.sh
/tmp$ sh /tmp/confnetwork.sh 
Warning: Section @zone[1] (wan) cannot resolve device of network 'netkeeper'
 * Flushing IPv4 filter table
 * Flushing IPv4 nat table
 * Flushing IPv4 mangle table
 * Flushing IPv4 raw table
 * Flushing IPv6 filter table
 * Flushing IPv6 nat table
 * Flushing IPv6 mangle table
 * Flushing IPv6 raw table
 * Flushing conntrack table ...
 * Populating IPv4 filter table
   * Zone 'lan'
   * Zone 'wan'
   * Rule '51413'
   * Rule '9091'
   * Rule '9000'
   * Rule '6800'
   * Rule 'Allow-DHCP-Renew'
   * Rule 'Allow-Ping'
   * Rule 'Allow-IGMP'
   * Rule #11
   * Rule #12
   * Forward 'lan' -> 'wan'
 * Populating IPv4 nat table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv4 mangle table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv4 raw table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv6 filter table
   * Zone 'lan'
   * Zone 'wan'
   * Rule '51413'
   * Rule '9091'
   * Rule '9000'
   * Rule '6800'
   * Rule 'Allow-DHCPv6'
   * Rule 'Allow-MLD'
   * Rule 'Allow-ICMPv6-Input'
   * Rule 'Allow-ICMPv6-Forward'
   * Rule #11
   * Rule #12
   * Forward 'lan' -> 'wan'
 * Populating IPv6 nat table
   * Zone 'lan'
Warning: fw3_ipt_rule_append(): Can't find target 'prerouting_lan_rule'
Warning: fw3_ipt_rule_append(): Can't find target 'postrouting_lan_rule'
   * Zone 'wan'
Warning: fw3_ipt_rule_append(): Can't find target 'prerouting_wan_rule'
Warning: fw3_ipt_rule_append(): Can't find target 'postrouting_wan_rule'
Warning: fw3_ipt_rule_append(): Can't find target 'delegate_prerouting'
Warning: fw3_ipt_rule_append(): Can't find target 'delegate_postrouting'
Warning: fw3_ipt_rule_append(): Can't find target 'prerouting_rule'
Warning: fw3_ipt_rule_append(): Can't find target 'postrouting_rule'
 * Populating IPv6 mangle table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv6 raw table
   * Zone 'lan'
   * Zone 'wan'
 * Set tcp_ecn to off
 * Set tcp_syncookies to on
 * Set tcp_window_scaling to on
 * Running script '/etc/firewall.user'
 * Running script '/usr/lib/gargoyle_firewall_util/gargoyle_additions.firewall'
 * Running script '/usr/share/miniupnpd/firewall.include'
 * Running script '/usr/share/qos_gargoyle/firewall.include'
0
   ! Failed with exit code 1
 * Running script '/usr/share/shadowsocks/firewall.include'

不尽感谢！

Cumlnbus, 2016-11-03:
成功拨上号，应该是Toolchain用错了。十分感谢你的工作：）

Quofataferunt, 2016-11-23:
同样的问题，请问是怎么具体解决的，谢谢