miao1007/Openwrt-NetKeeper #109: 插件能不能用在ARM架构下

HuLian21, 2016-12-13:
接下来可能收个树莓派3B用，用来处理KCP，做做NAS，MT7620还是太弱了。不知道能不能支持，刚开始学这方面。

opUserZangYing, 2016-12-13:
应该是可以的，在编译脚本的设置里我看见有设置交叉编译器位置，你把那个换成相应的ARM编译器就行了，就是可能后续还需要处理一些比如pppd的依赖之类//毕竟是pppd的插件

Neutree, 2017-03-11:
下对应的toochain,手动编译

HuLian21, 2017-03-24:
我在试着编译，现在有两个小问题。我下到的toochain是这个 
https://downloads.openwrt.org/barrier_breaker/14.07/brcm47xx/generic/OpenWrt-Toolchain-brcm47xx-for-mipsel_mips32-gcc-4.8-linaro_uClibc-0.9.33.2.tar.bz2
CPU BCM47189 ARM V7
那个编译器用mipsel吗？
mipsel有多个版本啊，选择那个呢

HuLian21, 2017-03-24:
@Neutree 编译出现 environment variable 'STAGING_DIR' not defined
这个变量问题怎么解决啊

Neutree, 2017-03-24:
@HuLian21 如果是树莓派3B而且系统是14.07的话，用[这个](https://downloads.openwrt.org/snapshots/trunk/brcm2708/generic/OpenWrt-Toolchain-brcm2708-bcm2708_gcc-5.3.0_musl-1.1.16_eabi.Linux-x86_64.tar.bz2)，BCM47xx选哪个版本，你可以
```
cat /proc/cpuinfo
```
查看`cpu model`字段

xratzh, 2017-04-02:
老哥可以试试我给梅林编译的so文件，也是博通armv7的处理器，应该能通用的。地址在此：https://github.com/xratzh/hiwifi-netkeeper/tree/master/src
我的梅林双核1G，256ram也跑不动kcptun，一会儿就会断。所以放弃kcptun了

lee86, 2017-04-04:
我只有一个梅林固件的网件r6400，就是搞不清楚怎么弄的

xratzh, 2017-04-05:
@lee86 我写在readme里面了，麻烦你仔细看

HuLian21, 2017-04-12:
@xratzh 好的，那么对于梅林固件，能通过直接修改文件方式启用插件么，类似openwrt 里修改network 文件

xratzh, 2017-04-12:
@HuLian21 做好了，你测试一下吧，应该问题不大，主要是梅林会断电丢失时间，所以梅林的路由器时间一定要准确。搞好后可以手动重新拨号试试。

HuLian21, 2017-04-14:
@xratzh 对了，我想问下，我从同学买的破解路由器里提取到3个文件，看卖家介绍是解决心跳的，其中的达到了244KB，按理说心跳式不用那么大，并且按卖家说还是转发远程服务器的心跳到路由器再转给电信的，我觉得有点诡异，你有兴趣看下吗？我可以传给你。
PS. 一个文件为.cer的key文件，感觉有点意思，可惜不懂反汇编。