miao1007/Openwrt-NetKeeper #111: 如何在一个路由器用不同的帐号登录来叠加网速啊，

ljq917181927, 2017-01-02:
如题，我把sh文件中的uci delete这两行删除，然后把下边所有netkepper改成netkepper2，mac地址也改了。但是执行之后一会我一个拨号成功了，查看系统日志另一个daemon.notice pppd[32117]: pppd 2.4.7 started by root, uid 0 往下就没有了

Destinyzxn, 2017-01-11:
首先是op  然后装上插件  然后配置op里面的交换机  再然后需要修改network里面的东西  ，最后 还需要配置负载均衡  最后完成拨号 负载上线 网速就可以叠加了

1shanyue1, 2017-03-24:
路由器刷个潘多拉固件，里面可以田间vwan，每个vwan添加一个账号，就可以叠加了，

StchaoL, 2017-07-14:
我的做法是：安装**kmod-macvlan**驱动和**mwan3**. 当然直接装一个支持多拨的潘多拉就行。
然后使用 ``` ip  link add link eth0.2 name macvlan1 type macvlan ``` 新建一个虚拟wan口，下一步修改**/etc/config/network** 文件，新建一个接口并指定其**option ifname 'macvlan1"** ，其余选项与正常的netkeeper拨号接口类似，接下来从mwan3里添加接口、配置成员、添加规则，即可多拨。

NeuraxLin, 2017-11-09:
@StchaoL 我设置了新的接口和链接选项 填写了不同的账号 仍然只能使原有的那一条线路链接 难道一个Wan口不行要去把Lan虚拟到Wan再接一根线嘛？
日志里有这么一个提示貌似是我没做好哪一步？
Couldn't rename ppp1 to pppoe-netkeeper
 Timeout waiting for PADO packets
Unable to complete PPPoE Discovery
Mwan的设置我能理解
但是现在二线连不上呀：(

StchaoL, 2017-11-09:
试试macvlan, 安装 ``` luci-app-syncdial ```