miao1007/Openwrt-NetKeeper #112: (wan) cannot resolve device of network 'wan'

caiqiqi, 2017-02-20:
版本号，OpenWrt Barrier Breaker 14.07 r45620 五一版 by明月永在

```
root@OpenWrt:/tmp# sh confnetwork.sh
uci: Entry not found
uci: Entry not found
Warning: Section @zone[1] (wan) cannot resolve device of network 'wan'
Warning: Section @zone[1] (wan) cannot resolve device of network 'netkeeper'
Warning: Section @zone[1] (wan) has no device, network, subnet or extra options
 * Flushing IPv4 filter table
 * Flushing IPv4 nat table
 * Flushing IPv4 mangle table
 * Flushing IPv4 raw table
 * Flushing conntrack table ...
 * Populating IPv4 filter table
   * Zone 'lan'
   * Zone 'wan'
   * Rule 'Allow-DHCP-Renew'
   * Rule 'Allow-Ping'
   * Forward 'lan' -> 'wan'
 * Populating IPv4 nat table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv4 mangle table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv4 raw table
   * Zone 'lan'
   * Zone 'wan'
 * Set tcp_ecn to off
 * Set tcp_syncookies to on
 * Set tcp_window_scaling to on
 * Running script '/etc/firewall.user'
 * Running script '/usr/lib/gargoyle_firewall_util/gargoyle_additions.firewall'
 * Running script '/usr/share/miniupnpd/firewall.include'
 * Running script '/usr/share/colorbox/firewall.include'
 * Running script '/usr/share/qos_gargoyle/firewall.include'
0
   ! Failed with exit code 1
 * Running script '/usr/share/redsocks2/firewall.include'
redsocks2 is not enabled.Exit now.
   ! Failed with exit code 1
 * Running script '/usr/share/shadowsocks/firewall.include'
```

caiqiqi, 2017-02-20:
刚到的路由器，请教大神这个是什么问题啊

Destinyzxn, 2017-02-22:
虽然看不懂是什么 我可以给你一个方法  直接去修改network文件 来达到调用插件

caiqiqi, 2017-02-22:
谢谢师兄。我问学校的另外一个用openwrt的同学，他帮我搞定了。没有用那个confnetwork.sh，
直接通过web界面配置的。貌似如果开始不同步好时间就连接了netkeeper，可能会被电信禁止连接一段时间，也许是这个原因困扰了很久。最后他在/etc/config/network的netkeeper的接口上加了一句
option auto '0'。这样就不会在浏览器还没有同步时间的时候就连接netkeeper了。

P9J7, 2017-02-25:
可否详细说一下如何通过web界面配置？我也遇到了相同问题

Destinyzxn, 2017-02-25:
[op使用netkeeper .docx](https://github.com/miao1007/Openwrt-NetKeeper/files/801303/op.netkeeper.docx)

Destinyzxn, 2017-02-26:
你仔细检查一下吧   按照这个文档通常都不会出错

P9J7, 2017-02-26:
我感觉是so库的问题，不确定heibei_plugin.so是否为河北联通

Destinyzxn, 2017-02-26:
这个我就不能知道了  你可以在网上多找找  cpu型号对应就好

P9J7, 2017-02-26:
CPU为MT7620A，同一个省份的不同运营商so库通用？

Destinyzxn, 2017-02-26:
网上都只有一个河北的so文件