miao1007/Openwrt-NetKeeper #118: 湖北e信开了心跳，可不可以增加一个心跳功能

mn0604, 2017-03-10:
求助大神，现在开了心跳，几分钟就要掉线一次，怎么办

sunflyer, 2017-03-10:
你想下这个项目在淘宝被卖成什么样子就知道可不可以了。

Sung-yy, 2017-03-10:
并不是所有人都这样，我们也是来湖北读书的，那这些卖路由器的人到底是来了湖北这样的还是生来如此，谁知道这些卖路由器的人是什么心态。

Sung-yy, 2017-03-10:
你这是在地域黑？你清楚那些卖路由器的是什么来路，就一定是湖北人？其他省份就没有这样的事件？不是就你一个人会破解……

Sung-yy, 2017-03-10:
@sunflyer

sunflyer, 2017-03-10:
@Sung-yy  我在地域黑？搞笑。连我意思都没搞懂就在说地域黑，这么跳脚？哪一句碰到你的G点了？

这套东西开源以后贵省卖破解路由器的低于150卖都会被同行骂的要死，当年IPV6不限速的时候甚至卖到300，诞生不少拿着这套东西骗人的家伙。“IPV6不限速校园网破解专用版”整个淘宝满街跑。

何况当年我做Simple Netkeeper被贵省卖路由器的卖了多少钱你怎么知道？如果你自己写的软件清清楚楚的协议不得拿来商用偏偏一大票人跑出来当着你的面违反协议还骂你你是什么想法？你说他魔改你APP了连传单的宣传语都抄袭你的网站的东西他还嚣张“你他妈哪只眼睛看到我卖你软件了？”你心里什么想法？

这可都是贵省人民做的事情哦。我实话实说有问题么？

原作者在repo里面写了GPL协议no taobao use，有几个人真正遵守的？你觉得你清白了你们省的其他人就一定清白了？

如果这时候给你们心跳插件，你觉得这群人会比以前少还是比以前更猖狂？“校园网防心跳路由器质保三年”想想都很感人呐。

对，不是就我一个人会破解，心跳原理什么的本来就都在这个repo里面，你们有本事就请你们自己弄，不然请你们另请高明，反正隔壁某左软件50一份对你们不是也挺好的吗，呵呵。

mn0604, 2017-03-10:
@sunflyer 大神，我只是不想让电信这么流氓的赚钱，就想着宿舍大家一起用，那些拿着开源的东西去卖钱的人我也是看不起，好东西就应该大家一起分享，不应该用于商业。希望大神能帮帮我。

zaqzxc, 2017-03-11:
有人研究出服务器转发心跳了，绑定账号的，用于商业的，大神帮帮我们。。。

left906080850, 2017-05-09:
@sunflyer The innocent being hurt （tang qiang）

sunflyer, 2017-05-09:
@left906080850 呀，dalao

miao1007, 2017-05-15:
没时间啊...我日常9106呢