miao1007/Openwrt-NetKeeper #121: 湖北云心跳服务器搭建教程

zaqzxc, 2017-03-22:
0

tml><head>
<title>301 Moved Permanently</title>
</head><body>
<h1>Moved Permanently</h1>
<p>The document has moved <a href="http://123.207.13.194/ip/">here</a>.</p>
</body></html>
..............T....._...g.C.:.\.W.i.n.d.o.w.s.\.s.y.s.t.e.m.3.2.\.w.s.2._.3.2...D.L.L.....................................................................................................................................................................................................................................................................................................................................................................................................................................................................................-..*......_..gg..http://123.207.13.194/ip/....d.....A...h.t.t.p.:././.1.2.3...2.0.7...1.3...1.9.4./.i.p./.......................<.d.P.d.l
d.T.d.............P(g.^..\S....._..@e...............d...d.....Y..[^....._..@e.................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................o..nD....._..@e..........................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................._..@e.....................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................V...h.t.t.p.:././.1.2.3...2.0.7...1.3...1.9.4./.i.p./...................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................&....._..._.8._.8._..@g..................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................].....c...............c.........D............*.w....................@.g.@.g.....@.g.........@.g.@.g.@.g...................................c......................0c.(r............................b.....P...........(lc.PK..................
[... cut ...]
.........b.............t.e...............................................................................................c.................................@.g.........._c.....................
.....c.....................
...._c.....................
.........b.............t.e...............................................................................................c.................................@.g.........._c.....................
.....c.....................
...._c.....................
.........b.............t.e........................

https://item.taobao.com/item.htm?spm=a230r.1.14.55.OpkKmy&id=547118198649&ns=1&abbucket=19#detail

closed 2017-03-22 by miao1007

sunflyer, 2017-03-24:
这么嚣张？厉害了。

theflash2017, 2017-04-10:
这个是教程还是收费广告呀？