miao1007/Openwrt-NetKeeper #125: 上路由器日志,帮忙看一下什么问题!

Sung-yy, 2017-03-25:
经常掉线!有些地方我不理解!
Sat Mar 25 12:53:15 2017 daemon.info dnsmasq[18902]: read /tmp/hosts/dhcp.cfg02411c - 1 addresses
Sat Mar 25 12:53:15 2017 daemon.info dnsmasq-dhcp[18902]: read /etc/ethers - 0 addresses
Sat Mar 25 12:53:16 2017 cron.info crond[18923]: crond (busybox 1.24.1) started, log level 8
Sat Mar 25 12:53:17 2017 daemon.notice netifd: Interface 'wan' has link connectivity loss
Sat Mar 25 12:53:17 2017 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Sat Mar 25 12:53:17 2017 daemon.notice netifd: Interface 'netkeeper_6' is now down
Sat Mar 25 12:53:17 2017 daemon.notice netifd: Interface 'netkeeper_6' is disabled
Sat Mar 25 12:53:17 2017 daemon.notice netifd: Interface 'netkeeper_6' has link connectivity loss
Sat Mar 25 12:53:17 2017 daemon.notice netifd: Network alias '' link is down
Sat Mar 25 12:53:17 2017 daemon.warn dnsmasq[18902]: no servers found in /tmp/resolv.conf.auto, will retry
Sat Mar 25 12:53:17 2017 daemon.info pppd[18856]: Terminating on signal 15
Sat Mar 25 12:53:17 2017 daemon.info pppd[18856]: Connect time 0.1 minutes.
Sat Mar 25 12:53:17 2017 daemon.info pppd[18856]: Sent 511 bytes, received 1648 bytes.
Sat Mar 25 12:53:17 2017 daemon.notice netifd: Network device 'pppoe-netkeeper' link is down
Sat Mar 25 12:53:17 2017 daemon.err miniupnpd[5246]: Failed to get IP for interface pppoe-netkeeper
Sat Mar 25 12:53:17 2017 daemon.warn miniupnpd[5246]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Sat Mar 25 12:53:17 2017 daemon.notice pppd[18856]: Connection terminated.
Sat Mar 25 12:53:17 2017 daemon.info pppd[18856]: Sent PADT
Sat Mar 25 12:53:18 2017 daemon.info pppd[18856]: Exit.
Sat Mar 25 12:53:18 2017 daemon.notice netifd: netkeeper (18856): connect as d786e201
Sat Mar 25 12:53:18 2017 daemon.notice netifd: Interface 'netkeeper' is now down
Sat Mar 25 12:53:18 2017 daemon.notice netifd: Interface 'netkeeper' is disabled
Sat Mar 25 12:53:18 2017 daemon.notice netifd: Interface 'netkeeper' is enabled
Sat Mar 25 12:53:18 2017 daemon.notice netifd: Interface 'netkeeper' is setting up now
Sat Mar 25 12:53:18 2017 daemon.info dnsmasq[18902]: exiting on receipt of SIGTERM
Sat Mar 25 12:53:18 2017 daemon.info pppd[19143]: Plugin rp-pppoe.so loaded.
Sat Mar 25 12:53:18 2017 daemon.info pppd[19143]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Sat Mar 25 12:53:18 2017 daemon.info pppd[19143]: Plugin wuhan_sxplugin.so loaded.
Sat Mar 25 12:53:18 2017 daemon.info pppd[19143]: sxplugin : init
Sat Mar 25 12:53:18 2017 daemon.info pppd[19143]: -------------------------------------
Sat Mar 25 12:53:18 2017 daemon.info pppd[19143]: timenow(Hex)=58d5f7be
Sat Mar 25 12:53:18 2017 daemon.info pppd[19143]: Begin : beforeMD5
Sat Mar 25 12:53:18 2017 daemon.info pppd[19143]: 1.<?d?>
Sat Mar 25 12:53:18 2017 daemon.info pppd[19143]: 2.<?d?77777777777>
Sat Mar 25 12:53:18 2017 daemon.info pppd[19143]: 3.<?d?77777777777hubtxinli01>
Sat Mar 25 12:53:18 2017 daemon.info pppd[19143]: 4.length=<26>
Sat Mar 25 12:53:18 2017 daemon.info pppd[19143]: End : beforeMD5
Sat Mar 25 12:53:18 2017 daemon.info pppd[19143]: Begin : afterMD5
Sat Mar 25 12:53:18 2017 daemon.info pppd[19143]: 1.MD5use_1=< 1>
Sat Mar 25 12:53:18 2017 daemon.info pppd[19143]: 2.MD5use_2=< f>
Sat Mar 25 12:53:18 2017 daemon.info pppd[19143]: End : afterMD5
Sat Mar 25 12:53:18 2017 daemon.info pppd[19143]: -------------------------------------
Sat Mar 25 12:53:18 2017 daemon.info pppd[19143]: sxplugin : user  is <
Q^"]R 1f77777777777@hkd> 
Sat Mar 25 12:53:18 2017 daemon.info pppd[19143]: sxplugin : passwd loaded
Sat Mar 25 12:53:18 2017 daemon.warn pppd[19143]: [pppoe hook debug] syscall operations name probe pandorabox ok!
Sat Mar 25 12:53:18 2017 daemon.warn pppd[19143]: [pppoe hook debug] hook status=1
Sat Mar 25 12:53:18 2017 daemon.notice pppd[19143]: pppd 2.4.7 started by root, uid 0
Sat Mar 25 12:53:18 2017 daemon.info pppd[19143]: PPP session is 8474
Sat Mar 25 12:53:18 2017 daemon.warn pppd[19143]: Connected to 00:46:4b:71:8a:98 via interface eth0.2
Sat Mar 25 12:53:18 2017 daemon.info pppd[19143]: Using interface pppoe-netkeeper
Sat Mar 25 12:53:18 2017 daemon.notice pppd[19143]: Connect: pppoe-netkeeper <--> eth0.2
Sat Mar 25 12:53:18 2017 daemon.info pppd[19143]: syncppp not active
Sat Mar 25 12:53:18 2017 daemon.info pppd[19143]: Remote message: 18;User(77777777777@hkd) Checking LoginUser RBillServer, return Called Number Not Allowed, Request Deny by hb.cn
Sat Mar 25 12:53:18 2017 daemon.err pppd[19143]: PAP authentication failed
Sat Mar 25 12:53:22 2017 daemon.info dnsmasq[19267]: started, version 2.76 cachesize 150
Sat Mar 25 12:53:22 2017 daemon.info dnsmasq[19267]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack ipset no-auth no-DNSSEC loop-detect inotify
Sat Mar 25 12:53:22 2017 daemon.info dnsmasq[19267]: DNS service limited to local subnets
Sat Mar 25 12:53:22 2017 daemon.info dnsmasq-dhcp[19267]: DHCP, IP range 192.168.1.100 -- 192.168.1.249, lease time 12h
Sat Mar 25 12:53:22 2017 daemon.info dnsmasq[19267]: using nameserver 114.114.114.114#53 for domain valf.atm.youku.com 
Sat Mar 25 12:53:22 2017 daemon.info dnsmasq[19267]: using nameserver 114.114.114.114#53 for domain apple.com 
Sat Mar 25 12:53:22 2017 daemon.info dnsmasq[19267]: using local addresses only for domain lan
Sat Mar 25 12:53:22 2017 daemon.warn dnsmasq[19267]: no servers found in /tmp/resolv.conf.auto, will retry
Sat Mar 25 12:53:22 2017 daemon.info dnsmasq[19267]: read /etc/hosts - 1 addresses
Sat Mar 25 12:53:22 2017 daemon.info dnsmasq[19267]: read /tmp/hosts/dhcp.cfg02411c - 2 addresses
Sat Mar 25 12:53:22 2017 daemon.info dnsmasq-dhcp[19267]: read /etc/ethers - 0 addresses
Sat Mar 25 12:53:24 2017 daemon.notice pppd[19143]: Connection terminated.
Sat Mar 25 12:53:24 2017 daemon.info pppd[19143]: Sent PADT
Sat Mar 25 12:53:24 2017 daemon.notice pppd[19143]: Modem hangup
Sat Mar 25 12:53:24 2017 daemon.info pppd[19143]: Exit.
Sat Mar 25 12:53:24 2017 daemon.notice netifd: netkeeper (19143): connect as cb25153e
Sat Mar 25 12:53:24 2017 daemon.notice netifd: Interface 'netkeeper' is now down
Sat Mar 25 12:53:24 2017 daemon.notice netifd: Interface 'netkeeper' is disabled
Sat Mar 25 12:53:24 2017 daemon.notice netifd: Interface 'netkeeper' is enabled
Sat Mar 25 12:53:24 2017 daemon.notice netifd: Interface 'netkeeper' is setting up now
Sat Mar 25 12:53:25 2017 daemon.info pppd[19469]: Plugin rp-pppoe.so loaded.
Sat Mar 25 12:53:25 2017 daemon.info pppd[19469]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Sat Mar 25 12:53:25 2017 daemon.info pppd[19469]: Plugin wuhan_sxplugin.so loaded.
Sat Mar 25 12:53:25 2017 daemon.info pppd[19469]: sxplugin : init
Sat Mar 25 12:53:25 2017 daemon.info pppd[19469]: -------------------------------------
Sat Mar 25 12:53:25 2017 daemon.info pppd[19469]: timenow(Hex)=58d5f7c5
Sat Mar 25 12:53:25 2017 daemon.info pppd[19469]: Begin : beforeMD5
Sat Mar 25 12:53:25 2017 daemon.info pppd[19469]: 1.<?d?>
Sat Mar 25 12:53:25 2017 daemon.info pppd[19469]: 2.<?d?77777777777>
Sat Mar 25 12:53:25 2017 daemon.info pppd[19469]: 3.<?d?77777777777hubtxinli01>
Sat Mar 25 12:53:25 2017 daemon.info pppd[19469]: 4.length=<26>
Sat Mar 25 12:53:25 2017 daemon.info pppd[19469]: End : beforeMD5
Sat Mar 25 12:53:25 2017 daemon.info pppd[19469]: Begin : afterMD5
Sat Mar 25 12:53:25 2017 daemon.info pppd[19469]: 1.MD5use_1=< c>
Sat Mar 25 12:53:25 2017 daemon.info pppd[19469]: 2.MD5use_2=< 6>
Sat Mar 25 12:53:25 2017 daemon.info pppd[19469]: End : afterMD5
Sat Mar 25 12:53:25 2017 daemon.info pppd[19469]: -------------------------------------
Sat Mar 25 12:53:25 2017 daemon.info pppd[19469]: sxplugin : user  is <
AR!]1 c677777777777@hkd> 
Sat Mar 25 12:53:25 2017 daemon.info pppd[19469]: sxplugin : passwd loaded
Sat Mar 25 12:53:25 2017 daemon.warn pppd[19469]: [pppoe hook debug] syscall operations name probe pandorabox ok!
Sat Mar 25 12:53:25 2017 daemon.warn pppd[19469]: [pppoe hook debug] hook status=1
Sat Mar 25 12:53:25 2017 daemon.notice pppd[19469]: pppd 2.4.7 started by root, uid 0
Sat Mar 25 12:53:25 2017 daemon.info pppd[19469]: PPP session is 24259
Sat Mar 25 12:53:25 2017 daemon.warn pppd[19469]: Connected to 00:46:4b:71:8a:98 via interface eth0.2
Sat Mar 25 12:53:25 2017 daemon.info pppd[19469]: Using interface pppoe-netkeeper
Sat Mar 25 12:53:25 2017 daemon.notice pppd[19469]: Connect: pppoe-netkeeper <--> eth0.2
Sat Mar 25 12:53:25 2017 daemon.info pppd[19469]: syncppp not active
Sat Mar 25 12:53:25 2017 daemon.info pppd[19469]: Remote message: 0;User(77777777777@hkd) Authenticate OK, Request Accept by hb.cn
Sat Mar 25 12:53:25 2017 daemon.notice pppd[19469]: PAP authentication succeeded

Sung-yy, 2017-03-25:
[image]

sunflyer, 2017-03-25:
湖北电信心跳检测所以掉线。

Sung-yy, 2017-03-25:
那是不是一直这样掉线,连线.

Sung-yy, 2017-03-25:
那怎么解决呢?

Sung-yy, 2017-03-25:
谢谢大佬解答！

sunflyer, 2017-03-29:
我发现你这个帐号最近的活动基本集中在Netkeeper这块，而且star & fork的项目全是Netkeeper的东西。
而且上来直接问到了心跳算法。

再想想之前问问题那个样子。。。我个人有点怀疑你拿了算法到底是拿来做什么的。
反正心跳也已经pull了，如你所说的，不是只有我一个人会破解，代码都已经show 给你了，那就请你自己动手呗。

Sung-yy, 2017-03-29:
我拿了自用,没心情去坑别人!

Sung-yy, 2017-04-04:
[image]
这个wan_6是个怎么回事啊,我记得没有这个接口啊!

Sung-yy, 2017-04-04:
还有,为什么一直会出现获取ip地址失败Failed to get IP for interface pppoe-wan

Sung-yy, 2017-04-14:
[image]大佬这是什么,一直拨不上去号.