miao1007/Openwrt-NetKeeper #127: 那个心跳插件怎么编译啊？

shitou001, 2017-03-31:
In file included from main.c:19:0:
netkeeper.h:15:25: fatal error: mbedtls/aes.h: No such file or directory
 #include <mbedtls/aes.h>
                         ^
compilation terminated.
Makefile:9: recipe for target 'main.o' failed
make: *** [main.o] Error 1

zaqzxc, 2017-03-31:
交叉编译

shitou001, 2017-03-31:
具体怎么搞，那个so文件编译成功了，这个main.c编译出错了

sunflyer, 2017-04-01:
最好先看代码，别编译错版本了。

shitou001, 2017-04-01:
原谅我根本看不懂代码😂

zaqzxc, 2017-04-01:
贴运行make后面代码

shitou001, 2017-04-01:
@zaqzxc
 /home/xx/luyouqi/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_uClibc-0.9.33.2/bin/mipsel-openwrt-linux-gcc -o main.o main.c -g -Wall -Wimplicit-function-declaration -I /include/  -c
mipsel-openwrt-linux-uclibc-gcc.bin: warning: environment variable 'STAGING_DIR' not defined
In file included from main.c:19:0:
netkeeper.h:15:25: fatal error: mbedtls/aes.h: No such file or directory
 #include <mbedtls/aes.h>
                         ^
compilation terminated.
Makefile:9: recipe for target 'main.o' failed
make: *** [main.o] Error 1

sunflyer, 2017-04-01:
你还需要mbedtls的库。

sunflyer, 2017-04-01:
另外呢你没必要去艾特那位，那位估计是湖北的某个商人？

Sung-yy, 2017-04-03:
@sunflyer 问一下,main.c里的注释里的key有用吗,我编译的时候需要修改吗,还有那个makefile里的change radius在哪里修改啊,需不需要修改?

BabyGH, 2017-04-03:
@shitou001 您好，我也在尝试编译SO文件，但这个步骤“Read and edit makefile and confnetwork.sh TODOS carefully”只用修改这两个文件吗？makefile中在文本哪处Change Radius还有Change PREFIX1？还请您指点，谢谢哈

shitou001, 2017-04-03:
搞了几天还是没弄好，我要放弃了，先学好c语言能看懂源码再说

closed 2017-04-03 by shitou001

BabyGH, 2017-04-03:
@shitou001 您那so不是编译好了吗 可以告诉我咋弄吗

shitou001, 2017-04-03:
搭建好交叉编译环境，根据自己的需要修改需要修改的地方，makefile写的很清楚，然后make all就可以了

BabyGH, 2017-04-03:
@shitou001 关键是有些地方不知道要不要改 以及怎末该 看不懂啊 您可以加我q一下吗1752167203

BabyGH, 2017-04-04:
@shitou001 还是编译不成功啊 能教教我吗

theflash2017, 2017-04-05:
你编译成功了吗？