miao1007/Openwrt-NetKeeper #130: 自己尝试编译so文件但不知在哪修改Makeflie中的TODO: change RADIUS和 TODO: Change PREFIX1

BabyGH, 2017-04-07:
TODO: change RADIUS
# 
# ChongQing Netkeeper: cqxinliradius002
# ChongQing Netkeeper(0094): xianxinli1radius
# WuHan E xin: hubtxinli01
# Hangzhou(Tested on HDU): singlenet01
# qinghai 3.6.27: shd@xiaoyuan0002
# Xinjiang 3.7.3 : xinjiang@0724
# hebei: hebeicncxinli002
# ShanDong Mobile : shandongmobile13
# Shanxi Yixun : sh_xi@xiaoyuan01
# GanSu Telecom 3.7.1 : xiaoyuanyixun001

##  The following radius has been deprecated due to update. No more used.
#
# ShanDong Telecom : 560Ox!a0yuanOlIz  (For 3.7.3 Augest Version)
# ShanDong Telecom : shdOx!a0yuan01lz  (For 3.7.3 October 27th Version)
# NanChangV12~V17: radius
# NanChangV18~V29: nanchang3.0
# NanChangV32: jiangxi4.0
#

# TODO: Change PREFIX1
# 
# NanChangV32: '1'
# Others: '\n'
另请明示，只需要修改makefile就行吗

xratzh, 2017-04-09:
readme里面写清楚了
git clone后，再下载toolchain解压缩（完全照着readme里面做的）
在makefile里更改，LOCATION=~/OpenWrt-Toolchain-ramips-for-mipsel_24kec+dsp-gcc-4.8-linaro_uClibc-0.9.33.2/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_uClibc-0.9.33.2，这一段要根据你自己的toolchain的位置做修改。其他的基本不需要修改。直接在src/下make all就好了。
BTW，我自己用的ubuntu14.04，因为编译梅林的相关固件16.04会有问题，不知道openwrt是否也会这样，未测试

BabyGH, 2017-04-10:
@xratzh 我用的最新的ubuntu16 我想问的是 这段代码change RADIUS和 TODO: Change PREFIX1改不改

[image]

BabyGH, 2017-04-10:
`#TODO:可能需要修改pppd的版本，取决于路由器的/usr/lib/pppd/，目前主流的14.09是2.4.7
PPPD_VER=2.4.7
CFLAGS=-Os -Werror -I$(LOCATION)/include -fPIC -DPPPOE_VER='"${PPPD_VER}"'

# TODO: change RADIUS
# 
# ChongQing Netkeeper: cqxinliradius002
# ChongQing Netkeeper(0094): xianxinli1radius
# WuHan E xin: hubtxinli01
# Hangzhou(Tested on HDU): singlenet01
# qinghai 3.6.27: shd@xiaoyuan0002
# Xinjiang 3.7.3 : xinjiang@0724
# hebei: hebeicncxinli002
# ShanDong Mobile : shandongmobile13
# Shanxi Yixun : sh_xi@xiaoyuan01
# GanSu Telecom 3.7.1 : xiaoyuanyixun001

##  The following radius has been deprecated due to update. No more used.
#
# ShanDong Telecom : 560Ox!a0yuanOlIz  (For 3.7.3 Augest Version)
# ShanDong Telecom : shdOx!a0yuan01lz  (For 3.7.3 October 27th Version)
# NanChangV12~V17: radius
# NanChangV18~V29: nanchang3.0
# NanChangV32: jiangxi4.0
#

# TODO: Change PREFIX1
# 
# NanChangV32: '1'
# Others: '\n'`

BabyGH, 2017-04-10:
@xratzh make all后提示如下
../../OpenWrt-Toolchain-ramips-for-mipsel_24kec+dsp-gcc-4.8-linaro_uClibc-0.9.33.2/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_uClibc-0.9.33.2/bin/mipsel-openwrt-linux-gcc: line 82: /home/liulibin/OpenWrt-Toolchain-ramips-for-mipsel_24kec+dsp-gcc-4.8-linaro_uClibc-0.9.33.2/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_uClibc-0.9.33.2/bin/mipsel-openwrt-linux-uclibc-gcc.bin: cannot execute binary file: Exec format error
../../OpenWrt-Toolchain-ramips-for-mipsel_24kec+dsp-gcc-4.8-linaro_uClibc-0.9.33.2/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_uClibc-0.9.33.2/bin/mipsel-openwrt-linux-gcc: line 82: /home/liulibin/OpenWrt-Toolchain-ramips-for-mipsel_24kec+dsp-gcc-4.8-linaro_uClibc-0.9.33.2/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_uClibc-0.9.33.2/bin/mipsel-openwrt-linux-uclibc-gcc.bin: Success
make: *** [chongqing_sxplugin] Error 126

xratzh, 2017-04-10:
我刚在vps上重装了ubuntu16.04.2，root登陆后，按照顺序：
1、git clone --depth=1 https://github.com/miao1007/Openwrt-NetKeeper.git
2、wget https://downloads.openwrt.org/barrier_breaker/14.07/ramips/mt7620a/OpenWrt-Toolchain-ramips-for-mipsel_24kec+dsp-gcc-4.8-linaro_uClibc-0.9.33.2.tar.bz2 && tar -xjvf  OpenWrt-Toolchain-ramips-for-mipsel_24kec+dsp-gcc-4.8-linaro_uClibc-0.9.33.2.tar.bz2
3、cd  Openwrt-NetKeeper/src/
4、make all(没装make包的话会提示，就apt install make -y安装再执行)

一切正常，编译出来了。你可以再试试。实在不行我的github有前几天编译好的。

BabyGH, 2017-04-10:
@xratzh  make 包一直装着呢我什么文件都没改  刚按你的试了试还是一样的错误提示  出不来SO

BabyGH, 2017-04-10:
@xratzh 如果什么都不改的话各个省份的都默认编译出来吗

xratzh, 2017-04-10:
@BabyGH 对，make all后会把所有省份都编译出来，到时候文件自动出现在src下。

BabyGH, 2017-04-10:
@xratzh 我换一个14.04的会成功吗

xratzh, 2017-04-10:
@BabyGH 你有vps可以在vps上测试和编译，但是我之所以之前是14.04是编译梅林需要安装很多依赖，16和14的依赖软件包不同。但是编译这个不需要依赖，所以系统版本不是问题，我自己不也是在16.04测试成功了么。你重新下载toochain吧。我觉得是网络问题，所以推荐在vps上做这个。

BabyGH, 2017-04-10:
@VPS全称是什么 我是在虚拟机立的linux

xratzh, 2017-04-10:
虚拟专用服务器，虚拟机也可以，但我觉得你应该是网络问题。

BabyGH, 2017-04-10:
@xratzh 网络没问题 都能下载的

BabyGH, 2017-04-12:
@xratzh 可以加我Q吗1752167203

BabyGH, 2017-04-19:
@xratzh 还是编译不成啊

xratzh, 2017-04-19:
@BabyGH 我发给你QQ邮箱一个vps账号，你用了没？
我自己用的那个编译成功了的，如何使用我给你说清楚了
1、git clone --depth=1 https://github.com/miao1007/Openwrt-NetKeeper.git
2、wget https://downloads.openwrt.org/barrier_breaker/14.07/ramips/mt7620a/OpenWrt-Toolchain-ramips-for-mipsel_24kec+dsp-gcc-4.8-linaro_uClibc-0.9.33.2.tar.bz2 && tar -xjvf OpenWrt-Toolchain-ramips-for-mipsel_24kec+dsp-gcc-4.8-linaro_uClibc-0.9.33.2.tar.bz2
3、cd Openwrt-NetKeeper/src/
4、make all(没装make包的话会提示，就apt install make -y安装再执行)
直接复制粘贴这些命令，一步一步来

closed 2017-05-05 by BabyGH