miao1007/Openwrt-NetKeeper #133: 重庆邮电大学移通学院更新了过后用不了了。（已上传更新后客户端）

qq624966793, 2017-05-04:
http://pan.baidu.com/s/1slhPE2L
首先感谢大神，让我能够使用路由器。。。
但是最近学校更新了客户端。。然后就用不了了。
我这个金融系的，是心有余而力不足。。求大神帮助
不知道还需要什么东西。。。

qq624966793, 2017-05-04:
@sunflyer

chingcq, 2017-05-04:
重庆强制更新4.x版本了 2.5版本认证已经关掉

qq624966793, 2017-05-04:
ed May  3 13:28:49 2017 daemon.warn pppd[6110]: [pppoe hook debug] hook status=1
Wed May  3 13:28:49 2017 daemon.notice pppd[6110]: pppd 2.4.7 started by root, uid 0
Wed May  3 13:28:49 2017 daemon.info pppd[6110]: PPP session is 11930
Wed May  3 13:28:49 2017 daemon.warn pppd[6110]: Connected to 28:31:52:57:17:58 via interface eth0.2
Wed May  3 13:28:49 2017 daemon.info pppd[6110]: Using interface pppoe-netkeeper
Wed May  3 13:28:49 2017 daemon.notice pppd[6110]: Connect: pppoe-netkeeper <--> eth0.2
Wed May  3 13:28:52 2017 daemon.info pppd[6110]: syncppp not active
Wed May  3 13:28:53 2017 daemon.info pppd[6110]: CHAP authentication failed: old pin code algorithm ##old pin code algorithm 
Wed May  3 13:28:53 2017 daemon.err pppd[6110]: CHAP authentication failed
Wed May  3 13:28:59 2017 daemon.notice pppd[6110]: Connection terminated.
Wed May  3 13:28:59 2017 daemon.info pppd[6110]: Sent PADT
Wed May  3 13:28:59 2017 daemon.notice pppd[6110]: Modem hangup
Wed May  3 13:28:59 2017 daemon.info pppd[6110]: Exit.
Wed May  3 13:28:59 2017 daemon.notice netifd: netkeeper (6110): connect as 6aec2e65
Wed May  3 13:28:59 2017 daemon.notice netifd: Interface 'netkeeper' is now down
Wed May  3 13:28:59 2017 daemon.notice netifd: Interface 'netkeeper' is setting up now
Wed May  3 13:28:59 2017 daemon.notice netifd: netkeeper (6145): uci: Entry not found
Wed May  3 13:28:59 2017 daemon.info pppd[6155]: Plugin rp-pppoe.so loaded.
Wed May  3 13:28:59 2017 daemon.info pppd[6155]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Wed May  3 13:28:59 2017 daemon.info pppd[6155]: Plugin chongqing_sxplugin.so loaded.
Wed May  3 13:28:59 2017 daemon.info pppd[6155]: sxplugin : init
Wed May  3 13:28:59 2017 daemon.info pppd[6155]: -------------------------------------
Wed May  3 13:28:59 2017 daemon.info pppd[6155]: timenow(Hex)=%ßx
Wed May  3 13:28:59 2017 daemon.info pppd[6155]: Begin : beforeMD5
Wed May  3 13:28:59 2017 daemon.info pppd[6155]: 1.<ή?>
Wed May  3 13:28:59 2017 daemon.info pppd[6155]: 2.<ή?17316741797>
Wed May  3 13:28:59 2017 daemon.info pppd[6155]: 3.<ή?17316741797xianxinli1radius>
Wed May  3 13:28:59 2017 daemon.info pppd[6155]: 4.length=<31>
Wed May  3 13:28:59 2017 daemon.info pppd[6155]: End : beforeMD5
Wed May  3 13:28:59 2017 daemon.info pppd[6155]: Begin : afterMD5
Wed May  3 13:28:59 2017 daemon.info pppd[6155]: 1.MD5use_1=< 7>
Wed May  3 13:28:59 2017 daemon.info pppd[6155]: 2.MD5use_2=< 7>
Wed May  3 13:28:59 2017 daemon.info pppd[6155]: End : afterMD5
Wed May  3 13:28:59 2017 daemon.info pppd[6155]: -------------------------------------
Wed May  3 13:28:59 2017 daemon.info pppd[6155]: sxplugin : user  is <
A`BM` 7717316741797@cqyt> 
Wed May  3 13:28:59 2017 daemon.info pppd[6155]: sxplugin : passwd loaded
Wed May  3 13:28:59 2017 daemon.warn pppd[6155]: [pppoe hook debug] syscall operations name probe pandorabox ok!
Wed May  3 13:28:59 2017 daemon.warn pppd[6155]: [pppoe hook debug] hook status=1
Wed May  3 13:28:59 2017 daemon.notice pppd[6155]: pppd 2.4.7 started by root, uid 0
Wed May  3 13:28:59 2017 daemon.info pppd[6155]: PPP session is 31583
Wed May  3 13:28:59 2017 daemon.warn pppd[6155]: Connected to 28:31:52:57:17:58 via interface eth0.2
Wed May  3 13:28:59 2017 daemon.info pppd[6155]: Using interface pppoe-netkeeper
Wed May  3 13:28:59 2017 daemon.notice pppd[6155]: Connect: pppoe-netkeeper <--> eth0.2
Wed May  3 13:29:02 2017 daemon.info pppd[6155]: syncppp not active
Wed May  3 13:29:03 2017 daemon.info pppd[6155]: CHAP authentication failed: old pin code algorithm ##old pin code algorithm 
Wed May  3 13:29:03 2017 daemon.err pppd[6155]: CHAP authentication failed
Wed May  3 13:29:10 2017 daemon.notice pppd[6155]: Connection terminated.
Wed May  3 13:29:10 2017 daemon.info pppd[6155]: Sent PADT
Wed May  3 13:29:10 2017 daemon.notice pppd[6155]: Modem hangup
Wed May  3 13:29:10 2017 daemon.info pppd[6155]: Exit.
Wed May  3 13:29:10 2017 daemon.notice netifd: netkeeper (6155): connect as 8cde9751
Wed May  3 13:29:10 2017 daemon.notice netifd: Interface 'netkeeper' is now down
Wed May  3 13:29:10 2017 daemon.notice netifd: Interface 'netkeeper' is setting up now
Wed May  3 13:29:10 2017 daemon.notice netifd: netkeeper (6189): uci: Entry not found
Wed May  3 13:29:10 2017 daemon.info pppd[6199]: Plugin rp-pppoe.so loaded.
Wed May  3 13:29:10 2017 daemon.info pppd[6199]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Wed May  3 13:29:10 2017 daemon.info pppd[6199]: Plugin chongqing_sxplugin.so loaded.
Wed May  3 13:29:10 2017 daemon.info pppd[6199]: sxplugin : init
Wed May  3 13:29:10 2017 daemon.info pppd[6199]: -------------------------------------
Wed May  3 13:29:10 2017 daemon.info pppd[6199]: timenow(Hex)=%ßx
Wed May  3 13:29:10 2017 daemon.info pppd[6199]: Begin : beforeMD5
Wed May  3 13:29:10 2017 daemon.info pppd[6199]: 1.<ή?>
Wed May  3 13:29:10 2017 daemon.info pppd[6199]: 2.<ή?17316741797>
Wed May  3 13:29:10 2017 daemon.info pppd[6199]: 3.<ή?17316741797xianxinli1radius>
Wed May  3 13:29:10 2017 daemon.info pppd[6199]: 4.length=<31>
Wed May  3 13:29:10 2017 daemon.info pppd[6199]: End : beforeMD5
Wed May  3 13:29:10 2017 daemon.info pppd[6199]: Begin : afterMD5
Wed May  3 13:29:10 2017 daemon.info pppd[6199]: 1.MD5use_1=< 1>
Wed May  3 13:29:10 2017 daemon.info pppd[6199]: 2.MD5use_2=< 4>
[... 103 lines cut ...]
Wed May  3 13:29:41 2017 daemon.info pppd[6331]: -------------------------------------
[... cut ...]
Wed May  3 13:29:41 2017 daemon.info pppd[6331]: PPP session is 21012
Wed May  3 13:29:41 2017 daemon.warn pppd[6331]: Connected to 28:31:52:57:17:58 via interface eth0.2
Wed May  3 13:29:41 2017 daemon.info pppd[6331]: Using interface pppoe-netkeeper
Wed May  3 13:29:41 2017 daemon.notice pppd[6331]: Connect: pppoe-netkeeper <--> eth0.2
Wed May  3 13:29:44 2017 daemon.info pppd[6331]: syncppp not active
Wed May  3 13:29:45 2017 daemon.info pppd[6331]: CHAP authentication failed: old pin code algorithm ##old pin code algorithm

qq624966793, 2017-05-04:
啊。。@chingcq 大神有解决办法吗

sunflyer, 2017-05-05:
你们用 cqxinliradius002 的那个看看吧。毕竟是MAC客户端的应该还能用

qq624966793, 2017-05-08:
@sunflyer  感谢大神。。。但是下载里面没有cqxinliradius002 这个。。。只有xianxinli1radius这个。。

qq624966793, 2017-05-08:
我尝试了下想下载ubunt编辑so文件。。（不知道能不能编辑）
。。。来自金融系的无奈。当初为啥不学个计算机。。。
看wiki看了半天加上百度。。。但是无奈基础太薄弱。。 @sunflyer

Qlo1062, 2017-05-10:
改了账号算法了

chingcq, 2017-05-11:
@sunflyer cqxinliradius002一样提示CHAP authentication failed: old pin code algorithm ##old pin code algorithm 估计是合川强制使用4.x了

chingcq, 2017-05-11:
@qq624966793 合川这边应该是强制使用4.x了 我刚才编译了cqxinliradius002的so文件一样提示 CHAP authentication failed: old pin code algorithm ##old pin code algorithm

stawidy, 2017-05-17:
有人能测试下mac客户端还能用吗？真抛弃mac用户了？

sunflyer, 2017-05-17:
照这个样子看来，估计是准备上新版MAC客户端了。
重庆目前部分学校已经强制4.7的算法，这里的插件也因为没有算法无法更新。
算是一个蛋疼的麻烦事儿吧

stawidy, 2017-05-17:
营业厅工作人员说新版的mac客户端就是589，他们现在分发的新版mac客户端也是此前官网上的589，没见有区别。

sunflyer, 2017-05-17:
那个589不过是个套牌的589，2.5的算法。