miao1007/Openwrt-NetKeeper #138: 有个可以用路由器的新思路 不知道行不行

chingcq, 2017-05-23:
思路是 在openwrt上面搭一个pppoe server 当电脑端netkeeper拨号时候会向路由器发送用户名和密码的请求 这个时候路由器用拦截到的用户名密码去拨号就行了 这样无论netkeeper怎么变算法应该都能实现路由器拨号吧

sunflyer, 2017-05-24:
这个是可以的。

opUserZangYing, 2017-05-24:
这个6……

miao1007, 2017-05-24:
用tcpdump是可以的，可以抓到PPPoE的包，但是设计到解析pcap文件，这一步比较费时间，可以考虑用tshark等开源实现，最终难度不大，就是非常费时间。

stawidy, 2017-05-24:
@miao1007 但是在路由器上抓包分析挺考验硬件性能吧？

miao1007, 2017-05-24:
不会的，我抓过直接蹭网的人，没有问题。HTTP明文看的清清楚楚。

aiyijing, 2017-05-24:
其实我也试过。pppoe连接显示Unable to complete PPPoE Discovery 这个意思就是说服务器根本就没响应这个ppp连接，不知道是不是rp-pppoe的问题。还有就是账号和密码其实就在pppoe server的日志文件中，不用抓包。简单的用一下grep cut 等等命令就可以截取下来了。这是我唯一想到的办法了。居然都没实现。只能悲哀的中继cqupt 将就的用wifi了。

zaiweijian, 2017-05-26:
@chingcq 请问你是不是有一个关于编译arm梅林固件专用的so文件的项目，我想请教下梅林编译NetKeeper的so文件的方法？

chingcq, 2017-05-27:
@zaiweijian 我没试过梅林固件 一直用的openwrt .so的编译应该跟openwrt一致吧

chingcq, 2017-05-27:
@aiyijing syncppp not activey 应该是把”pin code checked failed“的消息覆盖了 用户名或密码不对 是不是你截取的时间过长导致用户名过期

aiyijing, 2017-05-27:
@miao1007 @chingcq @sunflyer 我现在已经搞定了。是前面那个特殊字符串的问题。但是搞不懂，为什么我路由器重启后居然又自动拨号成功了，我是mac。我写了两个插件，makefile不是太会写，不方便发出来，如果能得到你们的帮助，那是最好不过的了。

zaiweijian, 2017-05-27:
@ aiyijing   前面那个/r就是专门坑人的，就是让你不能拨号（因为大多数路由器pppoe拨号不能识别/r换行符）。那个截取的正确账号是在一段时间内都是有效的，你可以使用多次，所以重启后可以再自动拨号成功。还有期望大神早日搞出来代码造福社会。。。。。

aiyijing, 2017-05-27:
@zaiweijian 代码其实很简单，最开始我也是想弄一个抓包。代码都写完了，shell也行完了，我也在用。就差一个完整的Makefile。如果有人帮写一下Makefile 就发出来。

zaiweijian, 2017-05-27:
@aiyijing这只能指望上面的几个大神做了。。据我所知diybox应该就是这样的实现方式。

aiyijing, 2017-05-27:
@zaiweijian 都弄好了。不过配置就有点复杂了。

GageChan, 2017-05-27:
@aiyijing 大佬  发出来吧   我的路由器扔那儿快吃灰啦😄

zaiweijian, 2017-05-28:
@aiyijing那就发出来呗，把教程弄出来就好了

huipengly, 2017-05-29:
根据楼主思路弄出来了，只需要脚本就可以。不需要编译so文件。
没写过脚本，写的应该很差。这里抛砖引玉，希望大神能帮忙把脚本改好。
[netkeeper4.x pppoe服务器拦截账号法](https://github.com/huipengly/Openwrt-NetKeeper/tree/master/netkeeper4-use-pppoer-server)
不保证能用，不能用的参考下脚本思路，自己改吧，我基本把思路都写出来了。~~因为要搜每条怎么实现~~

sunflyer, 2017-05-29:
@huipengly PPPOE SERVER可以考虑配置成PAP模式而不是CHAP，然后看日志里面能不能出现密码。这样的话就不用设置OP里面的密码了。

miao1007, 2017-05-29:
#139 

楼上学习Shell速度好快哇，推荐一本书 《Linux Shell脚本攻略（第2版） (图灵程序设计丛书)》

chingcq, 2017-05-29:
楼上好人一生平安😂

huipengly, 2017-05-29:
@sunflyer 我晚上试一试，这个配置也是搜出来的，具体区别我也不懂。

zkh1992, 2017-06-01:
@huipengly  请问下3.12的rp-pppoe的ipk去哪才下得到啊，我是mt7620的CPU路由器，找了半天都找不到，只有3.11的，3.11的有好多问题

elimsc, 2017-06-02:
昨天还可以，今天电信加了心跳，好像不行了

colorlight, 2017-06-02:
opkg install rp-pppoe-server 提示没有这个包怎么办

zkh1992, 2017-06-02:
@colorlight 你要去14.07的openwrt里面的package找，好像是在oldpackage里面才有，而且只有3.11版本的

QinFanhua, 2017-06-03:
用3.11版本的脚本，发现wan口的账号和系统日志里显示拨出去的账号不同。
wan口：\r^J5RC]5 9418888888888  系统日志：^M^J5RC]5 9418888888888
请教一下

miao1007, 2017-06-03:
用dos2unix命令转一下试试，你这个没有问题

HuLian21, 2017-06-04:
闪讯无非是向那么几个端口发数据包，其中一个端口（A)是账号密码登录用的，那么如果心跳不走A端口的话，我觉得有的操作。
在上面前提成立情况下，那么按照楼主的思路，能不能做特定透明代理，A端口数据来自截取的账号密码或者由插件生成，其他心跳包由闪讯客服端生成该怎么发怎么发，这样的话就能完美规避心跳了。
当然适用面不是很广，毕竟电脑还要开。

再有客户端可以开wifi，如果路由器能接上信号，做负载均衡，那理论上有一台电脑在线就不会出现因为心跳断网的问题。当然我没考虑这段线路质量问题，桥接差也没得玩。

QSJHUA, 2018-03-23:
广东天翼客户端可以用？？？？

huipengly, 2018-03-24:
@QSJHUA 客户端只是加密账号就可以。

QSJHUA, 2018-03-25:
哦，我的是广东天翼客户端手机号+密码的。