miao1007/Openwrt-NetKeeper #142: [bug report]pppoe.log格式不一致

kuri-leo, 2017-06-03:
路由器: WNDR 3800
OS: LEDE LEDE Reboot 17.01.0-rc2
rp-pppoe-server 版本: 3.12-1
地点: CQUPT

根据issue #136 提供的新方法，测试无法成功，然后看了一下，似乎是pppoe.log的格式不对
我这边的格式为：
~~~
Plugin /etc/ppp/plugins/rp-pppoe.so loaded.
RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Connected to xx:xx:xx:xx:xx:xx via interface br-lan
Using interface ppp1
Connect: ppp1 <--> br-lan
no PAP secret found for ^M1=WA<- c1xxxxxxxxxxx@cqupt
PAP peer authentication failed for ^M1=WA<- c1xxxxxxxxxxx@cqupt
Connection terminated.
~~~
但根据nk4脚本，应该是包含`network.wan.username`和`network.wan.password`字段才对。。。

所以似乎脚本并不能通用Orz

gmfghp, 2017-06-03:
嗯，不能读取密码，而且，账号倒是可以改一下，我用的高通的CPU是这种情况。

kuri-leo, 2017-06-03:
@gmfghp 密码的话就是原始密码吧？不过账号是不是不能显示转义字符啊，我看pppoe.log里面并没有转义支付

gmfghp, 2017-06-03:
嗯，密码没变，账号的话是/r后面的

colorlight, 2017-06-05:
同样遇到了这种问题

lijinwill, 2017-06-05:
同问，用此方法没法成功。。。

kuri-leo, 2017-06-05:
@colorlight @lijinwill 目前来看，这个和路由器的CPU有关

我的思路是，用grep抓`@cqupt`可以把`no PAP secret found for ^M1=WA<- c1xxxxxxxxxxx@cqupt`和`PAP peer authentication failed for ^M1=WA<- c1xxxxxxxxxxx@cqupt`抓出来，然后用`dos2unix`转码得到真实账号

然而，目前的问题是
1. grep得到的内容存不进变量。。。（不会shell编程）
2. 似乎LEDE上没有dos2unix

所以目前仍然是无解状态。。。

colorlight, 2017-06-05:
这个功能脚本里已经实现了啊，问题是，还是这种情况
`   username=$(grep "No CHAP secret found for authenticating" /tmp/pppoe.log  | tail -n 1 | cut -b 43-)`
` uci set network.netkeeper.username="\r$username"`

Destinyzxn, 2017-06-05:
我9331的没有问题 可以用

gmfghp, 2017-06-05:
能获取密码？，，账号是可以获取的，密码不行呀

Destinyzxn, 2017-06-05:
@gmfghp 密码不行

kuri-leo, 2017-06-05:
@colorlight 之前没这个…只有grep network.wan.username的

@gmfghp 他之前的密码也是从pppoe.log中读取的，似乎密码也被加密了？

gmfghp, 2017-06-05:
grep命令在PPPOE.log里面搜索不到“network.wan.username”这个语句，3.12版本的我这里用9331的CPU连账号都获取不到。

Destinyzxn, 2017-06-05:
@gmfghp 你用3.11的那个脚本 只获取账号那个 就可以

gmfghp, 2017-06-05:
嗯，我改的3.12的nk4.sh的语句。可以获取账号，但是，PPPOE.log里面没有密码。

Destinyzxn, 2017-06-05:
虽然不能获取密码 但是也够了 @gmfghp

gmfghp, 2017-06-05:
嗯是的，但是，为啥 mtk的 cpu就可以呢？手上没有mtk的路由 也没试过， 但是有人说可以获取密码的，想知道是不是 pppoe server 的问题，我9331 的用3.12 的 就是获取不到密码、。

Destinyzxn, 2017-06-05:
我k2也不能获取到密码 就算是 原版lede openwrt  安装的是3.12   也不行

gmfghp, 2017-06-05:
哦？，用，原版的 潘多拉。 pandora   测试使用的潘多拉。  不是经过测试了吗、？ 说明里面 写的

kuri-leo, 2017-06-05:
@Destinyzxn 如果他那个算法没有修改密码，讲道理把 `nk4.sh` 中的 `uci set network.netkeeper.password="$(grep "network.wan.password" /tmp/pppoe.log  | tail -n 1 | cut -b 64-)"`改成你的密码就能行了

gmfghp, 2017-06-05:
@kuri-leo  ，哦，这个 还可以@ 人的。 主要就是怎么吧密码加到自动获取里面， 可以加到@。。。的后缀里面吗比如说，账号123456@qwer 密码567890 可以这样吗， 直接在帐号里面所手脚，123456@qwer567890 然后直接从账号里面提取真实账号， 并且可以在输入的账号里面吧密码也提取出来。 前提是@qwer这个后缀不参与运算。怎么样，能实现吗？

Destinyzxn, 2017-06-05:
哎呀 无所谓啦  反正手动输入也不麻烦  不存在的

gmfghp, 2017-06-05:
那是， 能用就已经很叼了，主要是追求完美，这毛病我得改改。

kuri-leo, 2017-06-05:
@gmfghp 然而并没有这种操作。。。

我猜的，在mtk下的`rp-pppoe-server`会记录下`network.wan.username`和`network.wan.password`两个字段，然后nk4.sh通过分别对log进行grep来获取真实username和password

所以，不需要你说的那个操作，直接在nk4.sh 的基础上改一下，把 `uci set network.netkeeper.password="$(grep "network.wan.password" /tmp/pppoe.log | tail -n 1 | cut -b 64-)"`改成`uci set network.netkeeper.password="你的密码"`就可以了

当然，这是在密码不变的前提上。。。

Destinyzxn, 2017-06-05:
然而每个月都要换一次动态密码的

kuri-leo, 2017-06-05:
@Destinyzxn 就用静态密码呀。。。我用了一年多都没出问题

Destinyzxn, 2017-06-05:
我学校有心跳 只能动态密码 @kuri-leo

kuri-leo, 2017-06-05:
@Destinyzxn 节哀顺变23333

~~其实你们可以去工信部举办，传说弱校当年就是这样把心跳干掉的~~

gmfghp, 2017-06-05:
@kuri-leo Connect: ppp1 <--> br-lan
no PAP secret found for ^M1=WA<- c1xxxxxxxxxxx@cqupt
PAP peer authentication failed for ^M1=WA<- c1xxxxxxxxxxx@cqupt
Connection terminated.
就是这样的呀，没有network.wan.password这个， 是3.12的版本， 但是是9331 的cpu  也是pap模式， 不显示密码的。
grep "network.wan.password" /tmp/pppoe.log
这个不就是搜索/tmp/pppoe.log 这个里面的字符“network.wan.password”然后截取的 。然而这个在9331 的里面根本没有这个字符串，在log里面不显示 密码。所以就不能自动获取了。

Destinyzxn, 2017-06-05:
怕是没这么简单哟 @kuri-leo

colorlight, 2017-06-05:
3.12版本是在哪获取的

gmfghp, 2017-06-05:
@colorlight lede直接安装 就有

kuri-leo, 2017-06-05:
@gmfghp 对啊，我说的改不是改`pppoe.log`，而是把`nk4.sh`的代码改了

1. 把`username=$(grep "network.wan.username" /tmp/pppoe.log  | tail -n 1 | cut -b 66-)`改成`username=$(grep "No CHAP secret found for authenticating" /tmp/pppoe.log | tail -n 1 | cut -b 43-)` （感谢@colorlight 提供）
2. 把`uci set network.netkeeper.password="$(grep "network.wan.password" /tmp/pppoe.log | tail -n 1 | cut -b 64-)"`改成`uci set network.netkeeper.password="你的密码"` （我猜的，不知道行不行）

改完最好重新部署一次试试

colorlight, 2017-06-05:
不行，我试过了

kuri-leo, 2017-06-05:
@colorlight 我的是LEDE Reboot 17.01.0-rc2，直接opkg install就可以

你如果没有的话可以去找一下OpenWrt的14版本的包试一试

kuri-leo, 2017-06-05:
@colorlight 我之前遇到的问题是 grep 获得的字符串无法赋值给变量。。。不知道是不是同样的问题

gmfghp, 2017-06-05:
@kuri-leo @colorlight 不行呗， 后边代码没改呢， 你要这样改就直接在3.11版本的后边加上这句
把uci set network.netkeeper.password="$(grep "network.wan.password" /tmp/pppoe.log | tail -n 1 | cut -b 64-)"改成uci set network.netkeeper.password="你的密码"  然后添加到 3.11版本的nk4，sh里面

kuri-leo, 2017-06-05:
@Destinyzxn 先给10000打电话投诉，就说频繁掉线，然后客服一般会打哈哈，然后你就去工信部网站投诉。。。

kuri-leo, 2017-06-05:
@gmfghp 我说的就是这个意思，改了的代码应该是这样的：

```
#!/bin/sh
#启动pppoe服务器。TODO：检测是否有pppoe服务器进程，再启动
sleep 1
pppoe-server -k -I br-lan

#删掉之前的log，加快读取速度
rm /tmp/pppoe.log

while :
do
#读取log最后一个账号
    username=$(grep "No CHAP secret found for authenticating" /tmp/pppoe.log | tail -n 1 | cut -b 43-)

    if [ "$username" != "$username_old" ]
    then
        ifdown netkeeper
        uci set network.netkeeper.username="\r$username"
        uci set network.netkeeper.password="你的密码"
        uci commit
        ifup netkeeper
        username_old="$username"
        echo "new username $username"
    fi
#    echo "wait"
    sleep 10

done
```

然而，我遇到的问题是`$(grep "No CHAP secret found for authenticating" /tmp/pppoe.log | tail -n 1 | cut -b 43-)`的值根本没法存到`username`里面去

gmfghp, 2017-06-05:
把username=$(grep "network.wan.username" /tmp/pppoe.log | tail -n 1 | cut -b 66-)改成username=$(grep "No CHAP secret found for authenticating" /tmp/pppoe.log | tail -n 1 | cut -b 43-) （感谢@colorlight 提供）
这里 ，如果用的是3.12 版的 。就不是chap了 是pap
是这个字符串了no PAP secret found for ^M1=WA<- c1xxxxxxxxxxx@cqupt，而且没有authenticating
应该改成 username=$(grep "no PAP secret found for" /tmp/pppoe.log | tail -n 1 | cut -b 24-)
据我研究，好像是这样的。。。。好像学编程，  太有意思了，不做职业的就做业余的，prefect，。嘻嘻

kuri-leo, 2017-06-05:
@gmfghp 我用的是3.12的。。。没仔细看，反正就是这个意思

另外，我建议是`grep @CQUPT`,来抓取账号，直接回避版本问题。。。~~当然，如果你用的手机号做账号当我啥都没说。。。~~

gmfghp, 2017-06-05:
@kuri-leo 抓这个后缀，可以但是实现起来得加代码， 我不懂编程，反正这么截取得加代码的，没有这样的简单。，就像我说的把密码加到@后缀里面，都得加代码的。，还不如自己手动输入，毕竟不会编程。，，，等大神的自动识别的版本吧，。估计会像你说的这样截取账号。

kuri-leo, 2017-06-05:
@gmfghp 我试过，直接`grep "@cqupt"`就可以，然后会返回两行字符串
```
no PAP secret found for ^M1=WA<- c1xxxxxxxxxxx@cqupt
PAP peer authentication failed for ^M1=WA<- c1xxxxxxxxxxx@cqupt
```

gmfghp, 2017-06-05:
@kuri-leo 对，大概意思是 “搜索 带有@cqupt的行” ，然后列出，然后，你就可以在这两行里面截取你需要的信息了

lijinwill, 2017-06-05:
@kuri-leo 你解决这个问题了吗？？
`RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7

Plugin /etc/ppp/plugins/rp-pppoe.so loaded.

RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7

Connected to 00:e0:4c:36:09:14 via interface br-lan

Using interface ppp0

Connect: ppp0 <--> br-lan

PPP session is 23377

Connected to 74:5a:aa:c0:da:f3 via interface eth0.2

Using interface pppoe-netkeeper

Connect: pppoe-netkeeper <--> eth0.2

no PAP secret found for ^M1#V2Z(A3a15310287720@cqupt

PAP peer authentication failed for ^M1#V2Z(A3a15310287720@cqupt

Terminating on signal 15

Modem hangup

Connection terminated.

Remote message: Username does not exist.##Username does not exist.

PAP authentication failed

Modem hangup

Connection terminated.

Sent PADT`

kuri-leo, 2017-06-05:
@lijinwill 这个应该是pppoe.log吧，有个大概的思路，~~等过几天有空我试试吧~~

lijinwill, 2017-06-05:
我改了密码为自己的六位密码，没用。。。
还有，我只是奇怪为什么4.7.9这个版本没有人像之前那样反编译客户端，目前已知有pc，mac，iPhone客户端。我试着用hopper disassenbler反编译mac客户端，没有结果，4.7.9 mac端有防反编译的措施，并不像2.5那样可以看到程序部分。

kuri-leo, 2017-06-05:
@lijinwill 因为PC端的也加了壳防反编译啊Orz

lijinwill, 2017-06-05:
刚才看了下，iphone客户端也没戏：（

gmfghp, 2017-06-06:
@kuri-leo 我弄成了，可以自动获取密码了。 很方便。嘻嘻。

kuri-leo, 2017-06-06:
@gmfghp 恭喜

我这边还是有问题，目前改出来的是这样，但是似乎我切割username的地方有问题。。。

```
#!/bin/sh
#启动pppoe服务器。TODO：检测是否有pppoe服务器进程，再启动
sleep 1
pppoe-server -k -I br-lan

#删掉之前的log，加快读取速度
rm /tmp/pppoe.log

while :
do
#读取log最后一个账号
    username=$(grep "PAP peer authentication failed for" /tmp/pppoe.log | tail -n 1 | cut -c 35-)

    if [ "$username" != "$username_old" ]
    then
        ifdown netkeeper
        uci set network.netkeeper.username="\r$username"
        uci set network.netkeeper.password="密码"
        uci commit
        ifup netkeeper
        username_old="$username"
        echo "new username $username"
    fi
#    echo "wait"
    sleep 10

done

```

gmfghp, 2017-06-06:
@kuri-leo 
username=$(grep "PAP peer authentication failed for" /tmp/pppoe.log | tail -n 1 | cut -c 37-)
这里改成37
然后你可以参考我那个直接获取密码。
我现在在研究编程，以前学的一点点 c 都快捡不起来了。