miao1007/Openwrt-NetKeeper #143: 自己研究的可以自动获取密码的代码。直接把密码加到账号里面。亲测可用

gmfghp, 2017-06-06:
上网查的代码，研究了半天，成功获取到了密码。 账号上面做手脚，如zhanghao@houzhui/mima 
也验证了后缀不参与账号的加密运算。
编程真的太有意思了，
这是 用的 3.11版本的， 如果用其他的 可以自己参照修改。

#!/bin/sh
#启动pppoe服务器。TODO：检测是否有pppoe服务器进程，再启势
sleep 1
pppoe-server -k -I br-lan

#删掉之前的log，加快读取速度
rm /tmp/pppoe.log

while :
do
    var=$(grep "No CHAP secret found for authenticating" /tmp/pppoe.log  | tail -n 1 | cut -b 43-)
    username=${var%/*}

    if [ "$username" != "$username_old" ]
    then
        ifdown netkeeper
        uci set network.netkeeper.username="\r$username"
        uci set network.netkeeper.password="${var##*/}"
        uci commit
        ifup netkeeper
        username_old="$username"
        echo "new username $username"
    fi
#    echo "wait"
    sleep 10

done

Destinyzxn, 2017-06-06:
在账号后面加密码  每次创翼拨号都要加    和     在路由器里面设置一次密码     算起来不如自己去路由器里面设置

gmfghp, 2017-06-06:
@Destinyzxn 哪里， 这样自动获取的话，可以随意的更换账号， 非常方便。我今天特意试了。

Destinyzxn, 2017-06-06:
可是为啥要切换账号呢

gmfghp, 2017-06-06:
我这里有好几个账号，如果有人用的话，我就得换另一个帐号，这样我还得进入路由器设置一下密码， 很麻烦， 但是自动获取的话就简单多了。

closed 2017-06-10 by gmfghp