miao1007/Openwrt-NetKeeper #145: pppoe-server帐号密码获取方法，日志账号提取求助

huipengly, 2017-06-08:
前几天一直在准备答辩，没时间看这个。今天答完了，发现之前没仔细看日志，以为版本问题。今天重新搜了一下。
结论：**无法获取账号密码的原因跟pppoe-server版本无关。**
方法：需要在/etc/ppp/option里打开debug，并且添加show-password。这样就能在日志里看到帐号密码了。logread命令看到的格式如下。
```
Thu Jun  8 20:09:21 2017 daemon.debug pppd[3149]: rcvd [PAP AuthReq id=0xf user="username" password="passwd"]
```
这样就可以直接提取了。但是**用户名那里什么符号都可能出现**，我没搜到很好地提取方法，来求助。

miao1007, 2017-06-08:
首先定义一个函数
```sh
function regex { gawk 'match($0,/'$1'/, ary) {print ary['${2:-'0'}']}'; }
```

接着进行测试
```sh
echo "Thu Jun  8 20:09:21 2017 daemon.debug pppd[3149]: rcvd [PAP AuthReq id=0xf user="username" password="passwd"]"|regex 'user=(.*) password' 1
```

最终应该是这样的，我没有测

```sh
pwd=`tail -f xxx.log|sed -e "/user=/q" | regex 'user=(.*) password' 1`
echo $pwd
```

然后自己在折腾一下就可以了

gmfghp, 2017-06-08:
4、破解路由器会自动重拨对AAA产生大量拨号
解决方法：需亚信平台进行观察，防止拥塞的现象发生。

 这是 文件里的， 意思会封号吗？
这个咋么弄呀，，把断线后 自动重拨关掉，怎么操作呀，大佬研究下。

huipengly, 2017-06-09:
已更新脚本，但是如果账号里出现双引号"，会获取失败。希望有人能解决。

log格式
```
rcvd [PAP AuthReq id=0xf user="username" password="passwd"]
```
我的处理方法：
```
username=$(grep 'user=' /tmp/pppoe.log | grep 'rcvd' | tail -n 1 | cut -d \" -f 2)
```

gmfghp, 2017-06-10:
开始也用了你的截取第几段我截取的"="然后第一次拨号测试，账号就出来一个  \r1   很尴尬账号符号里面出来了一个=，天意呀。然后干脆弄成这样了。大佬看一下把。
#read the last username in pppoe.log
var=$(grep 'user=' /root/pppoe.log | grep 'rcvd' | tail -n 1)
name=${var#'"'}
username=${name%'" password="'}
word=${var#'" password="'}
password=${word%'"'}

gmfghp, 2017-06-10:
发错了，尴尬 。把修改的副本发过来了，下本这个是修改测试过的。
while :
do
	#read the last username in pppoe.log
	var=$(grep 'user=' /root/pppoe.log | grep 'rcvd' | tail -n 1)
	name=${var#*'"'}
	username=${name%'" password="'*}
	word=${var#*'" password="'}
	password=${word%'"'*}

	if [ "$username" != "$username_old" ]
	then

gmfghp, 2017-06-10:
我擦， 不对呀，？？？ 为啥显示不出来呀， 没发错呀。
直接上传个 txt把   奇怪了。
${*****************}
下载了 然后修改到nk4文件里面吧
[nk4_xiugai.txt](https://github.com/miao1007/Openwrt-NetKeeper/files/1065923/nk4_xiugai.txt)

XiaoCool007, 2017-11-08:
装好ipk和执行完sh后用netkeeper拨号，报错651，路由器没有明显相关日志是什么问题

XiaoCool007, 2017-11-08:
南昌老算法似乎现在已经不能在用了，好像全部4.8了，希望能抱大佬的大腿

gmfghp, 2017-11-08:
在执行一下试试吧，651 应该是，怕婆婆儿server 没有开开。