miao1007/Openwrt-NetKeeper #146: 更新了获取帐号密码脚本，测试反馈到这里吧

huipengly, 2017-06-09:
更新后，版本无关，都可以直接获取帐号密码，然后断网之后会自动关闭pppoe拨号。
我暂时只测了，openwrt 14.09/17.01，pppoe-server 3.11。
马上要毕业了，也不再用这个坑爹网络了，应该就没条件测试了。去别的学校也不知道会不会还要战斗。
滚蛋前几天我尽量完善脚本吧。

gmfghp, 2017-06-09:
出现错误628；在连接完成前，连接被远程计算机终止。
和错误619；不能建立到远程计算机的连接，因此用于此连接的端口已关闭。
我是 9331 的 刷的lede 用的3.12的server。

huipengly, 2017-06-09:
@gmfghp 先看一下pppoe-server启动成功没，把pppoe.log发出来看看，看有没有帐号信息。
uci那句话是读取wan口的eth然后给netkeeper口。脚本没有删除wan口，我们学校需要留着上内网，你可以自己删掉。
mac地址那句删不删好像没什么影响，我是直接在原来的配置文件复制的。

yiheyang, 2017-06-09:
在孵化基地刚刚测试成功 孵化基地不断网
而且不用网线插lan 直接wifi连上nk拨号就好
前一段时间还在郁闷openwrt怎么办而且用不了simplenetkeeper
太棒了！！！

yiheyang, 2017-06-09:
我之前nk2.x成功过多nk账号多拨 用你这个思路应该也可以实现

gmfghp, 2017-06-10:
@huipengly 今天试了下。
一。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。
     今天看了下 pppoe.log报错是这样的，昨天太晚了脑袋不清醒也没注意。
pppd: Error relocating /etc/ppp/plugins/rp-pppoe.so: new_style_driver: symbol not found
pppd: Couldn't load plugin /etc/ppp/plugins/rp-pppoe.so
找到原因了:这两句没加上，加上之后可以了。
cp /etc/ppp/plugins/rp-pppoe.so /etc/ppp/plugins/rp-pppoe.so.bak
cp /usr/lib/pppd/2.4.7/rp-pppoe.so /etc/ppp/plugins/rp-pppoe.so

二。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。。
   读取的不带有”的账号，在拨号过程中已经出现拨号状态了然后拨号会失败（中断）。猜测可能是  sleep 5 时间短了点 的问题呢？  然后我改成了7 ，测试了下，感觉可以了。

运行一切正常，除了“的情况不能读取。我再测试下，大佬再研究下"的问题吧，我想了下，能不能吧出现“的账号分割一下，吧 ” 两侧的账号分割后再合并，然后中间加上“   
有的时候也会出现两个 ”    “    这就不好弄了。要是还有其它不能读取的符号也不好弄。

huipengly, 2017-06-10:
@gmfghp 之前测试遇到过rp-pppoe.so这个错误，后来测就没有了，所以删了那句话。看来还是有必要留着，已经重新添加上了。
分割那个好像也是个不错的想法，写个循环检测，我去试试。

Destinyzxn, 2017-06-10:
K2成了 3.11也可自动获取密码

gmfghp, 2017-06-10:
@huipengly 我试了下。这样写怎么样。应该可以读取所有帐号类型了。
#read the last username in pppoe.log
	var=$(grep 'user=' /root/pppoe.log | grep 'rcvd' | tail -n 1)
	name=${var#*'"'}
	username=${name%'" password="'*}
	word=${var#*'" password="'}
	password=${word%'"'*}

还有就是，这里的 sleep 5
        logger -t nk4 "new username $username"
        
        sleep 5
    fi
    
    #close pppoe if log fail
    if [ -z "$(ifconfig | grep "netkeeper")" ]
    then
        ifdown netkeeper
那个 sleep 5 是不是把时间改大1-2秒？避免还没完成拨号就关闭了端口 
if [ -z "$(ifconfig | grep "netkeeper")" ]我刚才运行ifconfig 实测了一下，5秒感觉确实出不来。我的理解对吗，这块不太懂。

gmfghp, 2017-06-10:
@huipengly 有点晕了，好像改掉了呢，7分钟前，   怪了， github玩不太6 呢。嘻嘻。

huipengly, 2017-06-10:
@gmfghp 是看到你发的我才改的。昨天看日志5s是可以拨上号的。既然有可能拨不上号，就调久一点嘛。  
因为多次拨号失败电信那边会暂时禁止当前设备拨号，写的时候就把间隔设置的稍微短一点。

gmfghp, 2017-06-10:
@huipengly 看下我那个提取的方法可以吗，我也不太懂，但是要是出两个 “ ” 也会进行拨号，然后我试了下我原来把密码加到账号里面的提取方法。貌似可以用。
#read the last username in pppoe.log
var=$(grep 'user=' /root/pppoe.log | grep 'rcvd' | tail -n 1)
name=${var#'"'}
username=${name%'" password="'}
word=${var#'" password="'}
password=${word%'"'}

gmfghp, 2017-06-10:
@huipengly 哪个账号提取的 ，发上来不能全部显示。真是醉了。
把这个txt的后缀改下把。
[nk4.txt](https://github.com/miao1007/Openwrt-NetKeeper/files/1065930/nk4.txt)

HuLian21, 2017-06-16:
@yiheyang  请问你是直接加入网络以后选择ppoe协议还是选择桥接模式呢

simonla, 2017-06-21:
每天早上要开电脑重连，有解吗

F-Light, 2017-07-06:
[root@PandoraBox_6D00:/root]#sh nk4conf.sh
uci: Entry not found
 * Flushing IPv4 filter table
 * Flushing IPv4 nat table
 * Flushing IPv4 mangle table
 * Flushing IPv4 raw table
 * Flushing IPv6 filter table
 * Flushing IPv6 mangle table
 * Flushing IPv6 raw table
 * Flushing conntrack table ...
 * Populating IPv4 filter table
   * Zone 'lan'
   * Zone 'wan'
   * Rule 'Allow-DHCP-Renew'
   * Rule 'Allow-Ping'
   * Rule 'Allow-IGMP'
   * Rule #7
   * Rule #8
   * Forward 'lan' -> 'wan'
 * Populating IPv4 nat table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv4 mangle table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv4 raw table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv6 filter table
   * Zone 'lan'
   * Zone 'wan'
   * Rule 'Allow-DHCPv6'
   * Rule 'Allow-MLD'
   * Rule 'Allow-ICMPv6-Input'
   * Rule 'Allow-ICMPv6-Forward'
   * Rule #7
   * Rule #8
   * Forward 'lan' -> 'wan'
 * Populating IPv6 mangle table
   * Zone 'lan'
   * Zone 'wan'
 * Populating IPv6 raw table
   * Zone 'lan'
   * Zone 'wan'
 * Set tcp_ecn to off
 * Set tcp_syncookies to on
 * Set tcp_window_scaling to on
 * Running script '/etc/firewall.user'
 * Running script '/usr/share/miniupnpd/firewall.include'
 * Running script '/var/etc/shadowsocksr.include'
   ! Skipping due to path error: No such file or directory
 * Running script '/var/etc/koolproxy.include'
   ! Skipping due to path error: No such file or directory

我的是newifi mini 刷的潘多拉，是不是我别的设置有了问题

F-Light, 2017-07-06:
[http://www.lbysg.com/20170706235312.png](url) 
[http://www.lbysg.com/20170706235105.png](url)
这是我抓的包，怕不是就是简单的0-9 a-z A-Z   ++

MSfans, 2017-09-05:
PAP peer authentication failed for ^M^JMV09?A4a07189428742
sent [LCP TermReq id=0x2 "Authentication failed"]
rcvd [LCP EchoRep id=0x0 magic=0x33744d79]
rcvd [LCP TermAck id=0x2 "Authentication failed"]
大大您好，账号密码都获取到了，最后出现这三个错误？
请问如何解决，在此谢过

MSfans, 2017-09-07:
@Zhuhao0922 同学，你和我一样的情况呢