miao1007/Openwrt-NetKeeper #148: 用netkeeper上网之前必须得有一个内网IP吗？

caiqiqi, 2017-06-12:
想请问各位大佬一个问题，要想成功登录netkeeper，是不是必须netkeeper所在设备必须有一个内网IP啊。印象中好像是得有，但是为什么我在这个环境下，没有内网IP也可以成功登录netkeeper呢？
[image]
这个是不是跟我的网络所在的网关设备的行为有关呢？

lijinwill, 2017-06-19:
不需要内网IP，只需要和学校的pppoe服务器在一个广播域即可。因为抓包分析可知，netkeeper拨号是纯二层交换的，就是建立连接的过程中只发mac包，所以和有无内网IP没有关系！

caiqiqi, 2017-06-23:
@lijinwill 哦，我对PPPOE协议不是很熟悉。对广播域的概念也理解的不透。但是我发现有这么三种场景。第一，也是最常见的，就是PPPOE客户端有内网IP，即直接在校园内网中；第二，PPPOE客户端通过有线以太网连接在某个处在内网中的设备上，而客户端本身没有内网IP，只有一个以这个边界设备作为网关的局域网IP，比如图中的10.0.0.77/24。我知道所有172.16.0.0/12中的主机是在一个广播域，但是这个主机10.0.0.77/22与内网的172.16.0.0/12是不是同一个广播域呢？第三，就是某个处在172.16.0.0/12内网中的设备是一台无线路由器，通过发WIFI出来，作为这个WIFI的局域网的网关，然后处在这个WIFI局域网的主机是不能登录netkeeper的，是因为这个边界路由器做了NAT吗还是因为802.11跟以太网协议的差异吗还是什么原因呢？这个问题我一直没有弄清楚，困惑着我

lijinwill, 2017-06-24:
第二个问题，不是很清楚10.0.0.77这个IP是怎么来的，还请大神回答。第三个问题，广播域的边界是路由器，你如果用路由器接入172.的内网，是没办法拨通netkeeper的。解决办法是关掉路由器的DHCP协议，这样就可以路由器开Wi-Fi，电脑连接Wi-Fi拨号上网了！

hydejoe, 2017-08-27:
@caiqiqi 对于第二个场景：

> 我知道所有172.16.0.0/12中的主机是在一个广播域

ip地址的广播跟pppoe拨号的广播是不一样的，ip地址的广播是形如“224.0.0.251”、“192.168.1.255”的三层目标地址，而pppoe的广播包目标地址则是二层的“FF:FF:FF:FF:FF:FF”。

一般来说，二层的广播是不能跨路由的，也就是说，如果你跟pppoe服务器不在一个以太网，pppoe包是传不过去的。

在linux的rp-pppoe包中，有一个命令叫pppoe-relay，作为边界设备时，可以转发二层pppoe协议（PADI、PADO、PADS、PADT等），实现跨路由pppoe拨号。

参见：https://linux.die.net/man/8/pppoe-relay

这也就能解释为什么第二个场景下能拨号而第三个场景下不能：第三个场景下的边界路由器没有转发你二层的pppoe包。

如果你将第三个场景下的路由器当作交换机用，即关掉dhcp，外部网络插LAN口，这样wifi里的pppoe包也可以发出去。